OctoPrint/OctoPrint #2305: Update from v1.3.4 to v1.3.6 failed / Webfront corrupted

mrcg, 2017-12-19:
I am using octoprint since very long time (several years) and never had any issues when updating.
While recent update from v1.3.4 to v1.3.6 the update failed.
Webfront seem to be corrupted and therefore octoprint cannot be used any more.

How to fix the corruption caused by the "bad" v1.3.6 update?
I would like to avoid performing re-installation.

Thanks for your help.
Attached the logfile from update.
[update_failed_log.txt](https://github.com/foosel/OctoPrint/files/1573226/update_failed_log.txt)

foosel, 2017-12-20:
Read & follow [this FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#i-got-some-error-after-attempting-to-install-monotonic-or-futures-during-update-and-now-my-server-wont-start). tldr: Issue with your environment being too old, easily worked around - put out an important announcement for this out as well.

foosel, 2017-12-20:
Closing because duplicate of #2289, #2278 and #2193

closed 2017-12-20 by foosel