OctoPrint/OctoPrint #2306: Update Issue 1.3.5 -> 1.3.6 - Missing Functions?

camronnie, 2017-12-20:
#### What were you doing?

Updating from 1.3.5 to 1.3.6.
I deleted my OctoPrint folder in the C drive, and started over from scratch, following the same Windows installation steps that I used to install 1.3.5.

#### What did you expect to happen?

To successfully install 1.3.6.

#### What happened instead?

1.3.6 installs, but I am missing functions.  I no longer have calibration functions, slicing does not initiate my Cura plugin for modification, and when I hit print on a previous part, I do not get a prompt for print temperature.

I tried to revert back to 1.3.5, but the OctoPrint web interface appears to be the same.  To revert I deleted the OctoPrint folder from my C drive, and followed the Windows instructions I followed initially to install 1.3.5 on the Wiki, but the missing functions are still missing.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

1.3.6.

#### Operating System running OctoPrint

Windows 7

#### Printer model & used firmware incl. version

M3D Micro - using iMe / M33 Fio

#### Browser and Version of Browser, Operating System running Browser

Firefox 57.0.2

#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/1575253/octoprint.log)

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

ntoff, 2017-12-20:
1.3.6 changed the way plugins are loaded, plugins that relied on global variables no longer function.

https://octoprint.org/blog/2017/12/01/heads-up-plugin-authors/

It's possible the plugins you rely on will need to be updated by the authors to be compatible with OctoPrint's new system. In the mean time you can open OctoPrint's settings dialog, go to features, and tick the checkbox to enable legacy plugin bundling.

kantlivelong, 2017-12-20:
@camronnie if the legacy option works please let us know as well as open a ticket with the plugin developer.

camronnie, 2017-12-21:
I tried the legacy plugin bundling, but that is not changing anything.  I am still missing the previous functions that I had.

Currently:
[image]

And a few screen shots from https://www.thingiverse.com/groups/m3d-the-micro-3d-printer/forums/general/topic:5096 of what I am missing:

[image]
[image]

I am able to connect to my printer, and move it, but am missing key features to be able to print.  I was also unable to correct this by trying to revert back to 1.3.5.  Maybe I was on 1.3.4.... I'll try that just to be sure.

camronnie, 2017-12-21:
So the same thing is still happening on 1.3.4.  It also seems/feels like the web/user interface is not changing?

I never had a webcam stream screen until I installed 1.3.6, and that is also when my calibration controls went away.

[image]

Is the best way to uninstall to install a different version to simply delete the OctoPrint folder and start over?

kantlivelong, 2017-12-21:
You can delete the config directory. https://github.com/foosel/OctoPrint/wiki/Configuration

Webcam stream can be disabled by clearing out the webcam related settings.

As for M33-Fio:
While it is possible that there could be a bug in OctoPrint you're going to need to open an issue on the M33-Fio repo. This issue tracker is specifically for OctoPrint core issues.

camronnie, 2017-12-22:
Got everything sorted back out, and I'm back on 1.3.5 for now, I'll try 1.3.6 later.  I had to find under my roaming files (Windows OS) an OctoPrint folder that needed to be deleted.  Once that was deleted, and 1.3.5 reinstalled, everything is working correctly.

closed 2017-12-27 by kantlivelong