OctoPrint/OctoPrint #2308: New Problems with Pause / Resume GCODE Commands (Prusa i3)

CoolNamesAllTaken, 2017-12-21:
I updated to the most recent version of Octoprint recently, and things have been working just fine, until I tried to pause a print today.
```
OctoPrint version : 1.3.6
OctoPi version    : 0.14.0
```
## Printer / Octoprint Info
My printer is a Prusa i3 MK2S running the latest firmware, [FW 3.1.0](https://www.prusaprinters.org/fw-3-1-0-linear-advance-drivers-1-9-1/).  The new Prusa firmware was installed in November and worked happily with the previous Octoprint version I had (1.3.5, I think), including numerous pause / resume cycles, but with the update to 1.3.6 the pause and resume commands seem to be having some issues.

The browser I am using is Google Chrome.  This issue has been repeated from Mac OS and Windows 10.

## GCODE Info
Here's what I have in my GCODE scripts:
After print job is paused:
```
M601 ; pause print
```

Before print job is resumed:
```
M602 ; Resume print
G4 P5000 ; Dwell
```

## Bug Description
The issue occurs when I hit "pause" or "resume" during a print job.  Expected events (what used to happen with the previous OctoPrint firmware) and actual events (what occurs with Octoprint 1.3.6) are detailed below.

Expected events on "pause":
Print head stops movements, turns off extruder heater, moves to heating position (in the air above home position) and awaits further instructions.  Bed heater remains on.
"Resume" button appears on online Octoprint interface.

Actual events on "pause":
Print head freezes in current position on build plate, turns off extruder heater.  Bed heater remains on.
"Resume" button appears on online Octoprint interface, but appears to have already been clicked.

Expected results on "resume":
Print head turns on extruder heater, remains in heating position.  Print head reaches print temperature, moves to pause position and resumes printing.

Actual events on "resume":
Print head remains in incorrect paused position (not in heating position).  Octoprint displays target extruder temperature as the correct printing temperature (240˚), but the actual temperature of the print head does not increase.  The printer displays the message "Heating..." on its LCD, but the target extruder temperature on the printer says 0˚.  Printer stays in this configuration for 10+
 minutes (note: bed heater is still on).

## Log file
[octoprint.log](https://pastebin.com/4XJkKsee)
The log is rather messy and includes a full system reboot (raspi and printer), with the printer being rebooted multiple times (shows up as a printer reset detected).

Please let me know if I need to add anything else!  I appreciate the help.

cnlson, 2017-12-21:
I was just coming here to report the same exact thing. so 2nd prusa mk3 with the same issue with 1.3.6

cnlson, 2017-12-23:
@CoolNamesAllTaken no idea why but you are still sitting at incomplete ticket so your issue is not being reviewed

CoolNamesAllTaken, 2017-12-24:
Hmmm...is there anything I’m missing that’s obvious?

It’s good to know that someone else has the same issue!

cnlson, 2017-12-24:
I think all of these have to be in the body: 

1) - I have read the FAQ.
2) #### Link to contents of terminal tab or serial.log
3) #### Browser and Version of Browser, Operating System running Browser

might want to attach the gcode also

CoolNamesAllTaken, 2017-12-24:
Those exact words, or that general content?  I’ve included 2) and 3).

It could also be that the devs are on break for Christmas...

cnlson, 2017-12-24:
It's automatic, so for sure #1

cnlson, 2017-12-24:
bingo  - @GitIssueBot GitIssueBot added type:potential bug and removed type:incomplete ticket labels 4 hours ago

CoolNamesAllTaken, 2018-01-11:
Boop!  I'm hoping this is in the right place?

cnlson, 2018-01-11:
I think this was fixed with a firmware update. I'm not running octoprint at all right now until the prusa firmware gets sorted.

foosel, 2018-01-11:
Please always use the issue template and fill it out completely. Because then I don't have to run after stuff like a terminal output/`serial.log` all the time, which I now once again have to ;)

If this bit from `octoprint.log` however was triggered by you pausing your print:

```
2017-12-20 17:10:51,331 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 689581, current line = 689582
| Last lines in terminal:
| Recv: ok
| Send: N689573 G1 X160.837 Y121.160 E0.01849*94
| Recv: ok
| Send: N689574 G1 X160.716 Y121.847 E0.02362*88
| Recv: ok
| Send: N689575 G1 X160.569 Y122.356 E0.01793*82
| Recv: ok
| Send: N689576 G1 X160.422 Y122.966 E0.02123*88
| Recv: ok
| Send: N689577 G1 X160.163 Y123.804 E0.02971*82
| Recv: ok
| Send: N689578 M105*26
| Recv: ok T:237.7 /240.0 B:90.0 /90.0 T0:237.7 /240.0 @:121 B@:75
| Send: N689579 G1 X159.930 Y124.718 E0.03192*89
| Recv: ok
| Send: N689580 G1 X159.868 Y125.161 E0.01515*83
| Recv: ok
| Send: N689581 G1 X159.838 Y125.567 E0.01377*87
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 689580
| Recv: Resend: 689581
```

then your problem is caused by a bug in the Prusa Marlin fork as reported in prusa3d/Prusa-Firmware#331 and this ticket is actually a duplicate of #2285. Closing this since this most likely the case here, if not please provide your terminal output or better yet a `serial.log` (see the contribution guidelines on where to find and how to enable that) and request a reopen.

closed 2018-01-11 by foosel

foosel, 2018-10-22:
See [this FAQ entry](https://discourse.octoprint.org/t/printer-specific-original-prusa/239).

foosel, 2018-10-22:
Please see [this FAQ entry](https://discourse.octoprint.org/t/printer-specific-original-prusa/239).

nkoeppel, 2018-10-22:
So no? Pause is just incompatible with the current firmware?

JohnOCFII, 2018-11-22:
> 3.4 MK3 definitely has this issue.

Not sure how current this list is, but it implies that M601/M602 for pause/resume on the Prusa firmware is only supported when printing from SD card, not over USB as from OctoPrint.

https://github.com/prusa3d/Prusa-Firmware/wiki/Supported-G-codes

forestqc69, 2019-01-21:
> So, what about M76? I am trying to use that to pause a print being sent from OctoPi to an MK3-MMU2 but it did not pause.

Same thing for me, but that code doesn't seem to be valid for the Prusa MK3..