OctoPrint/OctoPrint #2309: AnyCubic i3 Mega - non standard responses

gregopti, 2017-12-21:
I've recently received custom firmware from AnyCubic with ability to connect it with Octoprint. There was some baudrate and serial driver issue with original one. See: **[https://github.com/guysoft/OctoPi/issues/452](url)**

When trying to print I've faced another issue: non standard responses M190 and "OK" confirmations.
That caused OctoPrint to send new requests even if last one was still not finnished. At the end that meant printer started to print without heating the extruder and going to home position - those was skipped. Printer was sending wait responses but Octoprint kept sending new requests.

Example (Octoprint - Anycubic i3 Mega):

`Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok N:0
Send: N1 G90*17
Recv: //####checkSum error expect:0x7e,actual:0x7d!str:N0*125
Recv: 
Recv: ok
Send: N2 M82*27
Recv: //####checkSum error expect:0x0,actual:0x7d!str:*125
Recv: 
Recv: ok N:1
Send: N3 M106 S0*100
Recv: ok N:2
Send: N4 M140 S60*87
Recv: ok N:3
Send: N5 M190 S60*91
Recv: ok N:4
Send: N6 M104 S200 T0*37
Recv: wait
Send: N7 M109 S200 T0*41
Recv: T:27 E:0 B:33
Recv: T:27 E:0 B:34
Recv: wait
Send: N8 G28 X0 Y0*26
Recv: T:27 E:0 B:34
Recv: wait
Send: N9 G28 Z0*80
Recv: T:27 E:0 B:34
Recv: T:27 E:0 B:35
Recv: wait
Send: N10 M107*20
Recv: T:27 E:0 B:35
Recv: T:27 E:0 B:36
Recv: wait
Send: N11 M105*23
Recv: T:27 E:0 B:36
Recv: wait
Send: N12 G5*31
Recv: T:27 E:0 B:37
Recv: T:27 E:0 B:37
Recv: wait
Send: N13 G92 E0*117
Recv: T:27 E:0 B:37
Recv: T:27 E:0 B:38
Recv: wait
Send: N14 M105*18
Recv: T:27 E:0 B:38
Recv: wait
Send: N15 G92 E0*115
Recv: T:27 E:0 B:38
Recv: T:27 E:0 B:38
Recv: wait
Send: N16 M105*16
Recv: T:27 E:0 B:39
Recv: T:27 E:0 B:39
Recv: wait
Send: N17 G1 X0.000 Y0.000 Z0.000 E-5.5000 F3000*120
Recv: T:27 E:0 B:39
Recv: wait
Send: N18 G1 X0.000 Y0.000 Z0.200 F1002*35
Recv: T:27 E:0 B:39
Recv: T:27 E:0 B:40
Recv: wait
[... 382 lines cut ...]
Send: N2 M21*18
Recv: //####checkSum error expect:0x7e,actual:0x7d!str:N0*125
Recv: 
Recv: ok
Recv: //####checkSum error expect:0x0,actual:0x7d!str:*125
Recv: 
Recv: ok ZWLF make it.Date:Dec 19 2017 Time:20:47:53
Recv: //SD init fail!
Recv: ok N:2
Send: N3 M105*36
Recv: ok T:27 /0 B:47 /0 @:0 B@:0
Send: N4 M105*35
Recv: ok T:27 /0 B:47 /0 @:0 B@:0
Send: N5 M105*34
Recv: ok T:27 /0 B:47 /0 @:0 B@:0
Send: N6 M105*33
Recv: ok T:27 /0 B:47 /0 @:0 B@:0
Send: N7 M105*32
Recv: ok T:27 /0 B:47 /0 @:0 B@:0
Send: N8 M105*47
Recv: ok T:27 /0 B:47 /0 @:0 B@:0
Send: N9 M105*46
Recv: ok T:27 /0 B:46 /0 @:0 B@:0`

Is there a way how to set the Octoprint to communicate with the printer without using some other firmwares.

crckmc, 2017-12-31:
I have the same problem with the latest OctoPrint (Version 1.3.6) on Windows is there any known workaround?

electron23, 2018-01-02:
I have also the problem. Is there already a fix?

gregopti, 2018-01-04:
I guess I can share it here, since it is the same firmware, with only difference that it works on 115.2kbps.
Unzip file, Copy the modified firmware (update.zw) to SD card , plug it into the printer and then reboot the printer, wait for the firmware update is complete. Remove file from SD cards afterwards.
[update.zip](https://github.com/foosel/OctoPrint/files/1602873/update.zip)

foosel, 2018-01-11:
This actually rather looks like the firmware is sending out an extra `ok` here that confuses things:

```
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok N:0                         <- This obviously belongs to M110 (N matches)
Send: N1 G90*17
Recv: //####checkSum error expect:0x7e,actual:0x7d!str:N0*125
Recv:
Recv: ok                             <- Where does that come from? It can't belong to the G90...
Send: N2 M82*27
Recv: //####checkSum error expect:0x0,actual:0x7d!str:*125
Recv:
Recv: ok N:1                         <- ... because THAT seems to belong to G90 (N matches)
Send: N3 M106 S0*100
Recv: ok N:2                         <- M82
Send: N4 M140 S60*87
Recv: ok N:3                         <- M106
Send: N5 M190 S60*91
Recv: ok N:4                         <- M140
Send: N6 M104 S200 T0*37
[...]
Recv: ok N:5                         <- M190
```

So the question is, why does the firmware send an extra `ok` here? It shouldn't, and doing so causes the whole ping-pong-sending-protocol to collapse here. And also: What is going on with its parsing of that `M110` line there? It appears to think that the parameter `N0` [1] is its own command and then really hiccups hard when trying to parse that.

What could be is that this wonky parsing issue here triggers this extra `ok`. To check, you could try placing [this small plugin](https://gist.github.com/foosel/924a6518b2dea77c9a6ff1f2561b8bb7) in `~/.octoprint/plugins`. On OctoPi just SSH into your Pi, download the file to the plugins folder and restart OctoPrint:

    cd ~/.octoprint/plugins
    wget https://gist.githubusercontent.com/foosel/924a6518b2dea77c9a6ff1f2561b8bb7/raw/46dcd2cd651817683746772907917b72d0c201a5/strip_m110_parameter.py
    sudo service octoprint restart

Then test. What it does is it rewrites any `M110` commands to be parameterless, so `N0 M110 N0*125` becomes `N0 M110*35` instead.

[1] OctoPrint sends an `N` parameter here because that is what some firmware require to function. Others require `M110` to be sent with the line number its supposed to reset to. To be compatible to both without users having to endlessly toggle checkboxes, OctoPrint simply sends both and so far no firmware that I encountered had a problem with that... Anycubic's the first there, hooray!

PS: I really wish printer vendors would just stop messing with firmware if they don't fully understand the protocol in place and just use what's already available instead of cooking up their own variants... Trying to keep things working in this environment of vendors just doing what they feel like and breaking stuff left and right in the process for end users is making this job extraordinarily frustrating to a point where I sometimes just want to scream.

gregopti, 2018-01-11:
The plugin unfortunately didn't help.
I think the main problem is that when printer sends "wait", Octoprint sends new request instead of waiting for the request to complete.

I'm sending complete serial.log from the moment of connecting, then starting printing a cube, canceling due to unheated extruder and skipped commands (heating extruder, home position, intial script and begining of print) and finally disconnecting the printer.
https://pastebin.com/FiF71gWG

I think the solution could be, that whenever printer sends "wait" it doesn't sends anything new till the printer sends something else than "wait".

foosel, 2018-01-11:
What OctoPrint does is correct, Anycubic is just misusing the `wait` here. It should only be sent if the printer hasn't received a command in a certain timeout and its planner buffer is dry. Instead here it is being abused as something similar to the busy protocol.

Another idea, [new plugin](https://gist.github.com/foosel/77e4792921db744a157764f66d1d72a5). Installation:

```
cd ~/.octoprint/plugins
wget https://gist.githubusercontent.com/foosel/77e4792921db744a157764f66d1d72a5/raw/89995d42b2cae5a0f403bea49d161e7959f9f3ce/rewrite_wait_to_busy.py
sudo service octoprint restart
```

Don't uninstall the other one, you need both.

As a side note:

```
2018-01-11 12:02:09,278 - Send: N1 M115*39
2018-01-11 12:02:09,286 - Recv: ok ZWLF make it.Date:Dec 19 2017 Time:20:47:53
```

Nope... Just nope. That is *not*  a well formatted response to an `M115`. Also:

```
2018-01-11 12:04:44,385 - Send: N103 G1 X110.754 Y95.480 E0.5554*103
2018-01-11 12:04:44,387 - Recv: ok N:101
2018-01-11 12:04:44,389 - Send: N104 G1 X114.520 Y99.246 E0.7148*110
2018-01-11 12:04:44,434 - Recv: ok N:102
2018-01-11 12:04:44,436 - Send: N105 G1 X114.520 Y99.812 E0.7318*99
2018-01-11 12:04:44,437 - Recv: Resend:103
2018-01-11 12:04:44,438 - Recv: ok
2018-01-11 12:04:44,439 - Send: N103 G1 X110.754 Y95.480 E0.5554*103
2018-01-11 12:04:44,442 - Recv: //last:102 curr:104 rcv:N104 G1 X114.520 Y99.246 E0.7148*110
2018-01-11 12:04:44,442 - Recv:
2018-01-11 12:04:44,443 - Recv: 3
2018-01-11 12:04:44,445 - Recv: Resend:103
2018-01-11 12:04:44,446 - Recv: ok
2018-01-11 12:04:44,447 - Send: N103 G1 X110.754 Y95.480 E0.5554*103
2018-01-11 12:04:44,451 - Recv: //last:102 curr:105 rcv:N105 G1 X114.520 Y99.812 E0.7318*99
2018-01-11 12:04:44,454 - Recv: N103 G1 X110.754 Y95.480 E0.5554*103
2018-01-11 12:04:44,455 - Recv:
2018-01-11 12:04:44,455 - Recv: 3
2018-01-11 12:04:44,461 - Recv: ok N:103
```

Resend request lacking the `Error:` line that indicates what the problem even was (which in some scenarios sadly is needed) 😕

gregopti, 2018-01-11:
I've tried the new plugin/hook and result was it couldn't connect to printer.
https://pastebin.com/23TfUvPH
Leter I went to change the hook and reboot OctoPrint. (I added "line" after "return")
`def rewrite_wait_to_busy(comm_instance, line, *args, **kwargs):
    if line != "wait":
        return line
    return "echo:busy processing"
`
I tried it first time. The result was, that printer did everything as it should and printed a cube although with too low first layer. (note for later: printer was already in home position) Unfortunately I didn't save the serial log. 

So I tried to print it again... No success this time. Printer started to print in middle of air with cold extruder.
Try #2: https://pastebin.com/Y5yE0xpw 
(Serial error on the end because I turned the printer off.)

In third try, nothing else changed... Printer heated the extruder, but didn't go to home position and started printing from the point where extruder was.
Try #3: https://pastebin.com/rrZGYKdE
(Serial error on the end because I turned the printer off.)

Forth try was in principle same as second one, unheated extruder, mid-air printing.
Try #4: https://pastebin.com/HQVZAKbp
(Serial error on the end because I turned the printer off.)

For some reason it is skipping some commands and doesn't send "Resend:" when it should...

foosel, 2018-01-11:
First of all sorry for the missing return parameter... That was a copy paste error from the previous plugin 😳

But from the logs it looks like it's not actually doing anything anymore starting with your second try. Let's try adding some logging, just to make sure it's actually running into that code. I've updated the [gist](https://gist.github.com/foosel/77e4792921db744a157764f66d1d72a5) (including your fix), install should work with `wget https://gist.githubusercontent.com/foosel/77e4792921db744a157764f66d1d72a5/raw/0e23f2867971cc9b2347dcaa269fd0e31572753c/rewrite_wait_to_busy.py` at the usual location (you might have to delete the old version first via `rm rewrite_wait_to_busy.py`, I'm not sure right now if `wget` will overwrite existing files by default or not).

SchizoDuckie, 2018-01-11:
> githubusercontent.com/foosel/77e4792921db744a157764f66d1d72a5/raw/
> 0e23f2867971cc9b2347dcaa269fd0e31572753c/rewrite_wait_to_busy.py at the

That looks like a brilliant solution!

I have one of the upgraded models myself. I'll try to verify if the
'updated version' has the same behavior as soon as my pi arrives.

gregopti, 2018-01-12:
Now this seems to be working :)
Octoprint now waits and doesn't sends new requests after "wait" signals, also you can see in octoprint.log that it is replacing the wait signals with echo:busy processing. (I have later disabled this logging) And it prints quiet well.
https://pastebin.com/8JsAuLyf

But after succesful print I have noticed that ending script was not as set, and I got this errors, repeating on terminal:
https://pastebin.com/VGT51Pz9

foosel, 2018-01-12:
```
Recv: Resend:7034
Recv: ok
Send: N7034 G28 X*91
Recv: //last:7033 curr:7035 rcv:N7035 M84*46
Recv:
Recv: 3
Recv: //sd cmd error cnt:16175 len:15 :N7034 G28 X*91
Recv:
Recv: ok
Send: N7035 M84*46
Recv: Resend:7034
Recv: ok
Send: N7034 G28 X*91
Recv: //last:7033 curr:7035 rcv:N7035 M84*46
Recv:
Recv: 3
Recv: //sd cmd error cnt:16176 len:15 :N7034 G28 X*91
Recv:
Recv: ok
```

This is funny :D

> **Printer**: Hey, please resend line 7034!
> 
> **OctoPrint**: Sure, here you go! *sends line 7034*
> 
> **Printer**: Ok, great! Ready for the next line!
> 
> **OctoPrint**: Perfect, here you go! *sends line 7035*
> 
> **Printer**: Errr.... eek... abort, abort... Hey, please resend line 7034!
> 
> **OctoPrint**: Sure, here you go! *sends line 7034*
> 
> **Printer**: Ok, great! Ready for the next line!
> 
> [and they lived angrily shouting at each other ever after]

So... I have no idea what triggers this merry go around, so I don't know if might be able to fix this by yet another workaround plugin (to be then merged into one I guess because it's starting to become a bit much). What I'd need is a full `serial.log` from a whole print, including the end of it and the start of the loop.

Also, please share your start and end script and the printed GCODE file so all of it can be matched up against what we'll be seeing in the `serial.log`.

I can't promise though that we'll get it to work fully... this firmware is really messed up, and I'd strongly suggest to write an angry letter or two to the manufacturer. What they are doing there roughly resembles the regular GCODE based communications on first glance, but behaves nothing like it or outright goes against it in some places (e.g. the `wait` usage).

crckmc, 2018-01-12:
I have (had) the same problen and am very grateful for your patches 👍 
I wrote the Anycubic support and send them here.
They say they are working on a fix in firmware:
"Hello:
Thanks very much for your email and support.
Our software engineer will update this feature, and also will give the results as soon as possible.
Thanks again for your suggesttion and support."

Lets hope they fix their software as the hardware is really great for the price.

gregopti, 2018-01-12:
Ok, so here we go...
I printed a simple cube and everything went fine except the ending is where things go wrong way.
https://pastebin.com/14wTqYJx
(I had to cut repating error sections otherwise the file is too big, since this error goes up several times in one second - only thing what changes it the "error cnt:" number which keeps increasing ) 

Here is also complete gcode file:
https://pastebin.com/vihfWvUZ

The `G28 X` if where it seems to go wrong. Extruder in reality moves towards X home position but never reaches end switch, unless I've missed it, but it stays 1cm from it.

foosel, 2018-01-12:
@crckmc I'll keep my fingers crossed!

@gregopti Is this `G28 X` part of your slicer's end code? I'd try a variant of this... `G28 X0` instead of `G28 X`. From my experience the correct format of a `G28` with axes limitation depends a bit on the firmware variant - and depending on who you ask, either `X` or `X0` is the one and only true form ;)

It's tough to say since it doesn't look like anything regular firmware produces, but maybe this `sd cmd error` debug output:

```

2018-01-12 11:56:05,767 - Recv: //sd cmd error cnt:0 len:15 :N2021 G28 X*90
2018-01-12 11:56:05,769 - Recv:
2018-01-12 11:56:05,770 - Recv: ok
```

is supposed to mean something like "Could not understand this command, seems to be malformatted, please fix and send again" and this then causes the loop?

I'd rather have expected an `Error: Unknown command` here or something like that. Beats me though why it doesn't immediately request a resend then but instead first sends a normal `ok`, then asks for the same line that just caused an issue again.

gregopti, 2018-01-16:
Ok, problem solved. Thanks for advice.
`G28 X0` is the right command.

`Send: N7395 G92 E0*127
Recv: ok N:7395
Send: N7396 M105*28
Recv: ok T:198 /200 B:60 /60 @:99 B@:71
Send: N7397 G1 E-5.5000 F3000*33
Recv: ok N:7397
Send: N7398 G1 X0 Y195 F2400*113
Recv: ok N:7398
Send: N7399 M106 S0*83
Recv: ok N:7399
Send: N7400 M104 S0*86
Recv: ok N:7400
Send: N7401 M140 S0*87
Recv: ok N:7401
**Send: N7402 G28 X0*106**
Recv: wait
Recv: wait
Recv: wait
Recv: ok N:7402
Send: N7403 M84*47
Recv: ok N:7403
Send: N7404 M107*18
Recv: ok N:7404
Send: N7405 M300 S1500 P100*32
Recv: ok N:7405
Send: N7406 G4 P100*89
Recv: ok N:7406
Send: N7407 M300 S1500 P100*34
Recv: ok N:7407
Send: N7408 G4 P100*87
Recv: ok N:7408
Send: N7409 M300 S2500 P100*47
Recv: ok N:7409
Send: N7410 G4 P100*94
Recv: ok N:7410
Send: N7411 M300 S2500 P1000*22
Recv: ok N:7411
Send: N7412 G4 P500*88
Recv: ok N:7412
Send: N7413 M300 S2500 P1000*20
Recv: ok N:7413
Changing monitoring state from 'Printing' to 'Operational'`

Commander112, 2018-01-16:
*Thumb up*
Same Issues with same Printer (Firmware+Hardware)...
So I updated the Firmware of the Printer like @gregopti do, install the rewrite_wait_to_busy-Script, make a few changes on my Start/Ending-GCODE and now all is running perfect!!

Thanks to @foosel aka. Gina for her excelent support!
I think I can do a little support on patreon so you can help us next time again ;-)

electron23, 2018-01-16:
Can you explain the exact steps please.

coconuthero, 2018-01-17:
I also have the same problem with my new Ultrabase I3 Mega with firmware: V1.4.ORC23/P3
I yesterday wrote to anycubic about it and also referred to this place and their reply today was:

_Sorry the printer does not support octoprint Raspberry pi , pls use cura if you need to print on line_

Doesn't look like they are aware what octoprint is, or working on a firmware update... :-(  
I would definately be interessted to get a noob-friendly fix for this issue incl donation!

foosel, 2018-01-18:
@coconuthero since @gregopti *did* get a firmware from them that is capable of talking to hosts like OctoPrint, I'd say things don't look that grim. However, I have no idea if that firmware is compatible to your model. I have no idea of the Anycubic lineup of printers (and frankly after seeing bits of their self tinkered firmware I don't want to get too personal with that stuff either).

As a **summary on how to get things working** once there's a firmware flashed that *does* allow an initial connection to succeed, here's what to do to get it behaving (instructions for OctoPi *via SSH*, adapt accordingly if you installed from source):

``` 
cd ~/.octoprint/plugins
wget https://gist.githubusercontent.com/foosel/924a6518b2dea77c9a6ff1f2561b8bb7/raw/46dcd2cd651817683746772907917b72d0c201a5/strip_m110_parameter.py
wget https://gist.githubusercontent.com/foosel/77e4792921db744a157764f66d1d72a5/raw/8593ddb36e8d41a67ad9ccc01c36bb2863f5e7c2/rewrite_wait_to_busy.py
sudo service octoprint restart
``` 

This installs both plugins written over the course of this thread here that change the non-standard replies and behaviours of the firmware in such a way that communication can succeed (as long as your GCODE files don't contain stuff the firmware doesn't like in which case it starts a resend loop instead of sending an invalid command error).

It probably makes sense to merge these two plugins into one and publish that on the repository. I'll try to get around to do that as soon as I can and then close this ticket with a link to it.

thetobby, 2018-01-19:
@gregopti  Hi, Do you have the v1.1 or v1.4 firmware original running on your Anycubic i3 Mega Ultrabase? The model is shipped in two versions with v1.4 firmware running a 32bit instead of 8bit CPU, so patching the wrong version might brick.  @coconuthero I would recommend not trying, but if you did. Let me know if it worked since I know you got v1.4

electron23, 2018-01-19:
I've flashed the patch and I've the firmware 1.4 and all works well.

thetobby, 2018-01-21:
Flashed the patch, my Anycubic went from v1.4 RC1 to v1.4RC2 and it now works great with octopi with the plugins provided by foosel.

gregopti, 2018-01-22:
Thanks @foosel for your support and great cooperation!

For all those who are asking what is this patched version of firmware for, I can corfirm words of @thetobby.

The original firmware that I had was V1.4.1RC1/P5 which is then after uploading patched firmware addressed as V1.4.1RC2/P5

I cannot guarantee that this version will work with other firmware/software combination.

foosel, 2018-01-22:
Any official sources for the 1.4.1RC2/P5 firmware that I could link to from a plugin page? Or is it as of yet unreleased and the link in this thread is the only source so far?

gregopti, 2018-01-22:
I think they haven't make it official and we only got what they have sent me via email and I posted here.
The only change in firmware is actually in baudrate which was incompatible with RaspberryPi PL2303 drivers. Instead of 250kbit the baudrate is now 115,2kbps.

Therefore if Octoprint is running on some other OS or if we got different debian/linux drivers for PL2303, I'm almost sure this patched firmware won't be needed and original could be used instead.

thetobby, 2018-01-22:
@gregopti  Since you already have a dialog with Anycubic, can you poke them for some more information related to this? Perhaps they can release the firmware publicly or have some more information/plans

foosel, 2018-01-22:
@gregopti I second this suggestion by @thetobby. Additionally - if you think that would help - you can also bring us into contact and I'll try to poke them a bit as well.

coconuthero, 2018-01-27:
And after I had setup my raspberry again because I forgot my password it finally also works for me! 
Thank you very much foosel!

thetobby, 2018-01-30:
This is the source code for old 1.1 hardware not 1.4hardware, but might help? @foosel  
https://github.com/ANYCUBIC-3D/I3-MEGA/tree/master/Marlin_TFT_CR_V1.1.1

foosel, 2018-01-30:
@thetobby on first look more confusing than helping sadly. They've completely changed the serial communication, thrown out the gcode parser and replaced it with some custom "a codes". Code is riddled with commented out blocks, very hard to read, but it doesn't look like this firmware in this form is related to what we saw in use here (with this one OctoPrint wouldn't have been able to connect at all, let alone print). 

Word of caution though just in case anyone is using that firmware: disabled thermal runaway protection according to the bundled config, that is highly disturbing.

Angeal87, 2018-02-04:
Hello,
my Anycubic i3 Mega runs Firmware 1.4.0RC23/P3
I use OctoPi 0.14 / OctoPrint 1.3.6 with installed and activated patches strip_m110_parameter.py and rewrite_wait_to_busy.py
When I start to connect the OctoPi with my i3 Mega the serial.log tells me the following:

`2018-02-04 10:07:45,743 - Connecting to: /dev/ttyUSB0
2018-02-04 10:07:45,894 - Changing monitoring state from 'Offline' to 'Opening serial port'
2018-02-04 10:07:45,910 - Connected to: Serial<id=0x7449f950, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2018-02-04 10:07:45,914 - Changing monitoring state from 'Opening serial port' to 'Connecting'
2018-02-04 10:07:45,936 - Send: N0 M110*35
2018-02-04 10:07:55,955 - Recv: �������>2���2�����>���>��22������0���< �<���� ���>�� ����>>>2����`

Any idea so far?

electron23, 2018-02-04:
Set the baud to 115000

Angeal87, 2018-02-04:
Nearly the same problem

> 2018-02-04 10:47:33,561 - Connecting to: /dev/ttyUSB0
> 2018-02-04 10:47:33,731 - Changing monitoring state from 'Offline' to 'Opening serial port'
> 2018-02-04 10:47:33,738 - Connected to: Serial<id=0x6f463b50, open=True>(port='/dev/ttyUSB0', baudrate=115000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
> 2018-02-04 10:47:33,739 - Changing monitoring state from 'Opening serial port' to 'Connecting'
> 2018-02-04 10:47:33,747 - Send: N0 M110*35
> 2018-02-04 10:47:43,766 - Recv: ��#��#��+#��/7�+��4��#������#4�#��4�#���,�$�;3#�����

Commander112, 2018-02-04:
@Angeal87 
You ´ve to update your Printerfirmware with the Update above (Comment from @gregopti)!! You need V1.4.1.RC2/P3

Angeal87, 2018-02-04:
No idea how to get the firmware and how to flash it...

thetobby, 2018-02-04:
@Angeal87  Just put the file on the root directory of the SD card and reboot the printer. When it is done, delete the file.

Angeal87, 2018-02-04:
Thanks for the info how to flash. But I have no chance to find the 1.4.1 on the internet. Googled about an hour...

Angeal87, 2018-02-04:
Thanks for the info how to flash. Works great :-)

electron23, 2018-02-05:
When you read the hole thread you will get all answers!!!!

Download the update.zip in this thread (gregopti is the uploaded), extract it, put the update.zw in the root directory of your SD, shut down your printer, put the SD in the slot in your printer, start your printer and wait for the flash process.

pauledd, 2018-02-06:
Hi, just got my I3 Mega with firmware: V1.4.0RC24/P3. Its recognized by my linux host as:

```
[  490.408116] usb 1-1: Manufacturer: Prolific Technology Inc.
[  490.408590] pl2303 1-1:1.0: pl2303 converter detected
[  490.409292] usb 1-1: pl2303 converter now attached to ttyUSB0
```
Octoprint cannot detect its baudrate.
Can I use the firmware by @gregopti  on my printer?

electron23, 2018-02-06:
@pauledd Yes, you can

Angeal87, 2018-02-06:
I've printed Benchy two times.
Once via SD card and once via OctoPrint.
For both I took the same GCode file (Sliced on the PC with Cura 15.04.6)
I have the impression that the print quality on OctoPrint is not as good as from SD card

Any idea somebody?

pauledd, 2018-02-08:
@electron23, thanks, firmware update succeeded. Do I need to set something else beside the common settings for the anycubic i3 mega? Any special commands?

pauledd, 2018-02-09:
Okay, my first Test went ... scary. I set the bed dimensions to 210,210,205. Thats all. Loaded model, pressed print,the printer startet heating up the bed. As it reached it (hotend was still unheated) the motors started to move the bed to the middle upper edge, it seemed to start printing half on the bed, half in the air and there was a horrific rattle noise, I guess from that motor that feeds the filament, because the hotend was not heated it of cause resists to move...
Here is the log, I dont understand it but maybe it is helpfull...
https://pastebin.com/MpaERidx

Iam quite sure I will NOT try octoprint again until it is absolutely clear why this happened...

Commander112, 2018-02-09:
@pauledd 
Can you please poste your used Gcode-File to Pastebin?
The "Logfile" unfortunately starts at N668.

- Plugins (listed above) installed on your Raspi?
- Any Start/Stop GCode-Scripts used?

pauledd, 2018-02-10:
@Commander112 
Sorry, my fault! I'll give it a second try with plugins this time... Do I need any start/stop gcode scripts?

Commander112, 2018-02-10:
@pauledd
The two Plugins are needed! Anycubic uses the Commands on her own way ;-) 
For better Workflow you need GCode-Script... 
Put the following part in your Octoprint GCode-Script for start printing:
G28 X0 Y0
G28 Z0
G1 Z15 F500
M107 ; fan off
G5 ; enable resume from power failure
G92 E0 ; zero E axis
G1 E3 F200 ; prime

And that for after printer:
G1 X0 Y195 F2400 ; position part removal
M106 S0 ; fan off
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
G28 X0 ; home x
M84 ; disable motors
M107 ; turn off print cooling fan
M300 S1500 P100
G4 P100
M300 S1500 P100
G4 P100
M300 S2500 P100
G4 P100
M300 S2500 P1000
G4 P500
M300 S2500 P1000

Behind the Commands (after ;) are Comments for better knowing the usage of the line.

pauledd, 2018-02-10:
Thanks for the clarification! So I put it in here?:
[image]

Commander112, 2018-02-10:
Yes!
Start-Script at "Before print job starts" and the End-Script at "After print job completed"

With the two pyton-Plugins and these Codes my Printer is perfekt ;-)

pauledd, 2018-02-10:
Thanks! It works so far, but I am a bit concerned how close to the bed the hotend moves back after the print has finished. Is this normal? I printed this bed level test with one layer rectangles.
https://youtu.be/t95wq5-5vLo

pauledd, 2018-02-11:
What I also noticed is that if I cancel a print the printer immediately stops printing where it currentlyis is, but stays there, it does not move away from the hotbed nor goes home... I dont know if this is intended.

Commander112, 2018-02-11:
Sorry, at the moment back from a 24h-Shift.

Both is a GCODE-Script Thing. The Code I poste is very small and for Funktion, if you want more you can change the code with Functions you want (Beep, Liftup, Movement).

For Cancel I use the same GCODE-Script I use at End-Script.

pauledd, 2018-02-11:
Okay thanks, I will change that. So far octoprint works beautifully with the i3 Mega! :)

thetobby, 2018-02-15:
I noticed that using octoprint with the above setups it sometimes sound like it is scraping when printing on X/Y, but with the same Gcode it is fine if run directly from SD card. Anyone else with similar experience or any ideas?

jjort, 2018-02-27:
I cant seem to find the plugins directory when I SSH into octoprint as directed in the instructions
cd ~/.octoprint/plugins
Is there a way to install this plugin through the web interface?

tifrei, 2018-03-10:
Hi, i have also the firmware : V1.4.0RC24/P3 and my octopi wouldnt connect. i already downloaded the firmware mentioned above but how can i install it? i just need to put it on the sd card and than ?

Thanks for any help and sorry for asking that stupid qeustion.

pauledd, 2018-03-11:
@tifrei afair I put the firmware on the root directory of the SDCard and then power on the printer with it. It should start the FW Update.

kamaleonte, 2018-03-11:
If I update with the gregopti firmware, it disabled thermal runaway protection ?

acdapkus, 2018-03-22:
Printer: Xinkebot Orca2

I was having the same issue as the OP, however it is now fixed. Thank you.

My solution:
I applied the 2 fixes linked above, and was left with one error, also listed above. This error was "Recv: //sd cmd error cnt:16175 len:15 :N7034 G28 X*91" in nature.

To fix this final issue, I also had to turn off "Auto Detect Firmware" and check the "Always assume SD card is present" option. After I did this, the errors disappeared.

If anyone runs into an issue and thinks I can help, let me know. I would be happy to post any logs or settings if requested.

apos, 2018-03-31:
https://github.com/foosel/OctoPrint/issues/2309

Hey,

I applied the alterings to octoprint (Version 1.3.6) and the server to my anycubic i3 mega:

* original firmware V1.4.1RC1/P1
* on my pc connected to the printer installed the above two (!) plugins ( " foosel commented on 18 Jan ") into ~/.octoprint/plugins
* Installed the GCode-Scripts from above into octoprint ("Commander112 commented on 10 Feb2)
* upgraded the firmware with update.zip/update.zw from  "@gregopti commented on 4 Jan"  (extract the update.zw into the sdcard root and shutdown/shut-of the printer. while the update is running the printer beeps continuisly - don't interrupt) - now I have  Firmware V1.4.1RC2/P2

But: the printer does not start printing at all. I always get the same error messages - the printerhead does not move at all:

`Send: N8 G92 E0*79
`Recv: Resend:7
`Recv: ok
`Send: N7 G1 Z15.0 F{speed_travel}*15
`Recv: //last:6 curr:8 rcv:N8 G92 E0*79
`Recv: Recv: 3
`Recv: //sd cmd error cnt:196139 len:31 :N7 G1 Z15.0 F{speed_travel}*15
`Recv:
`Recv: ok
`Send: N8 G92 E0*79
`Recv: Resend:7
`Recv: ok
`Send: N7 G1 Z15.0 F{speed_travel}*15
`Recv: //last:6 curr:8 rcv:N8 G92 E0*79

Cheers
Axel

[16 more replies]

closed 2020-03-25 by foosel