OctoPrint/OctoPrint #2310: octoprint not able to continue after Prusa MK3 layer shift recovery

cnlson, 2017-12-21:
I had a crash while printing which caused a layer shift and the prusa recovered, came back to the position and then just stopped. 

in looking at the log it appears that the rehoming and returning to location caused octoprint to think it had reset instead of resending last command and continuing

Recv: start
Printer sent 'start' while printing. External reset? Aborting job since printer lost state.
Changing monitoring state from 'Printing' to 'Operational'
Recv: echo:echo: Last Updated: Dec 15 2017 18:45:44 | Author: (none, default config)
[octoprint_crash_recovery.txt](https://github.com/foosel/OctoPrint/files/1580040/octoprint_crash_recovery.txt)

Recv: Compiled: Dec 15 2017
Recv: echo: Free Memory: 1549  PlannerBufferBytes: 1392
Recv: echo:Hardcoded Default Settings Loaded
Recv: crashdet_enable
Recv: CrashDetect ENABLED!
Recv: PAT9125_init:1

JagerSprinkles, 2017-12-22:
Maybe related, this also happens when the printer detects that the filament has run out. Octoprint disconnects as if an error had occurred.

JagerSprinkles, 2017-12-30:
This issue for the layer shift recovery and the issue with the filament run out detection I mentioned has been fixed in the latest MK3 firmware (rc4_b143).

foosel, 2018-01-11:
```
N14 M900 K{if printer_preset=="Original Prusa i3 MK2 MultiMaterial"}200{elsif printer_preset=="Original Prusa i3 MK2 MM Single Mode"}200{else}45{endif}*11
```

This templating language there does not look like something that should be in a printed GCODE file, but rather should have been replaced by the slicer. So I'd definitely look into why that even made it into a file to be sent to the printer in the first place, wrong slicer version?

Then the firmware's resend requests appear to be running into the same issue as described in #2285 - they are missing an `ok`, causing communication issues because OctoPrint doesn't know it's allowed to send more data to the printer now.

> This issue for the layer shift recovery and the issue with the filament run out detection I mentioned has been fixed in the latest MK3 firmware (rc4_b143).

Does that mean this can be closed? I'm unclear on what besides the already tracked #2285 (for which I btw also opened a ticket in Prusa's tracker as prusa3d/Prusa-Firmware#331) and the not replaced slicer left overs should be looked into here.

I have to say though that I'm not sure that any further debugging from OctoPrint's side makes a ton of sense until the general serial issues with the Prusa 3.1 firmware that are currently reported all over the place here are solved definitely on their. Otherwise it's just trying to shoot at a moving target.

foosel, 2018-01-11:
Alright, closing then. Thanks!

closed 2018-01-11 by foosel