OctoPrint/OctoPrint #2311: Server not autoconnecting on startup

ismangil, 2017-12-21:
#### What were you doing?

Ticked auto-connect on server startup

#### What did you expect to happen?

Server will connect to printer when server starts or restarted

#### What happened instead?

Server did not try to connect on startup 

#### Did the same happen when running OctoPrint in safe mode?

Yes. 

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Marlin 1.1.4 GitHub

#### Browser and Version of Browser, Operating System running Browser

Any browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

4325123, 2017-12-26:
Just saying I have the same issue too.  I'm wondering if it has something to do with octoprint starting before the printer is being detected.

kantlivelong, 2017-12-27:
@ismangil can you provide fresh logs for both **octoprint.log** and **serial.log** with the issue occurring?

ismangil, 2017-12-28:
@kantlivelong Here it is: https://gist.github.com/ismangil/cc99d24ee982a04a2aba946609bf2cdf

Note the serial.log is only generated once I manually click ```Connect```, because the server is not auto-connecting.

kantlivelong, 2017-12-28:
Yeah not much of use there as far as I can tell. If you login to SSH and run `cat /var/log/syslog | grep kernel | grep -A5 -B5 "tty"` you should see when the serial device was detected by the kernel. Does it show up before OctoPrint starts?

ismangil, 2017-12-28:
@kantlivelong you may have something there. If I I'm looking at the log correctly, OctoPrint may be starting up before /dev/ttyUSB0 become available. I suppose in this case it's an OctoPi configuration issue on how to wait a bit or detect ttyUSB0.

kantlivelong, 2017-12-28:
The [PortLister](https://plugins.octoprint.org/plugins/portlister) plugin should work as a workaround. I personally use it as it also will reconnect after resetting the printer controller.

ismangil, 2017-12-28:
@kantlivelong however, that doesn't explain why it doesn't connect on server restart (not a reboot) doesn't it? The port is already there, yet OctoPrint is still not autoconnecting.

kantlivelong, 2017-12-28:
Hmm...

What's the make and model of printer/controller are you using?

ismangil, 2017-12-28:
MKS GEN L 1.0

ismangil, 2018-01-01:
I've found the cause: If port is left on `AUTO` then

`if port in connectionOptions["ports"]:`

fails, because `AUTO` is not on the list of ports.

Yet when I click connect, it's able to connect. So the solution seems the autoconnect logic needs to follow the manual button logic.

For my current use, if I specify the specific port, then it can autoconnect just fine.

I've updated the issue title.

kantlivelong, 2018-01-03:
Well that's interesting... I've confirmed your findings here as well.

kantlivelong, 2018-01-03:
Hi @ismangi,

I've submitted a PR that fixes this.

jneilliii, 2018-01-03:
Does this also effect the Virtual Printer port?  I can never seem to get that one to autoconnect when testing my plugins.

kantlivelong, 2018-01-03:
@jneilliii only if you had the serial port set to AUTO.

jneilliii, 2018-04-08:
I just tested against 1.3.7rc4 and the behavior is still the same for me with VIRTUAL serial port, and the option selected as AUTO.  This is the output in the terminal.

Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['VIRTUAL']
Connecting to: VIRTUAL
Changing monitoring state from "Detecting serial port" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: AUTO OSError: '[Errno 2] No such file or directory: 'VIRTUAL'' @ comm.py:_openSerial:2304 (hook default)

foosel, 2018-04-09:
`VIRTUAL` doesn't and has never supported auto detection, so this is expected behaviour.

foosel, 2018-04-09:
Closing as 1.3.7 was just released.

closed 2018-04-09 by foosel

nickm324, 2019-01-13:
This issue still exists in 1.3.10.

foosel, 2019-01-14:
@nickm324 then please fill out the ticket template so this can be looked into

Edo78, 2019-01-21:
I'm experiencing the very same issue, even in safe mode. I'm on OctoPrint 1.3.10 running on OctoPi 0.15.1

It's not a big deal since I set the right value and now I'm autoconnecting but it's stilla bug ... do you want me to create a new issue for this bug?

seltix5, 2019-08-26:
Hello,
I have the same problem. Is there already a new issue created for this bug sinse this one is closed?

foosel, 2019-09-04:
Please open a new ticket.

seltix5, 2019-09-08:
new issue created https://github.com/foosel/OctoPrint/issues/3267