OctoPrint/OctoPrint #2312: Software Update fails with (near) full disk

tonsofpcs, 2017-12-22:
Attempted to run software update without checking free space first.  Space had been filled by timelapses. An update failure notice appeared but I attempted to restart octoprint anyway.  Octoprint no longer starts fully (I receive a 503 Service Unavailable error via http) after this, even after clearing additional space. 

Log follows:

`2017-12-22 00:11:02,243 - octoprint.plugin.core - INFO - Found 16 plugin(s) providing 13 mixin implementations, 9 hook handlers
2017-12-22 00:11:02,314 - octoprint.plugin.core - INFO - 16 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/discovery
| !DisplayProgress (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_displayprogress
|  DisplayZ (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_displayz
|  Filament Manager (0.2.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
| !GcodeEditor (0.2.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GcodeEditor
|  HeaterTimeout (0.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_HeaterTimeout
|  Navbar Temperature Plugin (0.8) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_navbartemp
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/pluginmanager
|  Printer Stats (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats
|  SnapStream (0-unknown) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_snapstream
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/softwareupdate
| !TouchUI (0.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_touchui
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/virtual_printer
2017-12-22 00:11:02,328 - octoprint.plugins.pluginmanager - INFO - The plugin was installed successfully: GcodeEditor, version 0.2.4
2017-12-22 00:12:32,151 - octoprint.plugins.softwareupdate - INFO - Starting update of octoprint to 1.3.6...
2017-12-22 00:15:19,037 - octoprint.plugins.softwareupdate.updaters.update_script - ERROR - Target: octoprint, error while executing update script, got returncode 1
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/softwareupdate/updaters/update_script.py", line 91, in perform_update
    caller.checked_call(update_command, cwd=folder)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/util/commandline.py", line 69, in checked_call
    raise CommandlineError(returncode, stdout, stderr)
CommandlineError
2017-12-22 00:15:19,058 - octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 893, in _perform_update
    update_result = updater.perform_update(target, populated_check, target_version, log_cb=self._log, online=online)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5rc3-py2.7.egg/octoprint/plugins/softwareupdate/updaters/update_script.py", line 94, in perform_update
    raise UpdateError("Error while executing update script for %s", (e.stdout, e.stderr))
UpdateError
2017-12-22 00:15:19,135 - octoprint.plugins.softwareupdate - INFO - Starting update of filamentmanager to 0.5.1...
2017-12-22 00:16:02,598 - octoprint.plugins.softwareupdate - INFO - Update of filamentmanager to 0.5.1 successful!
2017-12-22 00:16:03,684 - octoprint.plugins.softwareupdate - INFO - Starting update of navbartemp to 0.9...
2017-12-22 00:16:44,216 - octoprint.plugins.softwareupdate - INFO - Update of navbartemp to 0.9 successful!
2017-12-22 00:23:10,832 - octoprint.server.api.system - INFO - Performing command for core:restart: sudo service octoprint restart
2017-12-22 00:23:11,304 - octoprint.server - INFO - Shutting down...
2017-12-22 00:23:11,504 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2017-12-22 00:23:11,521 - octoprint.events - INFO - Event loop shut down
2017-12-22 00:23:11,588 - octoprint.server - INFO - Goodbye!

`

kantlivelong, 2017-12-22:
Can you SSH into your OctoPi and run the following:

1. > cd ~/OctoPrint
2. > git fetch --all
3. > git reset --hard origin/maintenance
4. > ~/oprint/bin/python2.7 setup.py install
5. > sudo /etc/init.d/octoprint restart

kantlivelong, 2017-12-22:
Update:

Spoke with @tonsofpcs on IRC. Issued compounded by the pip fix not being run for OctoPi 0.12. Resolved with:

1. >  rm -rf ~/oprint
2. > python2.7 -m virtualenv ~/oprint
3. > ~/oprint/bin/pip install --index-url https://pypi.python.org/simple --upgrade pip
4. > printf -- "[easy_install]\nindex-url = https://pypi.python.org/simple\n" > ~/.pydistutils.cfg
5. > ~/oprint/bin/pip install webassets --upgrade     #See #1725 
6. > cd ~/OctoPrint
7. > git fetch --all
8. > git checkout master
9. > git reset --hard origin/master
10. > ~/oprint/bin/python2.7 setup.py install
11. > sudo /etc/init.d/octoprint restart

kantlivelong, 2017-12-22:
Reproduced issue (to a lesser broken extent) by upgrading on a filesystem with 10MB.

StuckInLoop, 2018-01-20:
had plenty of storage space but, seems to had an issue which caused me unable to startup octopi,   followed your steps after a full reboot and gave it some time it was able to boot up again.   thanks

techfreek, 2018-10-03:
This seems like something I could take on. Any objections into me looking at this?

kantlivelong, 2018-10-03:
@techfreek go for it!

techfreek, 2018-12-26:
I'm finally looking at this again. Having a few difficulties, so I thought I might share my current thought on approach. This error came from update_script.py, which is called as a result of using the Update functionality in the webui.

My thought was that for each target we plan to update, we should figure out the size of that component (as a rough way of guessing how much extra space they may use during upgrade), and make sure the sum is less than the total amount of disk space remaining. In order to not add bloat elsewhere, I added this as a pre-check as I personally didn't think it made sense to allow some targets to update if we didn't have enough space to update all.

The problem I'm hitting is that I'm having trouble finding the installed path of the package. I have found the `update_folder` and `checkout_folder`, but it doesn't always seem set. I'm wondering the installed path doesn't normally matter. So maybe this is a case where I should add a get_path method to the different updaters so I can still maintain abstraction. 

Let me know if you have thoughts on my (admittedly rambling) thoughts. Thanks!

edumaker-limited, 2019-02-26:
Hi, thanks for your efforts in this issue.  

HELP! I have unfortunately paranoia checked this problem hoping to confirm it wasn't one, AFTER starting a 9-DAY, 1.2 metre tall print.  We are over 1.5 days in already and I really don't want to lose the work done already, let alone have it fail close to the end.  Ideally I'd like to keep the timelapse photos as they are good, but that's secondary.

My Setup: 
I have a generic clone Raspberry Pi CSI camera, and it is set to record a frame only on Z change,  However I also have some GCODE at the Z layer change which does a little 0.5mm Z hop - which is the same as the layer height.  So it could potentially be taking 2 frames per layer.
There are 2,338 layers so 4,676 photos potentially.
It's a 4Gb SD card, onto which I flashed the latest OctoPi 0.16 image - so it could be set lower than that.

-What is the best, light touch, approach to dealing with this?  
I expect I will need to SSH into the Pi (using PuTTY from a local Windows 7 machine), find the photo storage and check how much each picture consumes, and how many it has taken so far and compare with the layer height.
From this extrapolate how many will be needed to complete the print and the storage needed
Work out the free space available on the system and compare the two.
If not enough to clear with some overhead, I will need to copy photos off the Pi and delete them or just delete them.
- is there a safest method to copy the photos in terms of not interfering with the running print?
- any tips for dealing with the situation in the lightest touch manner possible?
- How much free space is advisable to leave on the Pi to minimise risk of system issues?

foosel, 2020-06-18:
Implemented by the above commit. Looks like this (obviously not the default minimum value, just set for purposes of the screenshot, default is 150MB):

[image]

To be released as part of 1.4.1

commit 462f2e60 (2020-06-18): ☂ SWU: Minimum free storage requirement

We now need to have at least 150MB (configurable) of free disk space on
all of site-packages, plugins and temp folder locations available in
order to enable updates.

Closes #2312
closed 2020-08-04 by foosel