OctoPrint/OctoPrint #2313: {Request] Plugin Blacklist needs individual Descriptions

sbts, 2017-12-23:
#### What were you doing?

Currently the blacklist only lists the plugin name and relevant version numbers.

#### What did you expect to happen?

A brief description of why the plugin is blacklisted should be present.
This should include any relevant issue numbers or links to other discussion.

The reason I'm suggesting this, is as the list grows longer (hopefully it doesn't....), it will help both users and plugin maintainers know why blacklisting has happened.
That will allow a blacklist entry to be added immediately, with no specific notification issued to the maintainer in advance

sbts, 2017-12-23:
I'm hoping the typo in the `[Request]` tag in the Subject was the cause of the incomplete ticket trigger.
It would really help if GitissueBot indicated to the user what information was missing etc

foosel, 2018-01-11:
I initially thought about this as well but then figured the blacklist would hopefully not see that much use (it IS for emergencies only). I agree with your points though, and it should be an easy addition. Best of all - no need to adjust OctoPrint for it, only the [blacklist page on plugins.octoprint.org](https://plugins.octoprint.org/blacklist/), so this could be implemented quickly "in between".

I've added an issue for that in the plugin repository tracker (better location since it's what would need to be adjusted, not OctoPrint), see OctoPrint/plugins.octoprint.org#151. Closing this ticket and proceeding over there.

closed 2018-01-11 by foosel

foosel, 2018-01-15:
Just for the record, this is now implemented and live on https://plugins.octoprint.org/blacklist/