OctoPrint/OctoPrint #2315: Temperature stuck on temperature tab

blinkey0, 2017-12-23:
So this has happened multiple times at different temperatures now so I thought I'd post about it.  The temp graph freezes for the extruder temperature but not the bed. I can still send new temperatures to change it but it won't change on the display. The terminal does show it updating as well as the display on the printer. Works after a reboot for a little while, then freezes again

#### What were you doing?

printing a file/cooldown after a print I'm monitoring octoprint from android and chrome on windows.

#### What did you expect to happen?
Hot end temperature to continually update to the reading of the thermistor
#### What happened instead?
Temperature freezes at some value
#### Did the same happen when running OctoPrint in safe mode?

I have not yet, currently a print underway but I will when it's done

#### Version of OctoPrint

OctoPrint 1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Rigidbot using MKS Gen 1.4 and Marlin 1.1.6
#### Browser and Version of Browser, Operating System running Browser

Android 6.0.1 - Chrome 63.0.3239.111
Windows 10 - Chrome Version 59.0.3071.115 (Official Build) (64-bit)

#### Link to octoprint.log

https://pastebin.com/ZSFf1u66

#### Link to contents of terminal tab or serial.log

https://pastebin.com/4iLBq5Cn

#### Link to contents of Javascript console in the browser

https://imgur.com/vsUpeZn

#### Screenshot(s)/video(s) showing the problem:

kantlivelong, 2017-12-27:
Hello,

If you send `M105` when this occurs does the graph update?

ntoff, 2017-12-27:
Looks like a duplicate of https://github.com/foosel/OctoPrint/issues/2302 

Both respond with "invalid extruder" after being issued a "T1" command, both using marlin 1.1.x

akf-was-here, 2017-12-28:
Sending M105 from the terminal doesn't make the graph update:

Recv:  T:220.94 /220.00 B:23.98 /0.00 @:84 B@:0
Send: M105
Recv: ok T:220.67 /220.00 B:23.98 /0.00 @:86 B@:0
Recv:  T:220.58 /220.00 B:24.02 /0.00 @:85 B@:0

foosel, 2018-01-11:
Agreed, looks like an exact duplicate. Closing.

closed 2018-01-11 by foosel