OctoPrint/OctoPrint #2316: OctoPrint 1.3.6 - Can't upload none printable files (png, jpg, etc..)

Thisismydigitalself, 2017-12-23:
#### What were you doing?
Upload a png file via the GUI to Upload folder
OctoPrint 1.3.6, OctoPi 0.14

#### What did you expect to happen?
For OctoPrint to upload the file.
I already have a PI for my other printer. i can upload any file without any problem.
OctoPrint 1.3.6, OctoPi 0.13

#### What happened instead?
Internal Server Error
The server encountered an internal error and was unable to complete your request.  Either the server is overloaded or there is an error in the application.

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
1.3.6, 0.14

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint
Win10/64
[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]
0.14
#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Chrome - Same for the other printer/OctoPrint which works just fine.
[If applicable, always include if unsure.]

kantlivelong, 2017-12-27:
Upload is currently limited to stl, gcode, gco, and g. I've marked this as a request.

Thisismydigitalself, 2017-12-27:
This is a very useful tool to copy files from windows to Debian. I use it on my other pi3 running img 0.13. it will be nice not to limit this option on this new 0.14 image. 

if there is a way around this for now or if there is a plugin that can do this until Octopi will support that, please let me know what to change for this to work on my newest octoprint img 0.14

Thx.

kantlivelong, 2017-12-27:
I'm not seeing anything in OctoPi that enabled uploading of any file type nor have I found any recent changes in OctoPrint that limited what could be uploaded. 

Can you provide fresh OctoPrint logs for the instance that allows you to upload?

In the meantime you can use something like WinSCP to upload. (I would suggest using this normally anyway.)

Thisismydigitalself, 2017-12-27:
Thank you. Interesting findings. 
Since my new OP running on img 0.14 do not allow me to upload files I looked for a solution and found WinSCP which I am trying out. 

Since I did not change OS settings on my 0.13 image, I wish for anyone who reads this thread to try to upload jpg or png using latest OctoPrint 1.3.6 running on img 0.13 and see if this works for them as well or not.

kantlivelong, 2017-12-27:
@Thisismydigitalself OctoPrint has limited upload to stl, gcode, gco, and g for quite some time. Plugins can extend what is allowed so I am guessing you have a plugin on the other instance that is allowing it. Please submit the logs if you would like assistance.

ntoff, 2017-12-27:
>This is a very useful tool to copy files from windows to Debian and I was sorry to see this feature got removed.

So you're not actually using this feature as part of octoprint, but just merely as an easy way to copy files to your debian device? If so, that's quite far out of the scope of OctoPrint (and this bug tracker). I suggest you look into either setting up some sort of writable samba network share, or tftp server.

kantlivelong, 2017-12-27:
Certainly not out of the question for a plugin though. I've been wanting to make a file browser/editor plugin but have not had time.

Thisismydigitalself, 2017-12-27:
Most interesting.
Following your suggestion i started comparing plugins between my two pi's and sure enough i found that my new pi running 0.14 image is missing some of the plugins that i am using on my other pi running 0.13.

Just installed:
+ OctoPrint-CustomBackground
+ OctoPrint-Gcodebar

And now I can upload jpg and png's. interesting.
Something weird thou. each time i upload a file my browser refresh itself - this does not happen on my other pi. i did notice that my 0.13 image states that it is:
 Using pip of "/home/pi/oprint/bin/python", **Version 8.1.1**

and my pi running on img 0.14:
Using pip of "/home/pi/oprint/bin/python", **Version 9.0.1**

Where are my OctoPrint logs located ?

ntoff, 2017-12-27:
logs are in the settings page (wrench icon in the nav bar) -> logs

kantlivelong, 2017-12-27:
On OctoPi - /home/pi/.octoprint/logs but it sounds like you figured out the issue.

CustomBackground is performing the refresh.

Thisismydigitalself, 2017-12-27:
Thank you. learned new things today.
CustomBackground is the "cultprit". without it i cannot upload files like jpg and such. with it - i can. each time i upload a file browser refreshes but this is no different than my other pi. howeve when uploading gcode file - browser will not refresh and will flash the file name with yellow background notice that upload is complete.

So. i wont bother you anymore..for now :)
Thanks for your help and tips.

ntoff, 2017-12-27:
>each time i upload a file browser refreshes

Probably because it's setting the image background to the temperature graph and refreshing to make sure the new image is loaded. OctoPrint doesn't need to refresh to load gcode scripts, the refreshing is a choice of the custom background plugin author.

kantlivelong, 2017-12-27:
Correct. Closing.

closed 2017-12-27 by kantlivelong