OctoPrint/OctoPrint #2317: Timelapse post roll taking too long

token47, 2017-12-24:
#### What were you doing?

Trying to capture a timelapse while printing. Configs are as on the screenshot on the end.

#### What did you expect to happen?

Post roll to take only a few (configured) seconds.

#### What happened instead?

Post roll took a long time (in this case 37 min, it seems to be a function of the time i put in the config (5s in this case).

#### Did the same happen when running OctoPrint in safe mode?

Did not try.

#### Version of OctoPrint

1.3.6 updated a few days ago. It seem to have started on this very update, didn't recall to see this before that.

#### Operating System running OctoPrint

Installed by hand on linux on a ODROID U3. But this installation has a long time, has been upgraded through many versions and this problem started recently. Don't think it's a problem with the installation.

#### Link to octoprint.log

https://pastebin.com/cXHHiAs4

#### Screenshot(s)/video(s) showing the problem:

[image]

[image]

token47, 2017-12-24:
Yes I omitted some information on the issue report but they do not seem to be related to the problem. Will revise it if needed.

token47, 2017-12-24:
Hummm, just wondering if "5s" in the config means 5 extra seconds in the final video, taken using the specified interval, that would explain it. But then, the config would be odd because the obvious thinking is to make sure to extend the footage a little after print ending and not to take a crazy amount of time to extend the final video with that increase in lenght. Hummm....

token47, 2017-12-24:
Yes, it is. At least the math almost matches. 37min == 2,220s. Also, 15s interval times 30 fps times 5s of final video is exactly 15*30*5 == 2,250.

If that is the case, please consider this a feature request to change the unverstanding of the config option.

ntoff, 2017-12-25:
If you don't need actual video (i.e. your printer parks itself and just sits there stationary after a print) just untick the box that says "capture post roll images". Then octoprint will just use the last image it captured during the print for the final 5 seconds of your video.

>If that is the case, please consider this a feature request to change the unverstanding of the config option.

It seems pretty clear to me, what would you propose to change?

token47, 2017-12-25:
No! It is in fact perfect as it is. I must be an idiot because it's explained below the "Capture post roll images". And now it makes sense. The problem was that at first (quick) look, it seemed that if I disabled the tick box it would disable the functionality completely.

Thank you!

closed 2017-12-25 by token47

tedder, 2017-12-29:
Aha. Mine was saying "125 minutes" for post roll, now I see why. This is going to repeat itself, it should be addressed in the UI..

tedder, 2017-12-29:
Developed in #384, was even confusing then. I think the word "rendered" is vague in this case, which is causing the problem.

ntoff, 2017-12-29:
How is it vague? It means 5 seconds worth of video after it's been rendered out by ffmpeg. As as I said to the  other person, if you'd like something changed then what would you propose it be changed to? If you think it's "too vague" then how would you make it more clear?

As for it "taking too long to capture them", that's just how timelapses work.

tedder, 2017-12-29:
If multiple people have been independently reporting the same thing since the feature was developed, that seems like a decent argument.

Current text:

>OctoPrint will take additional pictures to add this many seconds to the end of your rendered timelapse.

One example of how to word this would be to place the calculation next to the number- so "10sec", "25fps", "30sec" would say:

> OctoPrint will take 750 additional pictures over 75 minutes after print completion for the end of your rendered timelapse.

That may not be perfect- but not having the answer doesn't invalidate the concern.

JeJeff, 2018-01-02:
As a new Octoprint user I just got hit by this not-a-bug.  I was not expecting the post roll footage to continue to be taken in timelapse mode - after all, what is the point of it being a timelapse when the print is already complete?  When I configured 10 seconds of post roll I was expecting that to take 10 seconds of real time footage and append that to the timelapse, and not take 41 minutes to create a timelapse of something that isn't changing.

fiveangle, 2018-01-12:
I hit this on first use way back when as well, as I expected the post-roll capture time to be real-time to the setting, and not timelapse-stretched (for example to capture the print head moving off the print in real-time to the home position).   I didn't misunderstand it because the wording is unclear, but because I assumed the wording was a human overlook in describing the feature, because who in their right mind would ever want to take a timelapse-stretched timelapse to amass a specific number of post-roll seconds after completion anyway ?  But I quickly realized it was not a misstatement after first print, and so unchecked the box, never to be touched again.

The solution as I see it would be to remove the [absolutely useless as far as I can tell] feature entirely, or changed it to be what people expect, which is a real-time post-print capture of the number of seconds configured.  If there is some edge case in some alternate universe where a timelapse-stretched post-print capture would be useful, it should be available in the config.yaml only.  But that is just my opinion.  Perhaps there are droves toiling away in silent desperation awaiting to spring to action the moment this obscure "feature" is removed ;)

-=dave

foosel, 2018-01-12:
> I was not expecting the post roll footage to continue to be taken in timelapse mode

> a real-time post-print capture of the number of seconds configured

Yep, would be nice, I agree. But keep in mind that the timelapse currently depends entirely on a single snapshot URL. I can't capture from that at 25fps, the delay of connection establishment alone would kill this. 

The stream URL as is can't be used - it's needed to be in a form that can be embedded in the browser, and that's not necessarily a URL that makes sense to record from from the server itself (maybe not even reachable - stream and snapshot do not have to be the same camera).

So to do a continuous recording here we'd need a third camera URL. Or switch from fetching snapshots from a URL to constantly consuming an MJPG stream (which - as a side note - would then create new problems once this gets replaced by something a bit less anachronistic, once a good solution for such a server component is found). Which probably eats up quite some resources in the process for capturing timelapse stills, just to in the end maybe be able to capture a continuous real time post roll.

I don't know about you, but I think that doesn't sound like a good cost/usefulness ratio. 

Tempted to remove capturing those still myself now though. The only reason for adding post roll to begin with was so that the movie wouldn't simply end abruptly at the end of the print but instead allow "things to settle" visually. A still does serve the same purpose.

> Perhaps there are droves toiling away in silent desperation awaiting to spring to action the moment this obscure "feature" is removed ;)

Pitchfork wielding crowds are what I suspect to happen here as well.

(As a side note - scare quotes ('"feature"'), votes of temporary insanity ('who in their right mind') and similar ('absolutely useless') are exactly what I meant yesterday. These things do add up over time. A little less sarcasm and negativity really would help here :))

token47, 2018-01-12:
My problem was that even with the help/tip text right below perfectly explaining what was going to happen (and I failing to read it completely), my intuition was that checkbox off == no captures and checkbox on == capture of still frames for some more time (clock time, not rendered time).

Not sure if this is a real problem, but maybe a change of semantics would help.

JeJeff, 2018-01-12:
Personally, as long as clear warnings are given of the effects, I'd be fine with a real time postroll capture with doing any (or all) of:

- Stealing the the stream from the control page for the duration of the real time post-roll capture,
- Consuming lots of resources,
- Completely locking the server down until the postroll is completed.

But whilst that would work for me I realise not everyone uses octoprint the same way.

fiveangle, 2018-01-12:
Of course now thinking about it, it certainly would be a neat feature to be able to view the print head move off the print in realtime for the post-roll.

If there is no easy way to switch from timelapse capture to realtime capture for the post-roll, would it be worth testing the theory if anyone uses this feature by removing it (including the change in the release notes) and then waiting to see if anyone complains ?  I suspect we may all be surprised when nobody does.  Having the feature as-is seems more of an easy trap for those new to Octoprint so if nobody is really using the feature, it seems it may have more negative side effects than benefits.

[thanks for the comment Gina - I will try to quell the hyperbole 😄]

foosel, 2018-03-07:
The above commit removes the post roll capturing for good. Available on `maintenance`, soon `devel`, will be part of 1.3.7