OctoPrint/OctoPrint #2318: [Request] Gcode Viewer: show printing head location with a marker/pointer.

jvalencia80, 2017-12-25:
Is it possible to add an option to show where the printing head is currently? A simple marker would be nice.

I'm attaching a mockup of the feature.

Thanks for this awesome piece of software.
[image]

ntoff, 2017-12-25:
Since most printers don't offer any kind of feedback, the host doesn't know exactly where the print head is at any given point. The printer has an internal buffer that's slightly behind what the host sends and marking the head's position on the gcode viewer wouldn't necessarily be in sync with where it actually is.

jvalencia80, 2017-12-25:
It doesn't need to be perfect, it can be drawn on the last point of the current segment. Sometimes you just don't know where is it.

foosel, 2018-05-05:
Implemented in the above commit. Available on `maintenance`, soon `devel`, to be released with 1.3.9.

[image]

commit 8d9034b5 (2018-05-05): Gcode viewer: show approx. print head position

Also change retract/resume from dot to triangle/inverted triangle.

Closes #2318
closed 2018-07-25 by foosel