OctoPrint/OctoPrint #2320: Login button in Octoprint not responding

PrintedAlaska, 2017-12-26:
#### What were you doing?

Ideally provide exact steps to follow in order to reproduce your problem:

1. I opened octoprint as usual (wirelessly from the same wifi network)
2. I downloaded a new plugin called AstroPrint Cloud
3. I rebooted octoprint after the plugin successfully downloaded and now I can't log in or connect

#### What did you expect to happen?
I expected to be able to use the Astroprint plugin through octoprint.

#### What happened instead?
When I open the octoprint page, I cant click the Login button in the top right. Instead, I click it and nothing happens.

#### Did the same happen when running OctoPrint in safe mode?
I cannot access safe mode without logging in.

#### Version of OctoPrint
1.3.4
#### Operating System running OctoPrint
Windows 10 (google chrome)
#### Printer model & used firmware incl. version
Original Prusa i3 MK2

#### Browser and Version of Browser, Operating System running Browser
Google Chrome with Windows 10 x64

#### Link to octoprint.log
I typed the line to get to the logs into the raspberry pi and it said access denied. I am guessing I did it wrong. I did check the contributing guideline

#### Link to contents of Javascript console in the browser

See attatched .jpeg

#### Screenshot(s)/video(s) showing the problem:

[image]

Thank you for your help!
Mitch T.

I have read the FAQ.

ntoff, 2017-12-26:
If it stopped working after installing the astroprint plugin then it's probably safe to assume that plugin is to blame, might want to bring it up on their own issue tracker (if they have one).

To open octoprint in safe mode, close down octoprint and run it with ``octoprint serve --safe``.

kantlivelong, 2017-12-27:
Safe Mode documation can be found [here](http://docs.octoprint.org/en/master/features/safemode.html). Once you are in Safe Mode please disable the AstroPrint plugin and restart. If  functionality is restored then open a ticket with the plugin developer.

PrintedAlaska, 2017-12-27:
@ntoff The raspberry pi is the one running the octoprint server, it is connected to the printer. My computer is just what I control it with wirelessly through chrome.

ntoff, 2017-12-27:
Then that is what you should put in the section where it asks for the device running octoprint (for future reference). Knowing how it's installed helps us determine whether it's a standard raspberry pi install or a custom install on some other OS, instructions differ based on install type.

Were you able to figure out how to enable safemode? If not, the basic steps are:

* SSH into the raspberry pi https://www.raspberrypi.org/documentation/remote-access/ssh/ (skip to step 4)
* `sudo service octoprint stop`
* `source ~/oprint/bin/activate`
* `octoprint serve --safe`

and when you're done, you can either reboot the entire pi with `sudo service shutdown -r now` or stop octoprint with ctrl + c and `sudo service octoprint start` to start the service back up again. Rebooting isn't strictly necessary after uninstalling a plugin.

PrintedAlaska, 2017-12-27:
@ntoff That worked great! I restarted octorprint after deleting the astroprint plugin in safe mode and now everything is back to normal. Thank you for your help

kantlivelong, 2017-12-27:
@PrintedAlaska You can open an issue with the plugin developer [here](https://github.com/AstroPrint/OctoPrint-AstroPrint)

Closing.

closed 2017-12-27 by kantlivelong

CoDanny, 2017-12-28:
Sorry about this guys! It seems to be a javascript error in an edge case which prevents the rest of the javascript from being loaded. We're looking into it...

CoDanny, 2017-12-28:
This has now beed fixed in the plugin. Version 1.0.1 has this fix

kantlivelong, 2017-12-28:
Thanks @CoDanny

ntoff, 2017-12-28:
@CoDanny https://octoprint.org/blog/2017/12/01/heads-up-plugin-authors/ Just in case you weren't aware, that "use strict" clause was also the reason some changes were made to the way plugins are loaded. In case you haven't seen it, there's a blog post about it.

gdachs, 2017-12-30:
for what reason is there a request to doubleclick.net in the picture above?

foosel, 2018-01-11:
> for what reason is there a request to doubleclick.net in the picture above?

I'd like to know that too.

What plugins do you have installed @PrintedAlaska? Because that certainly isn't coming from stock OctoPrint and if a plugin out there is embedding ads, I'd like to know which one so I can shout a bit at the author.