OctoPrint/OctoPrint #2321: Print starts and hotend is never heated

michaelwu21, 2017-12-27:
#### What were you doing?
Clicked print on a gcode file, but forgot to click the preheat button first(from preheat plugin), as I like the bed and hotend to heat up at the same time, as it reduces time.

I paused the print, and canceled it, and now click the preheat button. Nothing happens(even clicking it multiple times), hot-end is not set to 215 degrees, and bed continues to heat to 70 degrees.

I decide whatever, I'll just click print again, and wait for the bed to hit 70 degrees, and after, it should automatically set the hotend to 215, and start printing after(I dislike this as it takes forever to heat up one after the other which is why I use the preheat button)

#### What did you expect to happen?
Bed heats up to 70-- Hotend then heats to 215, print starts great.

#### What happened instead?
The bed heated up to 70 degrees, and the hotend went to 215, as shown in the temp graph, and it starts printing. However, the heaters are shut off immediately, both the bed and hotend, and the temperatures start dropping while it prints. After about 2 minutes, the bed turns back on and climbs back to 70, but the temperature of the hotend drops to 30 degrees. I only noticed a bit later when I saw no filament coming out of the extruder.

I reproduced the issue and this is what the printer did after starting.
After homing axis, it goes to print but immediately activates the pause sequence, which in my case moves the head off to the side. It then goes back to the printing spot and starts printing, but the heaters for the hotend and bed are turned off.

I think it's an issue with pausing the print before the print actually starts, and even if you do cancel the print, it still pauses at the beginning of the print for some reason.

#### Version of OctoPrint
Octoprint 1.3.6 running on Octopi 0.14.0

#### Operating System running OctoPrint
Octopi on Pi Zero- I'm accessing octopi.local/ from Chrome on Windows 10

#### Printer model & used firmware incl. version
Wanhao Duplicator i3- Melzi board running Marlin

#### Browser and Version of Browser, Operating System running Browser
Latest version of Chrome

#### Link to contents of terminal tab or serial.log
https://pastebin.com/RDqDxv6E

I shut down the printer to avoid printing without heated extruder which is why it says "Printer reset by external source?"

#### Link to contents of Javascript console in the browser
No console.log after "User has logged in"

#### Screenshot(s)/video(s) showing the problem:
[image]

kantlivelong, 2017-12-27:
Hi @michaelwu21 

This issue tracker is not for plugin support. Please try to recreate the issue in [Safe Mode](http://docs.octoprint.org/en/master/features/safemode.html). If you are able to replicate the issue please submit new logs while in Safe Mode and be sure to include **octoprint.log**.

foosel, 2018-01-22:
@michaelwu21 please try what @kantlivelong suggested. Also provide a gcode file you are seeing this with, as well as your configured GCODE scripts.

michaelwu21, 2018-01-22:
Sorry but I don't have time to try a safe mode print at the time being as I'm replacing my board and rebuilding parts of the printer.

However I did try it a few more times before and noticed it had nothing to do with the preheat plugin, but it occurs every time I pause/cancel the print **before** it actually starts, and then click print again(as I changed my mind). 

The printer will home, but then execute the pause/cancel sequence before starting the print, like I just hit the pause and then print button, even though I'd clicked them before the print started.

After the cancel sequence, the heaters are turned off. My gcode scripts for cancel have the heaters turned off, which is the issue here. I believe there's a bug that basically queues the stop sequence, and doesn't remove it(as it never occured), and the stop sequence will be executed during the next file that is printed.

foosel, 2018-01-23:
All commands that ever sent to the printer enter a queue, yes. But that queue is processed in order. So I don't see how you could have a cancel from a prior print come after the start of another (which I understand to be happening here?). 

Please share your GCODE scripts and a file you observed this with.

michaelwu21, 2018-01-25:
What I mean is that I click print, and as it should, octoprint waits for the hotend and bed to heat up. However, I cancel the print **before it's done heating up**(the first time was due to selecting wrong file). On the next print I start, what I described above occurs.

It happens no matter what file I print.

Cancel script:
; disable motors
M84

;disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
;disable fan
M106 S0

Booli, 2018-02-20:
Hey,

this is typical printer behaviour. As you found out, and foosel notes, all commands are put in _the printer's_ buffer. Once they are in the printers buffer, OctoPrint has no reliable way of knowing if it really has been executed yes or no. 

What happens here is, that the bed heat up commando from the first started print is a blocking commando(M190), this means the printer goes into full blocking mode until the bed is really heated up to 70 degrees. Only then, it will return an 'ok' and continue with the next commands in the buffer(which are actually just stop commands since you have just stopped the print). And after that will happily start the commands that follow, which is your new started print.

While heating up the printer does give some information to OctoPrint, so it keeps the connection live. It is weird UX, that you do not see that it is in blocking mode still from an earlier started print, but that because it's very hard to predict all printers firmare. 

Anyways, one a quick workaround could be, to connect and disconnect the printer after you have started the print but canceled it before it really started. If your printerboards firmware is setup to reset itself during serial reconnection, it can actually cancel the initial heat up commands. 

Other work arounds tie into specific firmware commands that can abort heating commands, but I am not familiar enough with your printersfirmware to know if it supports that.

foosel, 2018-03-05:
Actually it looks like this is the same as what was reported and thanks to all full logs also identified in #2449. 

In short: state switch to "Operational" even though the cancel script hasn't yet executed, due to that the possibility to start a new print job, and due to the way things are processed the very first line of the file being able to enter the send queue before what is still waiting to be sent from the configured cancel script. I'm not yet entirely sure how to intentionally reproduce this, but at least I can see how it would be possible.

foosel, 2018-03-05:
Solved by the above commit 59cbfaf, available on `maintenance`, soon `devel`, to be released with 1.3.7

foosel, 2018-04-09:
Closing as 1.3.7 was just released.

closed 2018-04-09 by foosel

stupudmunkee, 2018-04-15:
Not wanting to kick a dead post, but is it possible this was reintroduced in 1.3.8? I cancelled a print in a similar fashion, now unable to get the heaters to temp through octoprint. I'll troubleshoot and log in the morning, but if someone were familiar with the issue and had a moment to review it might be faster.

foosel, 2018-04-16:
The only change between 1.3.7 and 1.3.8 concerns a third party dependency that is completely unrelated to communication and that change is only relevant during update/initial install anyhow.

So if you are seeing this again in 1.3.8, it sounds more like another version of it that so far went unpatched.

jareem, 2018-05-09:
I confirm I have a similar issue. Now I am able to heat the bed through Octoprint, but no more the hotend.
If I set a temperature of let's say 80 degrees, the target is set, and the hotend heats up.
If I set a temperature of 205degrees, the target is set, hotend start to heat and goes back to OFF almost immediately.
Will provide logs later on.