OctoPrint/OctoPrint #2322: M117 Messages Clipped on Printer

lesds, 2017-12-27:
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
aka the Contribution Guidelines that are linked ^-- just up there
in the big yellow box.

Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ...", "I have a
question ..."), that belongs on the mailing list or the G+ community
(both linked in the "guidelines for contributing" linked above, read
them!), NOT here.

Mark requests with a "[Request]" prefix in the title please. For bug
reports fully fill out the bug reporting template (if you don't know
where to find some information - it's all described in the Contribution
Guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete ANY lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to report is still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the Contribution
Guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

Ideally provide exact steps to follow in order to reproduce your problem:

1. ...
2. ...
3. ...

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]

#### Printer model & used firmware incl. version

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

lesds, 2017-12-27:
Please forgive me it this isn't the right forum (Noob). I have a post-processing tool chain that creates embedded LCD messages of the form: "M117 Layer 525 of 733" that is being truncated on my printer.

ntoff, 2017-12-27:
Please don't abuse the bug tracker by posting blank issues, this is not the correct place to seek help. Either use the mailing list or the google+ community group. Links to both can be found in the contribution guidelines https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#issues-tickets-however-you-may-call-them (they're a little hard to spot, scroll down passed the flow chart, underneath where it says "read the FAQ" you'll find the links to the mailing list and the G+ community).

More than that, OctoPrint has no control over what your printer's firmware decides to display on the screen, whether it scrolls the text or truncates it. If the text is longer than the display's character count then it may truncate it, there's nothing OctoPrint can do about that, you'd be better off opening an  issue on the firmware's issue tracker asking them to implement text scrolling.

If you believe your printer should be able to display the full character length but octoprint *is* to blame (possibly sending the wrong commands) then please fill out the entire issue template properly, including all the information that you chose to just ignore.

lesds, 2017-12-27:
Well jeez I guess that this doesn't mean a bug either then:

kantlivelong, 2017-12-27:
@lesds you need to supply all of the required information.

@GitIssueBot I'm done fighting with you!

ntoff, 2017-12-27:
If you want to report an issue then, as I said, please fill out the issue template correctly. You still have it up there in your original post. In order to investigate any issues, you'll need to provide the logs (especially the serial logs). Nothing much can be ascertained from just your short little truncated terminal text. 

Please see the contribution guidelines https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report

closed 2017-12-27 by lesds