OctoPrint/OctoPrint #2323: Request 1.3.4 > 1.3.6 update broke camera stream (image display scrambled)

N3RDPWR, 2017-12-27:
#### What were you doing?

Installing Octoprint fresh on RPi (1) B+
Sony 32GB uSD Class 10
Logitech C270 USB webcam

Ideally provide exact steps to follow in order to reproduce your problem:

1. Download Octoprint image from website, Flash to SD card following Youtube guide on Octoprint site.
2. Boot for first time, go through initial setup wizard (camera working fine at this point)
3. Update to 1.3.6, restarted once prompted, once restart was complete camera was still working fine
4. Shutdown Pi through software, power down to relocate Pi
5. Power up again, log in and camera feed is scrambled. Tried several reboots/ power down, unplug replug, etc...
6. Installed full screen plugin in hopes of fixing problem, plugin features work but imagine still scrambled 
7. Repeat step 1 to start fresh, camera works with 1.3.5 image and breaks the same way as I update to 1.3.6

#### What did you expect to happen?

Update to 1.3.6 without breaking anything

#### What happened instead?

Camera feed is coming online, however imagine is scrambled

#### Did the same happen when running OctoPrint in safe mode?

Webcam looks the same in Safe Mode

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

Octopi 14

#### Printer model & used firmware incl. version

Tevo Tornado - Marlin Tornado 1.1

#### Browser and Version of Browser, Operating System running Browser

Google Chrome Version 63.0.3239.108

#### Link to octoprint.log

https://gist.github.com/N3RDPWR/588869051c65c6a1bd9f7551ba55c790

#### Link to contents of terminal tab or serial.log

(Serial Log not enabled)

Terminal log empty, likely due to not being connected to printer at the moment

#### Link to contents of Javascript console in the browser

https://gist.github.com/N3RDPWR/5ed725a913083e475c996a31502ec60c

#### Screenshot(s)/video(s) showing the problem:

[image]

Added update log for additional info

https://gist.github.com/N3RDPWR/342f91f2ea1be9b4a1837fd6c65efb7c

I have read the FAQ.

ntoff, 2017-12-27:
Wrong tracker, OctoPrint itself doesn't handle the webcam, all it does is embed the image in the page. Mjpeg streamer handles the webcam functionality, and octopi is the prebuilt image that contains octoprint + mjpeg streamer. Any malfunctions with the camera should either be directed towards mjpeg streamer's bug tracker, or the octopi bug tracker if it happens one of the default bundled configs is now broken.

https://github.com/guysoft/OctoPi/ octopi tracker

https://github.com/jacksonliam/mjpg-streamer mjpeg streamer tracker

You could also try asking for help in the mailing list https://groups.google.com/group/octoprint or in the google+ community https://plus.google.com/communities/102771308349328485741

N3RDPWR, 2017-12-28:
ntoff
http://octopi.local/webcam/?action=stream displays the same thing unfortunately.
I will look into the issue with octopi track or mjpeg streamer tracker. After troubleshooting some more I noticed in 1.3.4 the issue does come up as well so it's likely not strictly due to the update. I suspect some auto detect feature is picking an incorrect setting for the C270. I will try to manually configure the camera and see if I can get it to work properly.

I appreciate your response on the matter.

ntoff, 2017-12-28:
If it happens with that URL then it's mjpeg streamer settings or your camera, not octoprint / octopi.

Just to clarify what they all are

* OctoPrint = a host program like repetier host or pronterface (printrun). It does nothing but control your printer.
* Mjpeg streamer = a separate program that takes the images from your camera and sends them out over your network connection
* HAProxy = (short version) a bit of kit that can redirect urls you type in your browser to various servers running on different ports
* OctoPi = a modified version of Raspbian Lite that includes OctoPrint, Mjpeg Streamer, and HAProxy already installed and configured, along with a version of the old CuraEngine slicer.

So if you're having issues with your camera feed then it's most likely an issue between the camera and mjpeg streamer, you can check the known camera config page https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work to see if you need any extra configuration options to make it work.

I find the safest way to edit the config files is either with NotePad++ on windows, or the easiest way is  logging in to the pi with SSH and running `sudo nano /boot/octopi.txt` to directly  edit it on the pi without having to move the SD card around. "sudo" to give you super user access, "nano" is a lightweight text editor, and obviously "/boot/octopi.txt" is the path for the config file mjpeg streamer is looking for.

Happy bug hunting.

N3RDPWR, 2017-12-28:
I have gone as far as editing the octopi.txt in SSH w/ nano but nothing changes the scrambled image and using sudo systemctl stop/start webcamd when making a change. 

Oddly enough I do see shadows of what is in the picture, such as when I move my hand in front of the camera and I can tell the camera_usb_option="-r  -f" setting (in octopi.txt) does what it's supposed to in terms of resolution and fps. However forcing the camera to "usb" (camera="usb" or even camera="auto") does completely break the stream in octopi and http://octopi.local/webcam/?action=stream (refuses to load).

From what I can tell, I should change parameters in input_uvc.so but I currently don't know how to do/use that.

The camera I am using is a Logitech C270 and apparently a common compatible (and trouble free) camera.

ntoff, 2017-12-28:
Unfortunately I can't help as I don't have that camera and my camera "just works" so haven't had to fiddle with it. 

>From what I can tell, I should change parameters in input_uvc.so but I currently don't know how to do/use that.

The options you put into the octopi.txt file just automatically go into your input_uvc section, they're just put in a spot that's easier to get to.

Forcing the camera to usb isn't necessary since mjpeg streamer has obviously found your camera.

Sadly since this isn't related to octoprint itself and is more a support request, this  bug tracker isn't the right place for it. You could try the raspberry pi forums, or perhaps check out the mjpeg streamer bug tracker to see if anyone else has the same issues. There's also the google+ group or the mailing list that I linked to earlier.

Hope you get it working.

N3RDPWR, 2017-12-28:
I appreciate your input, I will direct my request for help to the proper group.

Thanks again

closed 2017-12-28 by N3RDPWR

N3RDPWR, 2017-12-30:
As it turns out, no amount of configuration and debugging could help my defective SD card. My problems kept getting worse to the point that I could no longer edit the contents of the SD card. A new SD with a fresh image worked like a charm. The issue did come up on the new image after a accidental power down the the RPi and I was able to bring the camera back up by doing a sudo systemctl stop webcamd and sudo systemctl start webcamd seems to bring the camera image back to normal.

Sorry if this is not directly related to Octoprint but since it's out there, I might as well put what resolved it.