OctoPrint/OctoPrint #2325: [Request] Make GCode Viewer ignore layers with no movements

GurliGebis, 2017-12-27:
I have a Prusa i3 MK2S where I use the following start G-code:

`M115 U3.1.0 ; tell printer latest fw version
M83 ; extruder relative mode
G28 W ; home all without mesh bed level
G0 Z60 ; get head above bed to keep it cool
M104 S150 ; set extruder temp to 150C cold enough to avoid ooze but get some temp on it
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S150 ; wait for extruder temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0`

It moves the head to Z60 to keep the pinda cool while the bed is warming up.
This move causes Octoprint to think that there is a layer at 60.00, even for small objects.

It would be nice, if there was a way to tell it to ignore the first part when calculating the object dimensions, or maybe make it ignore layers where there is no filament being printed. (Since it only moves up there, and then back down again when it is done heating the bed).

GurliGebis, 2018-01-23:
@foosel does this sound like a good idea? (I'm not that fluent in python to be able to easily evaluate if it can be done)

foosel, 2018-01-23:
I haven't yet gotten around to take a closer look. Too much stuff accumulated over my Christmas break, still catching up and concurrently fighting off a cold 🤧

GurliGebis, 2018-01-23:
Ohh, no worries - I'm not in a hurry :)
Get well soon.

LazeMSS, 2021-03-31:
@GurliGebis in the latest OctoPrint 1.6.0rc1 there is a new option in Settings -> Features -> Gcode viewer that allows you to configure a "Skip to this" option. 

Depending on your slices most slicers makes a "comment" like: `; layer 1` try this 

This should fix it.

Closing for now - comments if you disagree and we can reopen.

closed 2021-03-31 by LazeMSS

GurliGebis, 2021-05-03:
It works as excepted - thanks :)