OctoPrint/OctoPrint #2330: octoPrint API makes octopi (and then octoprint...) crashes since 1.3.6

cornebistouille, 2017-12-29:
#### What were you doing?

my printer is far from may computer, so i use Octoandroid since few month to begin the print just beside the printer and then i monitor print from octoprint on my PC... 
but since 1.3.6 when i connect an android application to octoprint via API or the new cura engine wich can use the api, i can use it only from the android or Cura and cannot connect via http on my pc...
 neither ssh connexion or ping to my rasberry is possible, and if i quit the app, impossible to reconnect the only thing i can do is restart the Pi...

Ideally provide exact steps to follow in order to reproduce your problem:

1. connect via http to octopi.local
2. put some gcode in octoprint
3. connect my android using API,
4. setting my printer via android (auto home for level, pre heating...)
5 launch printing
6 trying unsuccesfully to reconnect to octoprint on my PC... no ping on ip or octopi.local...

this were acting correctly before 1.3.6, i tried other applications (printoid, octodroid) with same bad results...

#### What did you expect to happen?

be able to monitor my printing on my PC, and on my android application

#### What happened instead?

octopi crashes, only solution is to restart it...

#### Did the same happen when running OctoPrint in safe mode?

yes

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

[OctoPi 0.14
i had to get update ffmpeg which was in a version that doesnt work with octopi streaming)

#### Printer model & used firmware incl. version

CR 10

#### Browser and Version of Browser, Operating System running Browser

Chrome 63.0.3239.108, W10

#### Link to octoprint.log

[octoprint (1).log](https://github.com/foosel/OctoPrint/files/1592388/octoprint.1.log)
i see a lot of http 409 here with the rest API...

#### Link to contents of terminal tab or serial.log

serial.log 
impossible : the sefver is down/stuck/unavailable : no way to get interce...

#### Link to contents of Javascript console in the browser

idem

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

kantlivelong, 2017-12-29:
According to [this](https://github.com/foosel/OctoPrint/blob/777c02135eeb256437081dfd81d1e0596b45ff4f/src/octoprint/server/api/printer.py#L25) you are getting the error because the printer is not operational.

Does everything work from the WebUI without using any 3rd party tools?

ntoff, 2017-12-29:
Looks like the same thing from this issue https://github.com/foosel/OctoPrint/issues/2329 some error with getting a timelapse video download link seems to nuke tornado (at least that's what it looks like to my highly uneducated eyeballs).

cornebistouille, 2017-12-29:
mmm... a this time octopi was effectivelly rending a video. an by thinking of it, i remember that somes crashes came when i stop the printing (and btw octoprint start rendering timelapse...
obviously, some changes have been made on the video part in 1.3.6, since I had to update mjpeg-streamer in order to have a correct lve streaming (it crashed after few seconds) may be this is related...
pi@octopi:~/mjpg-streamer $ ./mjpg_streamer -v
MJPG Streamer Version: 2.0
Compilation Date.....: Jun 25 2017
Compilation Time.....: 00:38:45

kantlivelong, 2017-12-29:
@cornebistouille could you try disabling the webcam stream service and disable timelapse in OctoPrint?

cornebistouille, 2017-12-29:
ok, on my next print (in a hour or two)
what i wil do : 
reboot octopi
disable timelapse
disable stream service (kill -9 mjpg_streamer)
connect printroid
do a few command on printroid
start printing on octoprint
cross finger ;)

ntoff, 2017-12-29:
>disable stream service (kill -9 mjpg_streamer)

instead do `sudo service webcamd stop`

cornebistouille, 2017-12-29:
ok... it continue to crash... my printing is on the way... the printer receive the command (or has received, i don't know) but octoprint is unavailable... no way to ping the pi, http://octopi.local doesn't answer, nor directly via ip adress...

cornebistouille, 2017-12-30:
@kantlivelong sorry i didnt see your first comment... the printer seems to be opérational since it is printing from octoprint even when the server is unreachable...
may be the serial log will tell us more.... but we have to wait for the impression in progres (20 hours...)

cornebistouille, 2017-12-30:
just a question, waiting the end of the print...
does octoprint send the whole file in one time to the printer or does it send it line by line? because i am asking myself about the state of the octoprint server : is it stuck or is it stil sending command?

ntoff, 2017-12-30:
It sends the gcode line by line, the printer does have a buffer in it but it's only stores a few lines of code

cornebistouille, 2017-12-30:
thanks @ntoff so that means that my octopi has'nt die, neither octoprint ... only the network connection... the logs wil be instructive :)

cornebistouille, 2017-12-31:
hello
here we are....
unfortunately it seems that serial log is ok but i dont see the period between 29th and 31st ... on all that period the printer was printing but with no access...
https://www.transfernow.net/786vg3r7hi70 (files are too big for github ;) )

jvdbrg, 2018-01-01:
I can confirm this bug. Previous version was rock stable but afther  upgrading octoprint i experience random crashes. The Pi is still accessible via ssh and a reboot results in a working ocroprint.

Yesterday it crashed half way during a 4 hour print and the printer also freezed.

kantlivelong, 2018-01-03:
@cornebistouille Are you restarting the Pi via a reboot command in ssh or physically by pulling power? Can you connect to SSH when the issue occurs like @jvdbrg?

@jvdbrg (and @cornebistouille if possible) what happens if you just restart OctoPrint via SSH? `sudo service octoprint restart`? Does OctoPrint restart properly?

kantlivelong, 2018-01-03:
@cornebistouille disregard last message. If the Pi is completely inaccessible it may be worth re-imaging it with a fresh install as that shouldn't be caused by OctoPrint (though still technically possible).

sbts, 2018-01-04:
@cornebistouille One other things it may be worth doing is connecting a Keyboard and Monitor to the Pi and running the following commands then paste the results back here....
- `ip a`
- `ip r`
- `ifconfig`
- `route -n`
- `free -h`

Yes, `ip a` and `ifconfig` return almost identical info, as do `ip r` and `route -n`
However, sometimes it's useful to see the output of both as they are not identical

cornebistouille, 2018-01-04:
oh, no problem i'll try that, even if it makes print crash... I am a man of sacrifice :) i away from home right now but I'll try this WE, it will let me the time to consider how to get a working screenwith HDMI near the printer :)

cornebistouille, 2018-01-04:
@kantlivelong that was the first thing i have done... with 3 different SD cards... always the same issue. qhen i connect to octoprint via a mobile application (printroid, octodroid...) which uses API, and do some moves, server turns off connexion and no way to connect to octopi neither via the app, SSH or web..

sameSEA, 2018-01-06:
Same thing started happening here after update. Looks like it has problems. And what else.. Installed a FilamentManager plugin but I don't think it is a cause according logs.

cornebistouille, 2018-01-06:
hello, i got some news :
i submit the bug to printoid (the app i use) [here](http://printoid.forumforall.com/sujet-583257-789609-94353-1-Bug-With-1-3-6.html#endtopic) and the developpers tells me that there an issue in the [raspbian kernel](https://github.com/raspberrypi/linux/issues/1342)  wich cause wifi to go to sleep mode...
so i try to put 
wireless-power off 
at the end of my /etc/network/interfaces (sudo nano /etc/network/interfaces) and we'll see :)

cornebistouille, 2018-01-06:
as you can see, wifi interface as a very high Tx excessive retrries (14k in minutes...)
that should explain the problem : may be thin changes in API charge more the network and when i updated it overflow the wifi and causes the bug...
if its true, its not really a bug, but my advice is that users who experienced (under octopi distribution) same problem may : 
ssh octopi and type :
`sudo iwconfig wlan0 power off`  
test if its ok 
then edit /etc/network/interface : 
`sudo nano /etc/network/interfaces`
by putting (on the wan part) : 
`wireless-power off`
then CTRL + x
and it should worlk...

cornebistouille, 2018-01-06:
tadaaa!!!
look at the tx excessive retries 👍 : 
`pi@octopi:~ $ iwconfig
lo        no wireless extensions.

wlan0     IEEE 802.11  ESSID:"Livebox-19B0"
          Mode:Managed  Frequency:2.412 GHz  Access Point: 90:4D:4A:E7:50:50
          Bit Rate=72.2 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=61/70  Signal level=-49 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:60169  Invalid misc:0   Missed beacon:0
`

kantlivelong, 2018-01-06:
Looks promising. Hopefully that is indeed the issue.

cornebistouille, 2018-01-06:
ok no more bug : problems solved :) may be it can be usefull to put a request on octoPI to tell them to permit to add wireless-power off in their conf file :)

kantlivelong, 2018-01-06:
Would definitely suggest submitting the ticket. Interesting that I haven't been affected but I use a Pi 3

@sameSEA @jvdbrg does this resolve the issue for you as well?

cornebistouille, 2018-01-06:
and for the beauty no more TX excessive retries 👍 : 
```
pi@octopi:~ $ iwconfig
lo        no wireless extensions.

wlan0     IEEE 802.11  ESSID:"Livebox-19B0"
          Mode:Managed  Frequency:2.412 GHz  Access Point: 90:4D:4A:E7:50:50
          Bit Rate=72.2 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=56/70  Signal level=-54 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:60169 Invalid misc:0   Missed beacon:0
```
thank you to all of you for your support :)

sameSEA, 2018-01-07:
I works for me unless I initialize time lapse - it will freeze within minutes if it is on. Even if print not yet started.
When it happens, Octopi not accessible even on Ethernet.

foosel, 2018-01-11:
@sameSEA What do you mean with "initializing time lapse" - configuring the timelapse settings to something else than "off" on the timelapse tab, or something more like plugging in a webcam (which is otherwise not plugged in)?

Also, @sameSEA and @jvdbrg please *both* of you specify what still works and doesn't work once the problem occurs, because right now it sounds like there are multiple expressed behaviours treated as one here and that doesn't help in looking into this:

  - Does the print continue?
  - When you try to connect to OctoPrint's web interface, what does happen - do you get an error page that tells you that OctoPrint isn't running, that the server is refusing connections or can your browser not connect at all?
  - Can you still SSH into the pi?
  - Is the OctoPrint server still running? (via SSH: `ps -ef | grep -i octoprint` and provide the output)
  - Can you ping the pi via its hostname?
  - Can you ping the pi via its IP address?

Also please both of you provide an `octoprint.log`.

foosel, 2018-02-05:
Closing due to lack of response from "me too"s and solved issue for the OP.

closed 2018-02-05 by foosel

MarScaper, 2018-05-27:
Many many thanks for the fix! My PI Zero works now fine with OctoPrint! :D