OctoPrint/OctoPrint #2332: my printer "stutters" and sometimes pauses completely when reloading web interface

pbbob, 2017-12-30:
#### What were you doing?

1. Printing something
2. reloading the octoprint web interface page. (i've also noticed the bug while logging in/out)
Note that the bug only occurs about 1 in 4 times.

#### What did you expect to happen?
my printer to continue running smoothly

#### What happened instead?

the printer slows down and "stutters", sometimes even stops completely, for varying times, up to 2-3 secconds.

#### Did the same happen when running OctoPrint in safe mode?

I am currently in the middle of a print, so i cannot restart octoprint right now. I will add a comment later with the results during safe mode.

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.14.0 

#### Operating System running OctoPrint

octopi

#### Printer model & used firmware incl. version

Printrbot simple metal, rev F5 printrboard, unknown firmware version

#### Browser and Version of Browser, Operating System running Browser

firefox 57, ubuntu 16.04

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

https://pastebin.com/RnZ7Q6Z6 (note that i copied the terminal contents about 3-4 minutes after the bug occured

#### Link to contents of Javascript console in the browser

https://pastebin.com/WwuhpiJg

#### Screenshot(s)/video(s) showing the problem:

vimeo.com/249199090
vimeo.com/249199097

I have read the FAQ.

fiveangle, 2017-12-31:
https://github.com/foosel/OctoPrint/issues/1241

kantlivelong, 2018-01-03:
Duplicate of #1241

closed 2018-01-03 by kantlivelong