OctoPrint/OctoPrint #2333: Delay when clicking Disconnect button

ScottWell1, 2018-01-01:
#### What were you doing?
I clicked the Disconnect button to disconnect Octoprint from the printer.

#### What did you expect to happen?
A reasonably immediate disconnect. 

#### What happened instead?
There is a delay approximately equal to the "Settings, Serial Connection, Communication timeout" value.  For example, with the default Communication Timeout value of 35 seconds, pressing Disconnect results in an approximately 35 second delay before the connection is closed.

Two console messages are issued when Disconnect is pressed, with a delay between them:
Changing monitoring state from 'Operational' to 'Offline'
    _(Delay here, approximately equal to Communication Timeout setting)_
Connection closed, closing down monitor

#### Did the same happen when running OctoPrint in safe mode?
NA

#### Version of OctoPrint
1.3.6 Stable

#### Operating System running OctoPrint
Octopi 0.13.0  (fully updated)

#### Printer model & used firmware incl. version
Lulzbot Mini with Aleph Objects Marlin build 1.1.5.64

#### Browser and Version of Browser, Operating System running Browser
NA

#### Link to octoprint.log
Nothing is logged to octoprint.log for this event.

#### Link to contents of terminal tab or serial.log
Relevant lines from serial.log (with Communication Timeout at 35 sec):
**2018-01-01 12:07:46,245 - Changing monitoring state from 'Operational' to 'Offline'
2018-01-01 12:08:19,959 - Connection closed, closing down monitor**
  _(note ~35 seconds elapsed)_

Relevant lines from serial.log (with Communications Timeout at 5 sec):
**2018-01-01 12:05:13,923 - Changing monitoring state from 'Operational' to 'Offline'
2018-01-01 12:05:17,041 - Connection closed, closing down monitor**
  _(note ~4sec elapsed)_

#### Link to contents of Javascript console in the browser
NA

#### Screenshot(s)/video(s) showing the problem:
NA

I have read the FAQ.

kantlivelong, 2018-02-05:
@ScottWell1 

Looks like this slipped by us. Could you please attach full octoprint and serial log files?

ScottWell1, 2018-02-05:
@kantlivelong @foosel 

As mentioned, nothing is logged to Octoprint.log when this occurs.  The problem starts when the "Disconnect" button is pushed -- and at that point, only two lines are added to serial.log -- also shown in Terminal.  Those lines from serial.log and Terminal view are included in the ticket above.

This should be super-easy to reproduce.  It isn't an odd intermittent problem, it happens every time.  Just click the Disconnect button and watch the terminal window for the delay, which matches the "Communications Timeout" value.  

Here are fresh, new logs I just created.  Scroll to the bottom of the serial.log and look at the last two lines, which (exactly like those provided earlier) show (1) When I clicked Disconnect, (2) The 20 second delay before "Connection closed" occurs (this was done with "Communications Timeout" set to 20 seconds).

[octoprint.log](https://github.com/foosel/OctoPrint/files/1696007/octoprint.log)
[serial.log](https://github.com/foosel/OctoPrint/files/1696009/serial.log)

ScottWell1, 2018-02-06:
> Can you reproduce it right after connecting or do you first have to 
> print or have the connection run for some time? 

Yes, it happens every time.  It does not matter whether it has been connected for a few seconds, or a few hours.  It does not matter whether any printing has occurred.   The serial.log I attached earlier demonstrates connecting, waiting a bit (no print initiated, just some auto-temperature reports), then clicking Disconnect -- and the delay occurs.

It also occurs in Safe Mode, so it isn't plugins.  

It is not affected by the "Enable automatic firmware detection" feature.  I tried disabling that, thinking perhaps the incoming auto-temperature reports were involved.  But the same delay occurs when manual M105 polling is being used.

It happens at both 250000 and 115400 baud rates.

Don't really know what else to check....

Always, upon clicking "Disconnect", the message "Changing monitoring state from 'Operational' to 'Offline'" is posted immediately.  There is then a long delay (equal to the "Communications Timeout" setting) before the connection is closed and the "Connection closed, closing down monitor" message is posted.  Changing the "Communications Timeout" value changes the delay to that value.

foosel, 2018-02-08:
Ah... It seems to be OS specific. I can't reproduce from my (windows based) development machine, but can reproduce from a Pi.

kantlivelong, 2018-02-08:
Confirmed here now too. The UI indicates that disconnection has occurred but the log has the delay. 

>2018-02-08 09:09:37,684 - Changing monitoring state from 'Operational' to 'Offline'
2018-02-08 09:10:07,421 - Connection closed, closing down monitor

Running OctoPi (OctoPrint maint) w/ Smoothieware

foosel, 2018-02-08:
So I just pushed something that is fixing this at least on my end to `maintenance`. But it requires an update of PySerial and applying this update appears to cause issues in some setup scenarios that I currently don't yet understand.

foosel, 2018-02-08:
I've reverted the PySerial update (and hence deactivated the fix in 98b71a5) for now, since I currently can't make head nor tails of the update errors I'm seeing. Seems to be related to PySerial 2.7 being installed through an earlier `python setup.py install` triggered for OctoPrint, in which case an update fails on `pip install` with a weird error:

```
Installing collected packages: pyserial, OctoPrint
  Found existing installation: pyserial 2.7
    Uninstalling pyserial-2.7:
      Successfully uninstalled pyserial-2.7
  Rolling back uninstall of pyserial
Exception:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pip/req/req_set.py", line 784, in install
    **kwargs
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pip/req/req_install.py", line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pip/req/req_install.py", line 1064, in move_wheel_files
    isolated=self.isolated,
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pip/wheel.py", line 247, in move_wheel_files
    prefix=prefix,
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pip/locations.py", line 140, in distutils_scheme
    d = Distribution(dist_args)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/setuptools/dist.py", line 263, in __init__
    for ep in pkg_resources.iter_entry_points('distutils.setup_keywords'):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 521, in iter_entry_points
    entries = dist.get_entry_map(group)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 2484, in get_entry_map
    self._get_metadata('entry_points.txt'), self
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 2257, in parse_map
    for group, lines in data:
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 2841, in split_sections
    for line in yield_lines(s):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 2078, in yield_lines
    for ss in strs:
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 2414, in _get_metadata
    if self.has_metadata(name):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 1399, in has_metadata
    return self.egg_info and self._has(self._fn(self.egg_info, name))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 1759, in _has
    return zip_path in self.zipinfo or zip_path in self._index()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 1639, in zipinfo
    return self._zip_manifests.load(self.loader.archive)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 1551, in build
    with ContextualZipFile(path) as zfile:
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pkg_resources.py", line 1600, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 756, in __init__
    self.fp = open(file, modeDict[mode])
IOError: [Errno 2] No such file or directory: '/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/pyserial-2.7-py2.7.egg'
```

Will have to investigate this further, but until I have a solution for this problem this issue - albeit solved in theory - will stay unfixed.

foosel, 2018-02-12:
So it looks like this is an issue with pip & setuptools updating packages that so far were installed as Python eggs. See 8ad0aadb52b9ef354cad1b33bd4882ae2fbdb8d6 for a bit more context.

What appears to solve it is simply running another install right after the first failed one, however OctoPrint 1.3.6 isn't yet equipped to detect this scenario and handle it. OctoPrint 1.3.7 however will be thanks to the above commit (and any versions prior to 1.3.6 won't run into this problem due to a different update procedure) and so I'll update the pyserial version in 1.3.8 with a big disclaimer that if one is updating from 1.3.6 they should manually update pyserial.

Part one of the fix for this bug will roll out with 1.3.7 though, I've adjusted the requirements for pyserial to include 3.4, which allows to cancel any read from the serial line in case of a disconnect, which I utilize in 98b71a5. You'll be able to manually update pyserial through `pip install "pyserial>=3.4,<3.5"` against 1.3.7 and utilize this fix already. 1.3.8 will then perform the second step and enforce this newer pyserial version in general.

In any case, setting milestone to 1.3.8, with intermediary workaround present already in 1.3.7.

ScottWell1, 2018-02-12:
@foosel - Fantastic...  Always appreciate your methodical, comprehensive approach that deals with entire problems end-to-end instead of just the immediate symptom.  That is what makes Octoprint so solid and reliable.  Thank you!

foosel, 2018-04-13:
Quick update on this: anything I said before about OctoPrint 1.3.8 now applies to OctoPrint **1.3.9**, 1.3.8 was just turned into a hotfix release.

Also I just noticed this accidentally got closed by one of the commits that is preparing for the actual fix. Reopening.

ScottWell1, 2018-04-14:
Understood -- and thanks again for the fix ("part 1") in 1.3.7; I manually upgraded pyserial and the delay is gone!   :-)

foosel, 2018-07-25:
1.3.9 has just been released.

commit 98b71a58 (2018-02-08): Tell PySerial to cancel reads & writes on connection close

Only done if the methods needed for that exist and are callable though
(to prevent issues with custom serial port implementations provided
through hooks, or operating systems that don't have support for
cancel_read/cancel_write in PySerial.

Fixes #2333 (at least as far as we'll ever be able to fix it)
closed 2018-07-25 by foosel