OctoPrint/OctoPrint #2334: Can´t  install any plugins after 1.3.6 update

Ppanda1337, 2018-01-02:
#### What were you doing?

I updated to 1.3.6 via the update announcement. After the update I wanted to install some of the plugins that needed updates for 1.3.6. However none of the plugins will install. It seems not to be the issue described [here](https://octoprint.org/blog/2017/11/01/octopi-0.12-pip-issues/).

Steps to reproduce the Problem:

1. Installed the update to 1.3.6
2. Did the setup wizard
3. Tried to install a plugin

#### What did you expect to happen?

The selected plugin should install.

#### What happened instead?

Installation returned with an error 

#### Did the same happen when running OctoPrint in safe mode?

I need to check this but trying to run anything with the pip installer seems to throw an error

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

OctoPi 0.13

#### Printer model & used firmware incl. version

self build around the Smoothieboard V1

#### Browser and Version of Browser, Operating System running Browser

Tested on Chromium and Chrome (Version 63.0.3239.84 (Offizieller Build) (64-Bit))
#### Link to octoprint.log

[Octoprint Log](https://pastebin.com/Lu6hUCZY)
[Pluginmanager Log](https://pastebin.com/YcspKRch)

#### Link to contents of terminal tab or serial.log

I am not even connected to the printer yet.

#### Link to contents of Javascript console in the browser

https://pastebin.com/6wALyPMN

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

#### Output of Terminal when trying to run the PIP fix

[Console Log](https://pastebin.com/SyyLaxnk)

kantlivelong, 2018-01-02:
Can you SSH into OctoPi, run `locale`, and come back with the results?

Ppanda1337, 2018-01-02:
https://pastebin.com/w26T0RTS

here you go

kantlivelong, 2018-01-02:
If you run `locale -a` do you see **en_GB.UTF-8** listed?

Ppanda1337, 2018-01-02:
no I don´t see it here is the output

pi@Anteater:~ $ locale -a
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_COLLATE to default locale: No such file or directory
C
C.UTF-8
POSIX
pi@Anteater:~ $

kantlivelong, 2018-01-02:
1. Run `sudo dpkg-reconfigure locales` and select the locales you want to use.
2. Run `locale -a` and confirm your locale was added.
2. Restart OctoPrint using `sudo service octoprint restart` and try installing the plugin again.

Ppanda1337, 2018-01-03:
Thanks a lot for the help! This fixed the problem.

closed 2018-01-03 by Ppanda1337