OctoPrint/OctoPrint #2335: [Request] Monitoring of prints started manually

atoktoto, 2018-01-02:
Currently Octoprint GUI does not show correct state of the printer if the print was started manually using printer controller (and printing from SD card). This might cause a conflict if someone tries to start a print while the printer is already in progress. 

It this a limitation coming from printer firmware / g-code limitations?
I'm using Prusa i3 MK2.

foosel, 2018-02-19:
The thing is, OctoPrint doesn't get any info from the firmware when a print job is started directly via the controller or something. It would need to constantly poll the firmware to "report SD print status" via `M27`" while idle to do this. 

I'm not sure if people would like the noise in the terminal that produces while idle (especially combined with the `M105` polls when the firmware isn't detected to support temperature auto reporting). But it's something that could be looked into.

marcboivin, 2018-02-21:
Maybe an event from the firmware would make this feasible in a clean manner

foosel, 2018-02-21:
Indeed.

teemuatlut, 2018-02-21:
Would this be as simple as
```cpp
if (card.sdprinting == true & ELAPSED(10s)) { enqueue("M27") }
```
Basically just automate sending the sd print status if a print is in progress.

marcboivin, 2018-02-21:
https://github.com/MarlinFirmware/Marlin/issues/9636

marcboivin, 2018-02-21:
@teemuatlut or offer the user a button to trigger the manual update.

marcboivin, 2018-02-22:
There is an experimental branch available to test. Can't test it myself until sunday. 

https://github.com/MarlinFirmware/Marlin/issues/9751#issuecomment-367813091

TheSFReader, 2018-02-26:
The pull request was merged, and the feature should be enable-able on Marlin 2.0 bugfix builds. (depending on the configuration. M115 will report whether the capability is activated with the following tag : AUTOREPORT_SD_STATUS )

marcboivin, 2018-03-01:
@foosel Would you see this as a core feature or a plugin?

foosel, 2018-03-01:
Core feature. I actually also have something semi working, still need to iron out the kinks though.

TheSFReader, 2018-03-01:
@foosel, please note that I've also fixed a bug in the SD status report that made it return a number of bytes even when not printing. With the correction, it'll return MSG_SD_NOT_PRINTING ("Not SD printing") instead. I suppose the regexp to filter sd_status should be adjusted accordingly. 

This fix should not be dependant on the enabling/disabling of the AUTOREPORT_SD_STATUS feature.

foosel, 2018-03-01:
@TheSFReader `Not SD printing` has always been supported :) I also noticed the `0/0` style message recently and since that has been in a while I might have to support it too. Not that big of a problem though.

TheSFReader, 2018-03-01:
On my version, and since I applied the fix, I see the "NOT SD Printing" message in the terminal even if the filter is active. Not fluent with regexps, I added a new filtering rule , but Wonder how it could be combined in the default one

foosel, 2018-03-01:
Sorry, didn't realize you were referring to the terminal filters. You are right, those don't include that yet. There's also some internal processing of that message inside the comm layer that detects that and considers it as an indication that something went wrong while SD printing, switching states accordingly.

foosel, 2018-03-07:
Thanks to 52bb94e272bc3541a8adf315ee9b94949658bcd1 and 98c7ad0c2045d1a962c65097a61912cbc54a7d47 auto detection of SD prints started from the controller *while connected to OctoPrint* should now work *IF the following requirements are fulfilled*:

> * Firmware must send a "File opened: ..." message on start of the print
> * Firmware must respond to an immediately sent M27 with `SD printing byte <current>/<total>`
> * Firmware must stay responsive during ongoing print to allow for regular M27 polls (or push those automatically) or M25 to pause/cancel the print through OctoPrint.

e6ec48b8a1221b4d77cf393767bd4295f0b3b409 implements support for sd status auto reporting.

bb61bbb2b49c5fcbebf3bd79495ba2654d0348bc extends the terminal filters for sd status filtering with the "Not SD printing" message.

All of that is available on `maintenance`, soon `devel`. Will be part of 1.3.7

foosel, 2018-04-09:
Closing as 1.3.7 was just released.

closed 2018-04-09 by foosel

goncalossilva, 2019-08-08:
For cases when Octoprint is restarted with an ongoing print (from SD card), is there a way to make it start tracking the print progress again?

I can see it by sending `M27` myself, but it would be great to have it monitored by Octoprint itself. I understand this use case breaks the 2 first conditions in the earlier comment, as the print had already started. What I'm wondering is if I, as a user, can _tell_ Octoprint to restart monitoring `M27`.

Esteban16899, 2019-10-24:
Hey! I'm sorry to bother, but looking around i having trouble on this subject.
My printer is missing this condition:
> > * Firmware must send a "File opened: ..." message on start of the print

I'm Running An Ender 3 with an SKR1.3 wich firmware I updated myself (Marlin Bugfix 2.0.x). AutoSDreport is enabled, and I see it working, but I haven´t found the way for the printer to send a "File Opened" notice whenever I start an SDPrint manually from the LCD.
Thank you!