OctoPrint/OctoPrint #2339: [Request] Insert code checking for Control and Terminal commands

demonskull, 2018-01-05:
Hi all,

I often use the Terminal tab to do some calibrations and tests.

Perhaps I have a noisy environment, or there's a loose wiring in my system, but sometimes my controller does receive a corrupted message from OctoPrint. For example, if I issue the command "G1 X230 F1000", the controller may lose the last zero and do instead "G1 X230 F100" - Other message corruption events have more undesirable effects...

I believe that commands sent through the terminal are sent without "CRC" checking. Would it be possible to have the server add the corresponding "CRC" code to a manual command in the Terminal?

I have also noticed the occasional code corruption when using the manual Control tab. Would it be possible to add the checking code to this tab as well?

Thanks in advance!

kantlivelong, 2018-01-05:
Checksums should be sent with any commands sent. Can you supply serial logs with the issue occuring?

demonskull, 2018-01-05:
Hi kantlivelong, thanks for your comment.

On the Octoprint Terminal I can see checksums for other commands, such as when the server initialises the printer. But when I send Gcode manually, the Terminal shows the code as I type it. If it includes the checksum in the code actually sent, it is not echoed to the Terminal.

I'm using RAMPS/marlin, and I don't think they keep a log of received commands (please correct me if I'm wrong). I'll see what I can do at the printer's end.

kantlivelong, 2018-01-05:
Hmm it appears that you are correct. I could have sworn it sent checksums with everything. We'll keep this request open and perhaps it will be implemented.

demonskull, 2018-01-05:
That's fine, thank you!

kantlivelong, 2018-01-05:
Hey @demonskull 

I found it! In order to enable you will need to uncheck "Enable automatic firmware detection" in Settings -> Features and an option called "Send a checksum with the command" will appear.

demonskull, 2018-01-08:
Hi @kantlivelong,

It works! Thank you very much!

closed 2018-01-08 by demonskull