OctoPrint/OctoPrint #2340: OctoPrint upgrade to 1.3.6 failed.

deejayspinz, 2018-01-05:
Just tried to upgrade to OctoPrint 1.3.6 and it  failed.  Logs attached.  When I tried to load it, it loads to an HTML only view (attached) of the interface, and it's unusable.  Is there any way to recover from this?  Reload 1.3.5?

[octoprint.log](https://github.com/foosel/OctoPrint/files/1606802/octoprint.log)

[capture](https://user-images.githubusercontent.com/1494332/34612800-3b99a182-f1f9-11e7-9e22-50fe65e9bf76.PNG)

kantlivelong, 2018-01-05:
Can you check for a **plugin_softwareupdate_console.log**  file and attach as well?

deejayspinz, 2018-01-05:
After trying to restart /power down to see if it came back up, I can no longer get in to OctoPrint to get the logs now.   All I get is the raw HTML view of the home screen (shown in the screenshot I provided).  Sorry.

kantlivelong, 2018-01-05:
You can SSH in or use SCP and pull the logs. They are located in /home/pi/.octoprint/logs

deejayspinz, 2018-01-05:
Thanks for the tip.  It was easier for me to start from scratch and re-download and install OctoPi/print.  Installed, upgraded no problem to 136.  Must be an incompatible plugin.  Issue is, I don’t use my printer a lot, so forget all the plugins I had installed.  At a min I had the one that displays the temp in the top bar.  Had 5-8 others installed, just cant remember names… sorry.  You can close the issue if you like.

kantlivelong, 2018-01-05:
Alright. Just so you are aware, there are a few methods of getting into safe mode. See [the doc](http://docs.octoprint.org/en/master/features/safemode.html).

closed 2018-01-05 by kantlivelong

deejayspinz, 2018-01-05:
Good info. Thanks. Will keep it in mind if it happens next time.  I turned on the new feature in 136 that blocks flagged incompatible plugins.  It’s a shame that the update process did not have a way to do something like turn off plugins, update, then some sort of test /validate and disable the bad ones.  Wordpress does something like this.  That way a simple plugin does not cause the grief I had.  I’m a simple user from the perspective of Linux/RPi.  I don’t like getting into SSH etc..  Just want it to work and have an easy safe mode option that I could hit from the browser without the need for a login.

 

Thanks again for your help.

 

Dave

kantlivelong, 2018-01-05:
Disabling the plugins before an upgrade may be a good idea. I'll bring it up.