OctoPrint/OctoPrint #2341: [Request] Show current layer number on state panel

dapug, 2018-01-05:
Seeing current layer number is as useful as seeing % in progress bar, and for the same reasons.

Important: when a model is too large to render in g-code viewer there is NO other way to see what layer you are on, and yet this is when it's even more valuable to monitor where you are in layers!  So this really warrants including it as part of the state and progress bar info.

Ideally, show layer number and total, i.e.  "21 of 50" for a 50 layer print.

ntoff, 2018-01-06:
There's a plugin for that https://plugins.octoprint.org/plugins/displayz/ and a short explanation of why that info isn't there by default.

dapug, 2018-01-08:
Lame. a deeper issue than I thought.  Seems there would need to be a new standard established in printer firmware/communication itself in order to do this right.

closed 2018-01-08 by dapug

chatrat12, 2018-07-17:
I created a plugin that does this, it should be in the plugin manager soon. 

https://github.com/chatrat12/layerdisplay