OctoPrint/OctoPrint #2343: support for display of the IP address assigned by the DNS to R-PI on a printer display

bubnikv, 2018-01-07:
Hello Gina.

I am with Prusa Research. We would like to display IP address assigned to the Raspberry PI by a DNS directly on the 3D printer display. We did that before with the Toshiba Flash Air as the Flash Air offers an API to read out some of the WiFi statistics.

http://www.toshiba-memory.com/cms/en/products/wireless-sd-cards/FlashAir/index.html

Would you please propose and implement an API for the printer to receive an IP address over the serial line to be displayed on the 3D printer display? Likely a broadcast of some new M code would be sufficient.

Thanks, 
Vojtech

jneilliii, 2018-01-07:
There are a couple of plugins that do this. The detailed progress one does as well as my ip on connect plugin. 

https://github.com/jneilliii/OctoPrint-ipOnConnect

bubnikv, 2018-01-07:
Thanks @jneilliii. It seems I am guilty of not reading the documentation :-)

> There are a couple of plugins that do this.

Would you please share with me the names of the other plugins? What is the "detailed progress" plugin?

Thanks,
Vojtech

bubnikv, 2018-01-07:
I think I found what I needed. Thanks.

closed 2018-01-07 by bubnikv