OctoPrint/OctoPrint #2344: Anycubic I3 Mega - Baudrate can't be detected and also not set manually

Max123459, 2018-01-07:
#### What were you doing?
I installed octoprint with the image, updated the system and attached a logitech webcam and connected the printer with the delivered USB- Cable.

#### What did you expect to happen?
The printer is detected and completly working with octoprint.

#### What happened instead?

The baudrate isn't detected and also not connecting with set it manually.

Webinterface and Webcam Image are working.

#### Version of OctoPrint

OctoPrint version : 1.3.6
OctoPi version    : 0.14.0

#### Operating System running OctoPrint
Octo Pi

#### Printer model & used firmware incl. version

AnyCubic I3 Mega, Standard Firmware.

#### Link to console output
https://pastebin.com/G8G0rADT

ntoff, 2018-01-07:
>The baudrate isn't detected and also not connecting with set it manually.

Is it manually set to the same baud as the printer? It needs to match the setting in the printer's firmware, you can't just set an arbitrary value in OctoPrint. If your printer's baud is set to 115200 then you need to set OctoPrint to that, if the printer is set to something else then OctoPrint also needs to match. The most common ones seem to be 115200 and 250000 so try those.

Max123459, 2018-01-08:
Yes, I‘ve tried all baudrates (manually)...

KevHunter, 2018-01-08:
Hi, 
I have the same problem. I installed OctoPrint on a pi3 and I can’t get a connection to the Anycubic i3Mega. I tried all Baudrates and nothing happened. I also tried different usb cable but it seems that the printer doesn’t answer correctly. I get only question marks as an answer of the printer.

SchizoDuckie, 2018-01-08:
Note: this is a direct clone of #2309
In that issue you can already find a new firmware for the i3 mega that at least makes comms possible.

On top of that there seems to be some protocol errors too that make the printer not wait for things like heating wait responses.

foosel, 2018-01-11:
Sounds like an issue with Anycubic's firmware/hardware/* they solved on their end as described in #2350. 

Apparently though they didn't solve ALL their problems. The day I get back from a vacation and DON'T have to read about some printer vendors breaking communication left and right will be something to celebrate...

closed 2018-01-11 by foosel