OctoPrint/OctoPrint #2349: NameError: name 'unicode

TheLux83, 2018-01-08:
#### What were you doing?

I've installed Octoprint on my Windows 10 Workstation in order to let it run as a Octoprint Server.
I've followed this instructions: https://github.com/foosel/OctoPrint/wiki/Setup-on-Windows

#### What did you expect to happen?

Succesfully installing Octoprint on my Windows 10 Machine

#### What happened instead?

When I try to start the server, the following error Message appears:
```
C:\Progs\OctoPrint\venv\Scripts>octoprint.exe serve
Traceback (most recent call last):
  File "C:\Progs\OctoPrint\venv\Scripts\octoprint-script.py", line 11, in <module>
    load_entry_point('OctoPrint==1.3.6', 'console_scripts', 'octoprint')()
  File "C:\Progs\OctoPrint\venv\lib\site-packages\octoprint-1.3.6-py3.5.egg\octoprint\__init__.py", line 567, in main from octoprint.cli import octo
  File "C:\Progs\OctoPrint\venv\lib\site-packages\octoprint-1.3.6-py3.5.egg\octoprint\cli\__init__.py", line 137, in <module>
    from .client import client_commands
  File "C:\Progs\OctoPrint\venv\lib\site-packages\octoprint-1.3.6-py3.5.egg\octoprint\cli\client.py", line 166, in <module>
    @click.option("--str", "-s", "str_params", multiple=True, nargs=2, type=click.Tuple([unicode, unicode]))            
NameError: name 'unicode' is not defined
```

#### Did the same happen when running OctoPrint in safe mode?

Yes. The same thing happens when I try to start it with safe mode switched on:
```
C:\Progs\OctoPrint\venv\Scripts>octoprint.exe serve --safe
Traceback (most recent call last):
  File "C:\Progs\OctoPrint\venv\Scripts\octoprint-script.py", line 11, in <module>
    load_entry_point('OctoPrint==1.3.6', 'console_scripts', 'octoprint')()
  File "C:\Progs\OctoPrint\venv\lib\site-packages\octoprint-1.3.6-py3.5.egg\octoprint\__init__.py", line 567, in main
    from octoprint.cli import octo
  File "C:\Progs\OctoPrint\venv\lib\site-packages\octoprint-1.3.6-py3.5.egg\octoprint\cli\__init__.py", line 137, in <module>
    from .client import client_commands
  File "C:\Progs\OctoPrint\venv\lib\site-packages\octoprint-1.3.6-py3.5.egg\octoprint\cli\client.py", line 166, in <module>
    @click.option("--str", "-s", "str_params", multiple=True, nargs=2, type=click.Tuple([unicode, unicode]))
NameError: name 'unicode' is not defined
```

#### Version of OctoPrint

I wasn't able to start the webserver, but it is the newest version from 08.01.2018

#### Operating System running OctoPrint

Windows 10 Pro 
Version: 1709 
OS-Build: 16299.125
#### Printer model & used firmware incl. version

No Printer used

#### Browser and Version of Browser, Operating System running Browser

No Browser used

#### Link to octoprint.log

I don't have a log file yet. Searched at the known places

#### Link to contents of terminal tab or serial.log

No terminal tab used.

#### Link to contents of Javascript console in the browser

No Browser used

#### Screenshot(s)/video(s) showing the problem:

I've included the Error Messages above.

I have read the FAQ.

TheLux83, 2018-01-08:
Silly me.
I was using the wrong Python Version. Had an newer Python Verison from the past. Sorry for the trouble!

kantlivelong, 2018-01-09:
No worries. Glad you solved it.

closed 2018-01-09 by kantlivelong