OctoPrint/OctoPrint #2350: Anycubic I3 Mega no connection to OctoPrint

KevHunter, 2018-01-08:
#### What were you doing?

I want to connect Octoprint on a Rasperry Pi3 with my Anycubic I3 Mega ( new 32bit Version ).
I tried every Option of Baudrate and usb. 

#### What did you expect to happen?

The Printer connects to the Octoprint Server.

#### What happened instead?

The Printer doesn´t connect. It is only answering with question marks.

#### Did the same happen when running OctoPrint in safe mode?

Yes same Problem.

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Anycubic I3 Mega ( new32bit Version ) V1.4ORC23/P3

#### Browser and Version of Browser, Operating System running Browser

Windows 10 Edge and Chrome newest Version.

#### Link to octoprint.log

<script src="https://gist.github.com/KevHunter/8bd790397efb355ed8fa8b90ce14fb25.js"></script>

#### Link to contents of terminal tab or serial.log

<script src="https://gist.github.com/KevHunter/db5f0c373e0f44d1be5c1d1f2ee5584f.js"></script>

I have read the FAQ.

ntoff, 2018-01-09:
https://github.com/foosel/OctoPrint/issues/2344
https://github.com/foosel/OctoPrint/issues/2309

foosel, 2018-01-11:
Duplicate of #2344. See #2309 for solution & follow-up issues caused by Anycubic doing their own thing.

closed 2018-01-11 by foosel