OctoPrint/OctoPrint #2351: Octoprint does not listen on IPv6

tgunr, 2018-01-10:
#### What were you doing?

Trying to connect to my Ipv6 Octoprint server http://octoprint.polymicro.net

#### What did you expect to happen?
I expected to be connected to my server.

#### What happened instead?
Browser hangs because Octoprint is only listening to IPv4.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint
Debian Stretch

#### Printer model & used firmware incl. version
NA

#### Browser and Version of Browser, Operating System running Browser
Any browser

[If applicable, always include if unsure.]

#### Link to octoprint.log
NA

#### Link to contents of terminal tab or serial.log
NA

#### Link to contents of Javascript console in the browser
NA

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-01-11:
Please provide _all_ requested information, at the very least your `octoprint.log`. Without that information this ticket cannot be looked into and will be closed. Thanks.

tgunr, 2018-01-11:
A log has nothing to do with setting up to listen on IPv6. In your initialization you are only setting up IPv4.  Note the missing :::5000 local address.

[octoprint]/home/davec (626): netstat -lnpt
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      -
tcp        0      0 127.0.0.1:6010          0.0.0.0:*               LISTEN      -
tcp        0      0 0.0.0.0:5000            0.0.0.0:*               LISTEN      -
tcp6       0      0 :::111                  :::*                    LISTEN      -
tcp6       0      0 :::22                   :::*                    LISTEN      -
tcp6       0      0 ::1:6010                :::*                    LISTEN      -

foosel, 2018-01-12:
A log would show me what address your server is setup to bind to, which would allow me to verify that I'm testing against the same setup.

kantlivelong, 2018-03-09:
Closing due to lack of information and inactivity.

closed 2018-03-09 by kantlivelong