OctoPrint/OctoPrint #2353: USB Cable Issue

wucumbra, 2018-01-10:
#### What were you doing?
try printing with my tarantula and a shorter USB cable

#### What did you expect to happen?
that a print ended

#### What happened instead?
printing stops randomly showing timeout errors: 
Unhandled communication error
There was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: Too many consecutive timeouts, printer still connected and alive?

Terminal:
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok
Send: N1 M190 S80*81
Recv:  T:28.32 /0.00 B:33.36 /80.00 @:0 B@:0 W:?
Recv:  T:28.61 /0.00 B:33.26 /80.00 @:0 B@:127
Recv:  T:28.61 /0.00 B:33.26 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.64 /0.00 B:33.33 /80.00 @:0 B@:127 W:?
Recv:  T:28.49 /0.00 B:33.33 /80.00 @:0 B@:127
Recv:  T:28.49 /0.00 B:33.33 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.27 /0.00 B:33.54 /80.00 @:0 B@:127 W:?
Recv:  T:28.38 /0.00 B:33.62 /80.00 @:0 B@:127
Recv:  T:28.41 /0.00 B:33.65 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.30 /0.00 B:33.91 /80.00 @:0 B@:127 W:?
Recv:  T:28.55 /0.00 B:33.83 /80.00 @:0 B@:127
Recv:  T:28.64 /0.00 B:33.88 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.44 /0.00 B:33.93 /80.00 @:0 B@:127 W:?
Recv:  T:28.49 /0.00 B:34.19 /80.00 @:0 B@:127
Recv:  T:28.49 /0.00 B:34.19 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.55 /0.00 B:34.53 /80.00 @:0 B@:127 W:?
Recv:  T:28.55 /0.00 B:34.69 /80.00 @:0 B@:127
Recv:  T:28.55 /0.00 B:34.69 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.49 /0.00 B:34.82 /80.00 @:0 B@:127 W:?
Recv:  T:28.47 /0.00 B:35.00 /80.00 @:0 B@:127
Recv:  T:28.47 /0.00 B:35.00 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.47 /0.00 B:35.31 /80.00 @:0 B@:127 W:?
Recv:  T:28.49 /0.00 B:35.40 /80.00 @:0 B@:127
Recv:  T:28.58 /0.00 B:35.42 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.47 /0.00 B:35.63 /80.00 @:0 B@:127 W:?
Recv:  T:28.30 /0.00 B:36.10 /80.00 @:0 B@:127
Recv:  T:28.30 /0.00 B:36.10 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.47 /0.00 B:36.10 /80.00 @:0 B@:127 W:?
Recv:  T:28.49 /0.00 B:36.40 /80.00 @:0 B@:127
Recv:  T:28.49 /0.00 B:36.40 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.58 /0.00 B:36.50 /80.00 @:0 B@:127 W:?
Recv:  T:28.49 /0.00 B:36.73 /80.00 @:0 B@:127
Recv:  T:28.49 /0.00 B:36.73 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.44 /0.00 B:36.90 /80.00 @:0 B@:127 W:?
Recv:  T:28.38 /0.00 B:37.29 /80.00 @:0 B@:127
Recv:  T:28.38 /0.00 B:37.29 /80.00 @:0 B@:127 W:?
Recv: echo:busy: processing
Recv:  T:28.24 /0.00 B:37.58 /80.00 @:0 B@:127 W:?
No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from 'Printing' to 'Offline: Too many consecutive timeouts, printer still connected and alive?'
Connection closed, closing down monitor

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.14.0

#### Operating System running OctoPrint
Raspberry Pi 3 Model B

#### Printer model & used firmware incl. version

Tevo Tarantula/ Marlin 2.0

#### Browser and Version of Browser, Operating System running Browser

Safari / Mac OS High Sierra Version 10.13.2 (17C88)

#### Link to octoprint.log

https://gist.github.com/wucumbra/3b31436b737a8cc77426f986fb33ec6b

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

kantlivelong, 2018-01-11:
Hi,

I'm a little confused as it sounds as if your title answers the ticket...? If this is a USB cable issue what would OctoPrint be able to do about it?

wucumbra, 2018-01-11:
is a problem related to octoprint, because this problem I have at 90% only with octoprint, and not with astroprint or repetier Server. and I have no idea what it depends on ... if i need to change something in the firmware or octoprint ... or if it's just a bug

foosel, 2018-01-11:
Be sure to read and rule out *everything* listed in [this](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-keeps-running-into-communication-errors-and-timeouts) and [this](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-randomly-loses-connection-to-the-printer-with-a-serialexception) FAQ entry.

The most common cause of issues like this is an insufficient on unreliable power supply on the Pi.

foosel, 2018-02-12:
@wucumbra have you checked everything listed in the FAQ entries I pointed out?

wucumbra, 2018-02-12:
yesterday I got the new mainboard, now everything works fine, the problem was the mainboard usb data transmission

foosel, 2018-02-12:
Closing then since obviously not an OctoPrint issue.

closed 2018-02-12 by foosel