OctoPrint/OctoPrint #2355: File sorting does not update / show correctly when using by Upload date (descending)

hoegge, 2018-01-11:
#### What were you doing?

1. Set file sorting to "Upload date (descending)
2. Upload a new file
3. It will not be shown at the top

#### What did you expect to happen?
Last uploaded file first

#### What happened instead?
Last uploaded file second.
Trying to resort or change sort order or reload page does not correct it

#### Did the same happen when running OctoPrint in safe mode?
Have not tried

#### Version of OctoPrint
Version 1.3.6

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
not relevant

#### Browser and Version of Browser, Operating System running Browser
Chrome PC latest

#### Link to octoprint.log

N/A

#### Screenshot(s)/video(s) showing the problem:

[image]

foosel, 2018-01-11:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

hoegge, 2018-01-11:
@foosel It "can" be looked at but maybe you won't ;-) Not all the information is relevant (terminal, printer type) for a file upload and display problem or? I have added Octoprint log and Chrome console data.

foosel, 2018-01-11:
It can be looked at if it's the only ticket. It isn't, and running after information simply doesn't scale from a certain point, sorry. By providing all requested information you make sure your problem can be tackled in a timely manner, without endless round trips to request additional information. Isn't that worth the minimum overhead of collecting some logs?

ScottWell1, 2018-01-12:
FYI...   I have seen this problem when I manually (via SCP) created a folder in .octoprint/uploads and moved some of the files from .octoprint/uploads into the new folder.  After doing that, new files uploaded did not sort correctly.  I *suspect* it has something to do with the .metadata.yaml file being out-of-sync with the folder content after the manual file/folder changes.  Deleting .metadata.yaml and letting it be regenerated fixed the problem for me.

hoegge, 2018-01-29:
What information is missing? Would be smart if the automated mail said what was missing, I can't see what. Octoprint does often not sort correctly by time and I now have the same problem on a new Octopi/Octoprint, so probably not a one off.

foosel, 2018-01-29:
No information is missing since the OP added what was missing (logs and js console). I so far have not been able to reproduce this though, so any additional information regarding reproduction steps will be helpful.

foosel, 2018-02-26:
I still cannot reproduce this following the above steps. Also not via external modification of the `uploads` folder as indicated by @ScottWell1 (which - btw - is completely unsupported).

If anyone happens to run into this and can provide reliable reproduction steps, I'm happy to look into this further, but for now I'm marking it as unreproduced.

hoegge, 2018-02-26:
Ok, will try to see if I can find a way to reproduce it. E.g. my list right now looks like this:
[image]

[image]

hoegge, 2018-03-05:
BTW, does the color of the filenames have any influence? Sometimes some are red, sometimes some are green and sometimes black. Could not find the reason in the documentation.

arhi, 2018-03-05:
@hoegge red ones failed to print, green one printed successfully, black ones you did not try to print yet

BETLOG, 2018-04-25:
And again
[image]
[image]
[image]

foosel, 2018-04-25:
I *still* haven't been able to reproduce this myself. Does this also happen in safe mode?

BETLOG, 2018-04-26:
With a file list that is clear of all but one file, I cannot reproduce it in safe or normal mode.
But unlike my previous experience of the issue I do not have files in the list that are more then a day old to compare. I will avoid removing entries until tomorrow, and retry.

foosel, 2018-05-05:
@BETLOG anything new on that test?

BETLOG, 2018-05-06:
@foosel unfortunately not. Not noticing that I had inverted the connector on a piezo pcb has progressively fried two of my 3 arduinos regulators. So my printer has been completely inactive since our last posts. I was however not able to duplicate the issue even when attempting to do so.

foosel, 2018-05-08:
@BETLOG sorry to hear about the fried electronics :/ But good to know that you haven't been able to reproduce it so far anymore, I was starting to think I was doing something excessively differently than all of you.

BETLOG, 2018-05-08:
Hahah. Such is the nature of these things :)
I have been printing and sending/deleting small test files all day, with some being several weeks old, but so far I have not reproduced it. ...Or at least not like last time; when the topmost file was not the expected one. I have not been paying particular attention to the precise order of the later files, but superficially I'm pretty sure nothing is out of place.
I don't get it, either.

adrianmihalko, 2018-05-12:
I have similar problem. I set Sorty by upload date, but this setting is then not saved. Is there any trick to make it work?

BETLOG, 2018-05-12:
The trick is to figure out how to reproduce it, or why it's happening, so Gina can fix it. :)

Try deleting a single older file.

BETLOG, 2018-05-14:
Well, I managed to duplicate this issue a few days ago, but I have no idea how or why. I was just queuing a file and it wasn't at the top of the list. I have left it there, and deleted a few files under it, but everything I queue ends up in the correct chronological sequence... under a file that is several days old. Very strange. So it therefore it looks like my glib suggestion of "delete a file" should be "delete the out of place/top file".

BETLOG, 2018-05-14:
[OctoPrint-issues-2355.txt](https://github.com/foosel/OctoPrint/files/2002832/OctoPrint-issues-2355.txt)

BETLOG, 2018-05-15:
And as if by magic.. it works again....
All I did was queue a bunch of stl's dragged into the same Slic3r scene and send to printer.
Note that the file that was previously incorrectly on top of the list, is now still in the same sequence relative to the previous files, but the newly added file is (correctly) on top of them all.
[image]

luke321, 2018-05-15:
I have the same problem over and over again, suddenly files that I uploaded some days ago get to the top of my sort by upload date file list:

[image]

I will also try to figure out what's going on to help reproduce the bug, what I can say:

- I uploaded the files via REST API call
- The file on top was printed 3-4 times
- The files in the wrong order were only printed once or not at all

Maybe printing a file messes with the internal upload date? Or upload time is used over the date for sorting?

BETLOG, 2018-05-15:
This time the file I queued did not go to the top of the list...But rather quite a long way down the list.

Also, *possibly* related: for no apparent reason a print in progress and at about Z=1mm  suddenly started gouging the bed with the nozzle. As if the Z nemas had been inverted part way through a print. I've never seen anything like that before. Maybe this and the mis-sorting is a symptom of some kind of sdcard corruption on the pi? I have had some very unusual yet subtle behaviour because of that in the past. Seems unlikely though. For now I'm going to assume it's a separate issue, unrelated to octoprint.

[image]

[OctoPrint-issue2355__.metadata.yaml_2018-05-15--19-11-32.txt](https://github.com/foosel/OctoPrint/files/2004203/OctoPrint-issue2355__.metadata.yaml_2018-05-15--19-11-32.txt)
[OctoPrint-issue2355__octoprint.log_2018-05-15--19-11-32.txt](https://github.com/foosel/OctoPrint/files/2004204/OctoPrint-issue2355__octoprint.log_2018-05-15--19-11-32.txt)
[OctoPrint-issue2355__uploads_2018-05-15--19-11-32.txt](https://github.com/foosel/OctoPrint/files/2004205/OctoPrint-issue2355__uploads_2018-05-15--19-11-32.txt)

foosel, 2018-05-15:
> @foosel -- What other data, tests, or logging might be useful?

Since the upload date is basically the creation/modification date as attached to the file on disk (the younger of the two IIRC), the output of `ls -la ~/.octoprint/uploads` would indeed be the most helpful here. If the data there is ok, I'm leaning towards some strange caching issues causing this, since the only place that data is stored is in fact on the filesystem metadata and the cache - `metadata.yaml` isn't involved in any way here, not is any kind of internal OctoPrint logic.

John-Mc, 2018-05-15:
@foosel - Could the problem be related to **how** the file gets to the uploads folder? I know you've mentioned in the past not to manually place files in the Uploads folder and not to Samba share the Uploads folder (we should create a Watched Folder and upload to that instead), since bypassing OctoPrint's normal method for getting files into the uploads folder caused some necessary "housekeeping duties" to be skipped.

foosel, 2018-05-15:
@John-Mc Good idea, but that necessary house keeping doesn't involve the upload date. That's really just taken from the file metadata itself. Whatever `ls -la` shows should also be what OctoPrint sees. What also greatly confuses me is that it almost looks on those screenshots as if the dates are actually correct, it's just the sorting (which happens exclusively in the browser) that's off. Which I don't get because what happens there is the timestamps as retrieved from the backend (which in turn gets them from the file system) are simply sorted numerically by simple JS sorting functions, that should also not fail...

Long story short, this greatly confuses me. What might be interesting on top of an output of `ls -la` would actually be a full blown response body for the file list request from the server, which should be available in the network console (F12 at least in Firefox & Chrome, then switch to the "Network" tab):

[image]

@luke321 could you provide the above reponse headers and response body?

luke321, 2018-05-15:
@foosel No problem:

Response headers:

`
HTTP/1.1 304 NOT MODIFIED
Cache-Control: max-age=0
X-Clacks-Overhead: GNU Terry Pratchett
Set-Cookie: session_P80=.eJxNjUELgjAcxb9K_M8SUh1i4M0cBVsUie0k1iYbThM3MSd-98QuOz0e7_3emyAvO2EkoLLQRgSQKw5ogs0LENA63VGchgSfHHEXyZyUNCZ7lp2HRQ-sZl_idHWNSQRzAIqLxio7boveytyOrQDU9Fp7ibf-xtVK9UZ06yu0Cv7WCGPUp_Hb90fiWHgcsmdC-S1ayPkH2aM9kA.DdyIYQ.WBVlgiO0xzmcRdOLFu77IXlD6H8; Path=/; HttpOnly
X-Robots-Tag: noindex, nofollow, noimageindex
`

response body:

`
{"files": [{"date": 1526378199, "display": "MK3_akkus.gcode", "gcodeAnalysis": {"dimensions": {"depth": 137.13, "height": 4.799, "width": 164.275}, "estimatedPrintTime": 9382.315677075881, "filament": {"tool0": {"length": 6135.963800001853, "volume": 0.0}}, "printingArea": {"maxX": 164.275, "maxY": 134.13, "maxZ": 4.799, "minX": 0.0, "minY": -3.0, "minZ": 0.0}}, "hash": "02c79c756faeae48159e815d3d50fdcd2d260a28", "name": "MK3_akkus.gcode", "origin": "local", "path": "MK3_akkus.gcode", "refs": {"download": "http://10.0.0.20/downloads/files/local/MK3_akkus.gcode", "resource": "http://10.0.0.20/api/files/local/MK3_akkus.gcode"}, "size": 1823665, "type": "machinecode", "typePath": ["machinecode", "gcode"]}, {"children": [{"date": 1520490332, "display": "pla_test.gcode", "gcodeAnalysis": {"dimensions": {"depth": 112.785, "height": 1.999, "width": 144.785}, "estimatedPrintTime": 523.4055771781871, "filament": {"tool0": {"length": 333.16309999999385, "volume": 0.0}}, "printingArea": {"maxX": 144.785, "maxY": 109.785, "maxZ": 1.999, "minX": 0.0, "minY": -3.0, "minZ": 0.0}}, "hash": "31d6a4a70521d86b86013431533ce76d0db29ca4", "name": "pla_test.gcode", "origin": "local", "path": "Calibs/pla_test.gcode", "prints": {"failure": 0, "last": {"date": 1520491042.95241, "printTime": 626.4699320793152, "success": true}, "success": 1}, "refs": {"download": "http://10.0.0.20/downloads/files/local/Calibs/pla_test.gcode", "resource": "http://10.0.0.20/api/files/local/Calibs/pla_test.gcode"}, "size": 108304, "statistics": {"averagePrintTime": {"_default": 626.4699320793152}, "lastPrintTime": {"_default": 626.4699320793152}}, "type": "machinecode", "typePath": ["machinecode", "gcode"]}, {"date": 1519034572, "display": "prusa_ht_test.gcode", "gcodeAnalysis": {"dimensions": {"depth": 113.01, "height": 1.998, "width": 144.79}, "estimatedPrintTime": 271.4573236526969, "filament": {"tool0": {"length": 319.9139999999969, "volume": 0.0}}, "printingArea": {"maxX": 144.79, "maxY": 110.01, "maxZ": 1.998, "minX": 0.0, "minY": -3.0, "minZ": 0.0}}, "hash": "170c7332b9ce4dac88788bbd3f1bc62c9847adbe", "name": "prusa_ht_test.gcode", "origin": "local", "path": "Calibs/prusa_ht_test.gcode", "refs": {"download": "http://10.0.0.20/downloads/files/local/Calibs/prusa_ht_test.gcode", "resource": "http://10.0.0.20/api/files/local/Calibs/prusa_ht_test.gcode"}, "size": 66476, "type": "machinecode", "typePath": ["machinecode", "gcode"]}, {"date": 1525527602, "display": "MK3_Cleaner.gcode", "gcodeAnalysis": {"dimensions": {"depth": 0.0, "height": 0.0, "width": 0.0}, "estimatedPrintTime": 10.543088022903975, "filament": {"tool0": {"length": 50.0, "volume": 0.0}}, "printingArea": {"maxX": null, "maxY": null, "maxZ": null, "minX": null, "minY": null, "minZ": null}}, "hash": "bf085ffc120f44e88537aea2e859b0b12aa203f0", "name": "MK3_Cleaner.gcode", "origin": "local", "path": "Calibs/MK3_Cleaner.gcode", "prints": {"failure": 1, "last": {"date": 1526367568.851734, "printTime": 353.35611295700073, "success": true}, "success": 3}, "refs": {"download": "http://10.0.0.20/downloads/files/local/Calibs/MK3_Cleaner.gcode", "resource": "http://10.0.0.20/api/files/local/Calibs/MK3_Cleaner.gcode"}, "size": 166, "statistics": {"averagePrintTime": {"_default": 377.50834401448566}, "lastPrintTime": {"_default": 353.35611295700073}}, "type": "machinecode", "typePath": ["machinecode", "gcode"]}, {"date": 1519034572, "display": "prusa_petg_test.gcode", "gcodeAnalysis": {"dimensions": {"depth": 113.01, "height": 1.998, "width": 144.79}, "estimatedPrintTime": 271.4573236526969, "filament": {"tool0": {"length": 319.9139999999969, "volume": 0.0}}, "printingArea": {"maxX": 144.79, "maxY": 110.01, "maxZ": 1.998, "minX": 0.0, "minY": -3.0, "minZ": 0.0}}, "hash": "9d87913238e7bc883de1df3320812a6826d6545e", "name": "prusa_petg_test.gcode", "origin": "local", "path": "Calibs/prusa_petg_test.gcode", "prints": {"failure": 2, "last": {"date": 1521523968.762834, "printTime": 331.8189580440521, "success": true}, "success": 4}, "refs": {"download": "http://10.0.0.20/downloads/files/local/Calibs/prusa_petg_test.gcode", "resource": "http://10.0.0.20/api/files/local/Calibs/prusa_petg_test.gcode"}, "size": 66465, "statistics": {"averagePrintTime": {"_default": 342.99451518058777}, "lastPrintTime": {"_default": 331.8189580440521}}, "type": "machinecode", "typePath": ["machinecode", "gcode"]}], "display": "Calibs", "name": "Calibs", "origin": "local", "path": "Calibs", "refs": {"resource": "http://10.0.0.20/api/files/local/Calibs"}, "size": 241411, "type": "folder", "typePath": ["folder"]}, {"date": 1526318877, "display": "MK3_DACH.gcode", "gcodeAnalysis": {"dimensions": {"depth": 144.805, "height": 3.998, "width": 161.777}, "estimatedPrintTime": 2737.3659455981056, "filament": {"tool0": {"length": 3113.8162000000034, "volume": 0.0}}, "printingArea": {"maxX": 161.777, "maxY": 141.805, "maxZ": 3.998, "minX": 0.0, "minY": -3.0, "minZ": 0.0}}, "hash": "6a1b6af39a52543d883e19fba3b8b0ecd7cb813c", "name": "MK3_DACH.gcode", "or [... cut ...] G.gcode"}, "size": 1936701, "statistics": {"averagePrintTime": {"_default": 11664.364645719528}, "lastPrintTime": {"_default": 11663.639322042465}}, "type": "machinecode", "typePath": ["machinecode", "gcode"]}], "free": 13025038336, "total": 15232843776}
`

The order is still incorrect with MK3_akkus before MK3_OG in the file list:

[image]

But when I reorder the list by selecting order by name and than order by date I get to correct order, so the data from the response should be correct right?

foosel, 2018-05-16:
Yeah, it should, but now we know it *is*, which is better than *should* ;)

Ok, so this really smells like some kind of JS side sorting issue. I understand it only happens on refresh for you @luke321. What about the others? @BETLOG, @adrianmihalko, @ScottWell1, @hoegge can you fix it (until the next refresh) by switching to e.g. sort by name, then switching back to sort by upload date? I'm suspecting something being up with the local storage data (where the sorting preference is stored).

ScottWell1, 2018-05-16:
> What about the others? @BETLOG, @adrianmihalko, @ScottWell1, @hoegge can 
> you fix it (until the next refresh) by switching to e.g. sort by name, then switching back 
> to sort by upload date?

It has been a while since I had the problem but as best I recall....   

YES - switching sort to "by name" and then back to "by upload date" resulted in correct order, but after refreshing the "offending older file" was at the top again.   The date/time on the file itself was correct, and Octoprint correctly stated the "uploaded:" age of the file in the listing (as in examples posted by others above).

New files added (via watched folder) were inserted into the list below the older file; toggling sort order to name then back to descending date fixed it until a refresh or another file was added.

BETLOG, 2018-05-16:
@foosel No, toggling sort option did not rectify my list. I have attempted that twice, several weeks apart. Neither time had the desired effect.
A few days ago my list started getting difficult to navigate (my long filenames, fast scroll speed, and the tiny unresizable list frame being the main frustrations) and new files became difficult to find, so I deleted everything. But at that time there were several files incorrectly above the latest one. Previously I only had one file incorrectly at top, and everything else correctly sorted under that.

luke321, 2018-05-17:
@foosel Toggling the sort option did not always fix everything.
I had an instance which I describe in my first comment where toggling put the newest file at the bottom of my file list.

foosel, 2018-05-17:
Hm, too bad. If toggling had always fixed it (temporarily) I at least would have had a lead to go on to but now I'm completely in the dark again.

It sucks that I can't reproduce this myself, that would make debugging way way easier.

luke321, 2018-05-17:
I just tried Micrsoft EDGE and the order is correct.
So maybe the problem is Google Chrome specific?

foosel, 2018-05-17:
Should be solved by the above commit. Already on `maintenance`, soon `devel` and to be released with 1.3.9

commit 5456f5a7 (2018-05-17): Fix potential for invalid sorting of uploads

"date" and "size" might be undefined on both parameters (e.g. for
folders, SD files). We have no guarantee that "a" will always have
either defined, so make sure to test or a wrong sorting order might be
the consequence.

Fixes #2355
closed 2018-07-25 by foosel