OctoPrint/OctoPrint #2356: [Not An Issue] Letting you know about the serial protocol error handling implemented in LulzBot Cura 2

marcio-ao, 2018-01-11:
Hello Gina,

I heard that you were working on improving the serial coms in OctoPrint. I thought you might be interested in knowing that I wrote a Python class that is used to implement GCODE error handling and re-transmission using Marlin's new ADVANCED_OK feature. It seems robust to all sorts of errors. If you want to look at that module, you can find it here:

https://code.alephobjects.com/source/Cura2/browse/master/plugins/USBPrinting/MarlinSerialProtocol.py

The class essentially acts as a wrapper between the serial port and upstream code. On the sending side, it computes checksums, keeps a queue of commands for transmission and makes sure the Marlin buffer is not over-filled. On the receiving side, it takes care of listening to responses from Marlin and uses Marlin's ADVANCED_OK feature to keep track of how much space is available in the Marlin buffer, it automatically resends commands when requested by Marlin so the upstream code does not need to do any of that. It also has some support for emergency commands that need to bypass the normal buffering.

Anyhow, if this is of any use to you, feel free to use it.

-- Marcio

foosel, 2018-01-11:
@marcio-ao took the liberty to turn this into a brainstorming ticket (fit the bill the most). Thanks for the code link, I also had a user get in touch on an issue he's currently seeing where an `ok` gets stuck on the tail of an automatic temperature report and he mentioned your advanced ok tracking as well.

I've just returned from an overdue vacation and am currently wading through a mountain of mails, but I might get back to you with some questions once I've had a chance to look at this, would that be ok?

marcio-ao, 2018-01-11:
@foosel: Thanks for the super speedy reply :) Sure, I'd be glad to answer any questions you may have, just let me know. I hope it can be of some use; if it's not, no problem. I just wanted you to be aware of it :)

fieldOfView, 2020-01-06:
Unfortunately code.alephobjects.com is down and it does not seem to be coming back up any time soon after the recent events at Lulzbot. Fortunately I pulled their git repo for Cura 2 LE, so here is the referenced file in case someone is interested:

https://github.com/fieldOfView/Cura/blob/lulzbot-master/plugins/USBPrinting/MarlinSerialProtocol.py