OctoPrint/OctoPrint #2357: Plugin Manager - The pip command could not be found

purist69, 2018-01-12:
I have found a few related issues, but none seem to solve my problem.  I have a fresh install in Lubuntu and when I go to the plugin manager it says pip could not be found, but pip IS at ~/OctoPrint/venv/bin so I'm not sure why it's complaining.

Any help would be appreciated.

purist69, 2018-01-12:
#### What were you doing?

Trying to install plugins on a fresh install of OctoPrint 1.3.6

Ideally provide exact steps to follow in order to reproduce your problem:

Go to Plugin mananger and see message:
The pip command could not be found or does not work correctly for this installation of OctoPrint - please consult the log file for details and if necessary configure it manually. No installation and uninstallation of plugin packages is possible while pip is unavailable.

#### What did you expect to happen?
Install plugins
#### What happened instead?
Can't install plugins
#### Did the same happen when running OctoPrint in safe mode?
Yes
#### Version of OctoPrint
1.3.6
#### Operating System running OctoPrint
Lubuntu 16.04 kernel 4.4.49
#### Printer model & used firmware incl. version
not relevant
[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Version 63.0.3239.132 (Official Build) (64-bit) on Windows 10
[If applicable, always include if unsure.]

#### Link to octoprint.log
2018-01-12 02:57:33,610 - octoprint.startup - INFO - ******************************************************************************
2018-01-12 02:57:33,611 - octoprint.startup - INFO - Starting OctoPrint 1.3.6
2018-01-12 02:57:33,611 - octoprint.startup - INFO - ******************************************************************************
2018-01-12 02:57:33,677 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-01-12 02:57:35,098 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-01-12 02:57:35,493 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2018-01-12 02:57:35,501 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2018-01-12 02:57:35,653 - octoprint.server - INFO - Intermediary server started
2018-01-12 02:57:35,654 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-01-12 02:57:35,656 - octoprint.plugin.core - WARNING - Plugin "OctoPi Support Plugin" did not pass check
2018-01-12 02:57:35,896 - octoprint.plugin.core - INFO - Found 7 plugin(s) providing 7 mixin implementations, 4 hook handlers
2018-01-12 02:57:35,898 - octoprint.printer.profile - ERROR - Profile _default does not exist, creating _default again and setting it as default
2018-01-12 02:57:35,998 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-01-12 02:57:35,999 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-01-12 02:57:37,928 - octoprint.util.pip - INFO - Using "/home/pi/OctoPrint/venv/bin/python -m pip" as command to invoke pip
2018-01-12 02:57:39,818 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2018-01-12 02:57:39,820 - octoprint.util.pip - ERROR - Cannot use pip
2018-01-12 02:57:39,822 - octoprint.plugin.core - INFO - Initialized 7 plugin implementation(s)
2018-01-12 02:57:39,875 - octoprint.plugin.core - INFO - 7 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/discovery
|  Plugin Manager (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/virtual_printer
2018-01-12 02:57:41,393 - octoprint.environment - INFO - Detected environment is Python 2.7.12 under Linux (linux2). Details:
|  hardware:
|    cores: 8
|    freq: 1400.0
|    ram: 999739392
|  os:
|    id: linux
|    platform: linux2
|  python:
|    pip: 9.0.1
|    version: 2.7.12
|    virtualenv: /home/pi/OctoPrint/venv
2018-01-12 02:57:41,400 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-01-12 02:57:41,401 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2018-01-12 02:57:41,576 - octoprint.server - INFO - Shutting down intermediary server...
2018-01-12 02:57:41,863 - octoprint.server - INFO - Intermediary server shut down
[... 1063 lines cut ...]
2018-01-12 05:54:14,449 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2018-01-12 05:54:14,453 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
[... cut ...]
2018-01-12 05:55:40,403 - octoprint.server.api.system - INFO - Performing command for core:restart_safe: sudo service octoprint restart
2018-01-12 05:55:40,654 - octoprint.server.api.system - WARNING - Command for core:restart_safe failed with return code 1:
! STDOUT: 
! STDERR: sudo: no tty present and no askpass program specified
2018-01-12 05:59:49,189 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.116
2018-01-12 05:59:53,100 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.116

The rest of the requested details dont see relevant.

Thanks!

foosel, 2018-01-12:
Please also provide you JS console output (it might be relevant here after all)

purist69, 2018-01-12:
[image]

purist69, 2018-01-13:
I ran this, again...
./venv/bin/pip install pip --upgrade
and did an apt-get upgrade and it started working

closed 2018-01-13 by purist69