OctoPrint/OctoPrint #2359: [Request] Print time left - can we send some info to Octoprint from G-code itself?

arhi, 2018-01-13:
#### What were you doing?

Printing a large object that varies print speed significantly. For example first layer is printed at 8mm/sec, and number of other layers are printed from 12mm/sec till 120mm/sec

#### What did you expect to happen?

Octoprint to show progress percentage and time left accurately.

#### What happened instead?

The progress bar and time to print are out of mark by a lot. Example, if I have 10 layers cube and 1st layer is 10% of the speed of layers 2-10 on second layer we are not at 10% of the print but at 52%... if we have a cone the error is even greater.. 

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

armbian

#### Printer model & used firmware incl. version

smoothieware

#### Question

I have created a small gcodestat app https://github.com/arhi/gcodestat that I use as post-processor after s3d slicer, and it takes into account acceleration and junction deviation configuration options of the firmware + reads changes of acceleration and junction deviation from the g-code and calculates the ideal time to print. It cannot take into account the heatup times but movement times are properly calculated for smoothieware (I'll add marlin support and some others as soon as I can)... I also add every N % the M117 with % left and hh:mm:ss left to display (both on printer and as popup in octoprint) but what I'd like is to somehow "inform" the octoprint about this data... for e.g. add a g-code that will at the beginning and few other points in file give octoprint the info about "how much time left" and "what % we are at" .. something like 

M9999 P70 T56934767 ; 70% to go, 56934767seconds to go

I know how to intercept M9999 with a plugin but I don't see how from plugin to introduce this values into core octoprint to modify original time and progress bar calculation??

arhi, 2018-01-13:
The question might be trivial but I don't write Python so have some issues with figuring out how stuff links with each other inside octoprint :(

foosel, 2018-01-18:
What you want (replacing OctoPrint's own time and completion estimates with something determined by a plugin) is currently not possible.

jneilliii, 2018-05-06:
@foosel, it wouldn't be possible to use [set_additional_metadata](http://docs.octoprint.org/en/master/modules/filemanager.html?highlight=meta#octoprint.filemanager.storage.StorageInterface.set_additional_metadata) of the storage class?

arhi, 2018-05-06:
well the current implementation takes into account only bytes read and time spent till "now" and total bytes of  a g-code file.. it's not parsing file at all.. 
the only real solution would be to move _estimatePrintTimeLeft function
https://github.com/foosel/OctoPrint/blob/6d7f8a2a64b364504f11dafd5009a85fb785127a/src/octoprint/printer/standard.py#L769
out into the plugin together with TimeEstimationHelper class and allow bit more data flow between core and plugin.. so that way any number of plugins that calculate more/less accurate progress can be created... additional_metadata could be in theory used to read total print time of already printed part (that's not used now afaic) so it could be set in advance by a plugin (or a new property can be added) but that all looks more like a hack then a real solution... 

@triplepoint slic3r PE gives only accurate values if you don't limit acceleration settings on the printer, if you do limit acceleration/jerk it will be off, now it is one of the best yet as it does have control of the acceleration in the slicer itself so it will send this data to printer and use for calculation, if you do not limit it on printer and don't exceed maximums in the slicer itself it will be accurate for every printer.

[gcodestat](https://github.com/arhi/gcodestat) for e.g. takes the default values as input and reads the set values from g-code and calculates the time, and inserts M117 codes every X% to display progress report on the printer's lcd (and in the browser using m117 display message plugin for octoprint) but can't put that info back to octoprint itself :( ...

arhi, 2018-05-06:
@jneilliii yes I know about preprocessors, but what's the point when I can't push data into the octoprint :( ... (and I don't even need to run full preprocessor on the rpi as I already have postprocessor integrated into slicer ) ... now it would be needed to have bit more then preprocessor as the total time is from "when you start printing" not from when you "press print" as preheat time is not something one would like to guess ... so a code in the file "started printing" or "first G1 in the code" is required to set the "start" for the octoprint... 

but apart from the "total time" I'd like to "update octoprint" from g-code with "status" as I can embed 10%, 20%, 30%... done in the g-code itself (doing it already with M117, e.g. "M117 90% Remaining ( 03:09:21 )" ) so updating the main progress bar from g-code would be cool (I can easily add to processor to add there a M9117 90,03:09:21 that octoprint would parse like it's parsing M0 )...

eyal0, 2018-06-25:
@arhi I have an idea.

@foosel has made a hook so that anyone can add a PrintTimeEstimator. That estimator only gets some very basic info: the progress in the file as a percentage of bytes, the total time so far, and the total estimated print time. Currently, the print time estimation just assumes a linear relation between the percentage of bytes processed and the time to print. This may be inaccurate.

What if we used gcodestat or some other tool to preprocess the gcode file and determine a mapping between percentage of bytes complete and percentage of time elapsed? Then we store as metadata attached to the gcode file the mapping for 100 evenly spaced points in the file. For example, we might store that the first 10% of the file takes 15% of the time and the first 40% of the file takes 39% of the time. Those are just two points but we'd store 100, from 0% to 100%.

Then when we need to generate an estimate, just take the current progress and look it up in the map. A linear interpolation between points would be be good enough. That would get you a result within 1% already and usually better. The storage overhead would be 100 pairs of floats. We could trade between storage and accuracy by changing the number of points.

eyal0, 2018-06-25:
gcodestat looks really promising. I haven't looked at the code but I bet that it would be pretty easy to modify it so that it would output the mapping that I need. I could run that from OctoPrint. I would also put some comment into the gcode indicating that gcodestat has already been run on it so that the RPi could skip running gcodestat and instead grab that info from the file itself. I assume that running gcodestat on RPi is pretty slow. gcodestat could run asynchronously and niced so that it doesn't interfere with OctoPrint. The original time estimation algorithm would be used until there is enough metadata to switch to the gcodestat method.

arhi, 2018-06-25:
@eyal0 attm the https://github.com/arhi/OctoPrint-gcodestatEstimator reads the M117 codes from the gcode file put there by gcodestat and updates the octoprint with relevant details. if the M117 codes in format it understands are in the g-code file it will replace original estimator and use data from the g-code file (no need for any external pair of numbers, every X% of the time to finish gcodestat will embed info directly into gcode and this plugin will read that and use for estimation)... so with the estimator factory exposed for plugins @foosel gave enough for us to calculate the progress any way we want... 

now as for preprocessing gcodestat on "desktop" is super fast and it can be upgraded to perform some additional operations and transformations if needed. attm the most useful thing it does for me is send file to octoprint :D as s3d don't know how to do that itself...  since on rpi it takes some seconds to parse the file I kinda like to keep the gcodestat running on desktop with my slicer uploading the file to octoprint. attm I'm thinking about making a second plugin that will utilize the other hook @foosel introduced (octoprint-filemanager-analysis-factory) that will check if there's M117 100% line in the file and read the stats from there so that there's already metadata embedded when you upload the file - so processing is still on desktop... then, if time allows I might add an option to run gcodestat directly on the pi to process the file on the sod if file came without data but since this is something I personally will never use the priority is incredibly low as you can imagine..

as for gcodestat, it's super simple code, there's really not much you need to do there it's super basic g-code parsing and very simple time calculation based on current acceleration settings. it also uses junction deviation (so smoothieware and similar high end path finders use that) to tweak the last 1% accuracy :D ... unfortunately I don't know marlin that well so the jerk settings from marlin is ignored but that is under 1% error margin so I did not waste too much time with it..

closed 2018-07-01 by arhi