OctoPrint/OctoPrint #2361: Pyton setup fails on dependencies

VipSaran, 2018-01-13:
#### What were you doing?

"Basic setup" up to `./venv/bin/python setup.py install`.

#### What did you expect to happen?

To actually install dependencies listed un `setup.py` and finish the instalation.

#### What happened instead?

```
Installed /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg
Processing dependencies for OctoPrint==1.3.6
Searching for monotonic>=1.3,<1.4
Reading http://pypi.python.org/simple/monotonic/
Couldn't find index page for 'monotonic' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
No local packages or download links found for monotonic>=1.3,<1.4
error: Could not find suitable distribution for Requirement.parse('monotonic>=1.3,<1.4')
```

Please note that **HTTP** protocol is being used (instead of **HTTPS**).

#### Did the same happen when running OctoPrint in safe mode?

n/a

#### Version of OctoPrint

1.3.6

#### Operating System running OctoPrint

Linux (Raspbian)

#### Printer model & used firmware incl. version

n/a

#### Browser and Version of Browser, Operating System running Browser

n/a

#### Link to octoprint.log

n/a

#### Link to contents of terminal tab or serial.log

n/a

#### Link to contents of Javascript console in the browser

n/a

#### Screenshot(s)/video(s) showing the problem:

n/a

---

Solution (from `~/OctoPrint`):

```
sed -i s,http://pypi.python.org,https://pypi.python.org, venv/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/setuptools/command/easy_install.py
```

foosel, 2018-01-15:
This is not an issue with OctoPrint but with your environment and [even had its own FAQ item](https://github.com/foosel/OctoPrint/wiki/FAQ#i-got-some-error-after-attempting-to-install-monotonic-or-futures-during-update-and-now-my-server-wont-start) and [blog post](https://octoprint.org/blog/2017/11/01/octopi-0.12-pip-issues/).

closed 2018-01-15 by foosel

VipSaran, 2018-01-15:
My environment? Why the virtualenv is used then if not to set environment compatible with the install script?

foosel, 2018-01-15:
OctoPrint requires your python environment to be able to install packages. Your version of some python utilities however are so old that they still use http instead of https to access the python package index, and as explained in both links above there that was [disabled in October](https://mail.python.org/pipermail/distutils-sig/2017-October/031712.html). You need to update your environment. OctoPrint or OctoPrint's setup script can't do that for you. Virtual environments are used to allow the peaceful coexistence of various dependency versions on a system, but it can't fix issues with old utility versions.