OctoPrint/OctoPrint #2362: [Brainstorming] Add an option to increase snapshot timeout

emilio1625, 2018-01-14:
I'm currently using a smartphone to stream to octoprint using IP Webcam, when i test the snapshot url for the timelapse, the phone first focus then takes the picture, but the timeout is too short and it reports a failure, however when i put the url in the browser the picture is displayed, it takes some time, but is displayed.
I think it is convenient to have a timeout setting to allow slow cameras to take a picture and stream it to the snapshot url.
I have looked at the code, and found in timelapse.py the _perform_capture function, with a timeout parameter in a get request, but changing that doesn't seem to have any effect on the web interface, it keeps showing the same error message. I guess that the python files are precompiled and chached and that's why the change doesn't have any effect, or maybe i made the wrong change.
Any tips on where to look in the code to make this feature possible?

ntoff, 2018-01-14:
You could start with the documentation http://docs.octoprint.org/en/master/development/

and yes, the source code is built and stashed away elsewhere, so changing anything in the "src" directory won't affect octoprint if it's been built and installed.

Also if your phone takes longer than the interval between snapshots, you might run into issues, I'd say if the phone takes any more than 3 seconds (which is a LONG time to take a photo) then I'd ditch it and go with a faster solution if possible.

foosel, 2018-03-07:
I was adding a checkbox to disable SSL validation for the snapshot URL anyhow and adding an input for the timeout next to it as requested here was about the same amount of work. Implemented in the above commit, available on `maintenance`, soon `devel`, will be part of 1.3.7

foosel, 2018-04-09:
Closing as 1.3.7 was just released.

closed 2018-04-09 by foosel