OctoPrint/OctoPrint #2363: Octoprint losing connection to Prusa i3 MK2S since upgrading Octoprint to v1.3.6 and printer firmware to v3.1.0

tvi675, 2018-01-14:
#### What were you doing?

Printing a file that takes longer than 4-5 hours using Octoprint following updating Octoprint to v1.3.6 and the Original Prusa i3 MK2S to firmware version v3.1.0. Previous to updating this the Prusa was running v3.0.12 Final and Octoprint was v1.3.6 rc3, prints would run without issue. 

Since upgrading the printer firmware and Octoprint version, serial connection is lost on long prints causing print failures.

The USB cable is known good, and the issue occurs with other USB cables. The Pi power supply is known good, and the issue occurs with other known good power supplies.

Ideally provide exact steps to follow in order to reproduce your problem:

1 - Print a long file (doesn't matter what, multiple gcodes cause issue, including known previous good gcode files)
2 - Wait. A serial disconnect will occur.

#### What did you expect to happen?

Print to finish successfully.

#### What happened instead?

Serial disconnected. Print failed.

#### Did the same happen when running OctoPrint in safe mode?

Not yet tried.

#### Version of OctoPrint

1.3.6 Final

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Original Prusa i3 MK2S, firmware v3.1.0

#### Browser and Version of Browser, Operating System running Browser

n/a

#### Link to octoprint.log

https://gist.github.com/tvi675/a07d337ca8cae09d431ff8c8bbc59d62

#### Link to contents of terminal tab or serial.log

https://gist.github.com/tvi675/0df6095f6743a602133a05855057cfbe

#### Link to contents of Javascript console in the browser

n/a

#### Screenshot(s)/video(s) showing the problem:

n/a

I have read the FAQ.

token47, 2018-01-14:
Hello

I was about to open a new issue that seems like this one. So here is the indo:

- worked for months without issues on a manually installed octoprint on a ODROID / ubuntu 16.04, updated to the last version (1.3.6).
- last week swapped the hardware for a Raspberry Pi 3 and octopi 0.14, upgraded everything to the latest versions as of today (octoprint 1.3.6 and apt-get updated everything).
- since this change, having random stops on prints
- it just stops printing. No warning, no errors. Octoprint GUI goes on as if nothing has happened, thinks it's still printing
- the printer stops moving and stays in the last point it was, with the nozzle still over the last printing layer
- nothing wrong on the LCD display, temperatures are still maintaining (with occasional change in the temp value), rotary button still responsive, etc. In essence, the Marlin firmware seems to be running ok and just not receiving commands anymore.
- The serial.log shows that at that exact time the commands stop being sent, BUT IT IS STILL CONNECTED receiving temperature updates. The serial seems to be in a catatonic state because the RX is receiving status but the TX is not sending commands. And it seems that the octoprint does not want to send anything because they don't even get logged. If I click CANCEL on the interface, it "stops printing" and the joystick controls are enabled again. None of them will do anything, no command sent will do anything while in this state.
- If I click "disconnect" on octoprint it will disconnect (marlin still working ok), then I click "connect", marlin resets and everything is normal again (as I guess was expected).
- USB cable seems to be good, it's short, shielded and is the same I always used.

One thing to note is that there is no "ok" as a response of the last command sent. It sends "N25376 G1 X212.429 Y137.623 E0.4431*89" and receives "N25 T:240.23 /240.00 B:99.95 /100.00 @:46 B@:85", which seems wrong to me.

Below an excerpt or the serial.log at the exact time of freezing:

```
2018-01-14 16:32:37,794 - Send: N25367 G1 X210.169 Y133.851 E0.2619*91
2018-01-14 16:32:37,807 - Recv: ok
2018-01-14 16:32:37,810 - Send: N25368 G1 X210.243 Y134.351 E0.2820*87
2018-01-14 16:32:37,820 - Recv: ok
2018-01-14 16:32:37,823 - Send: N25369 G1 X210.365 Y134.840 E0.3021*80
2018-01-14 16:32:37,831 - Recv: ok
2018-01-14 16:32:37,834 - Send: N25370 G1 X210.535 Y135.315 E0.3223*81
2018-01-14 16:32:37,845 - Recv: ok
2018-01-14 16:32:37,849 - Send: N25371 G1 X210.751 Y135.772 E0.3424*84
2018-01-14 16:32:37,860 - Recv: ok
2018-01-14 16:32:37,864 - Send: N25372 G1 X211.011 Y136.204 E0.3626*82
2018-01-14 16:32:37,872 - Recv: ok
2018-01-14 16:32:37,876 - Send: N25373 G1 X211.311 Y136.610 E0.3827*94
2018-01-14 16:32:37,884 - Recv: ok
2018-01-14 16:32:37,888 - Send: N25374 G1 X211.650 Y136.984 E0.4029*90
2018-01-14 16:32:37,897 - Recv: ok
2018-01-14 16:32:37,901 - Send: N25375 G1 X212.023 Y137.323 E0.4230*86
2018-01-14 16:32:37,909 - Recv: ok
2018-01-14 16:32:37,912 - Send: N25376 G1 X212.429 Y137.623 E0.4431*89
2018-01-14 16:32:38,530 - Recv: N25 T:240.23 /240.00 B:99.95 /100.00 @:46 B@:85
2018-01-14 16:32:40,534 - Recv:  T:240.39 /240.00 B:100.08 /100.00 @:42 B@:53
2018-01-14 16:32:42,540 - Recv:  T:240.47 /240.00 B:100.05 /100.00 @:40 B@:73
2018-01-14 16:32:44,545 - Recv:  T:240.16 /240.00 B:100.12 /100.00 @:46 B@:61
2018-01-14 16:32:46,549 - Recv:  T:240.39 /240.00 B:100.19 /100.00 @:40 B@:48
2018-01-14 16:32:48,556 - Recv:  T:240.23 /240.00 B:100.17 /100.00 @:43 B@:58
2018-01-14 16:32:50,560 - Recv:  T:240.23 /240.00 B:100.12 /100.00 @:42 B@:77
2018-01-14 16:32:52,565 - Recv:  T:240.08 /240.00 B:100.15 /100.00 @:44 B@:65
2018-01-14 16:32:54,570 - Recv:  T:240.08 /240.00 B:100.14 /100.00 @:43 B@:66
2018-01-14 16:32:56,576 - Recv:  T:239.63 /240.00 B:100.19 /100.00 @:52 B@:56
2018-01-14 16:32:58,580 - Recv:  T:240.00 /240.00 B:100.22 /100.00 @:44 B@:51
2018-01-14 16:33:00,587 - Recv:  T:240.31 /240.00 B:100.30 /100.00 @:37 B@:35
2018-01-14 16:33:02,590 - Recv:  T:240.31 /240.00 B:100.26 /100.00 @:37 B@:50
2018-01-14 16:33:04,596 - Recv:  T:239.69 /240.00 B:100.27 /100.00 @:50 B@:51
2018-01-14 16:33:06,601 - Recv:  T:239.88 /240.00 B:100.31 /100.00 @:46 B@:39
2018-01-14 16:33:08,606 - Recv:  T:240.39 /240.00 B:100.29 /100.00 @:35 B@:47
```

token47, 2018-01-14:
It could be a Marlin issue. I opened a ticket there to ask for advice.

https://github.com/MarlinFirmware/Marlin/issues/9179

tvi675, 2018-01-14:
This looks to be a similar issue to 2299

tvi675, 2018-01-14:
Edit to the above report, issue is also occurring when in safe mode.

foosel, 2018-01-15:
The current prusa fork of marlin has some serious serial communication issues. See also #2285 and https://shop.prusa3d.com/forum/original-prusa-i3-mk3-f30/octoprint-issues-and-tips-t13156.html.

I wouldn't be surprised if the above is a manifestation of the same underlying problem. I suggest to downgrade either your firmware or OctoPrint to test. If the same issue persists with downgraded OctoPrint or vanished with downgraded firmware, it indeed is the firmware.

tvi675, 2018-01-15:
I'll roll back to the previous prusa firmware later on and will report back, thank you.

token47, 2018-01-15:
I'm on Marlin 1.1.8. Sorry if I'm poluting this issue, I can start a different one if you want.

foosel, 2018-01-15:
@token47 yours definitely looks like a different issue if this is stock Marlin. Missing ok rather looks like an issue in firmware or a really bad communication problem, especially with this weird spliced in "N25" which doesn't make any sense at all there. Make sure to read [this FAQ entry (and the one linked in it)](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-keeps-running-into-communication-errors-and-timeouts) on possible causes for the latter (especially check that if you are using a pi is power supply not only fits the specification on paper but actually does provide what it claims to provide in power).

token47, 2018-01-15:
Yeah, that's why I opened the firmware issue concurrently. I don't think it's a power supply issue because I not only use a 5V 7A (yeah, seven amperes) supply but also supply that same 5V directly to ramps also (to overcome the 1.2A limit of USB ports just in case).

From the log I guess it's using checksums and line numbering (things that in simplify3d I have to enable manually but in octoprint is automatic). Not sure if this is relevant.

Also, two questions:

Shouldn't there be a reaction for this kind of thing, like repeating a command or anything when an "ok" is not received in a short time in the middle of a print? I know there are long running commands but a common G0 or G1 should always have a very fast response. Not saying it's octoprint's fault, only trying to understand if there could be a workaround. At least octoprint would be officially acknowledging that it is missing the response from the firmware in a reasonable time and would help debug this type of problems. (a quick message in the serial log like "missing ok" or "timeout waiting for ok").

Secondly, is there any other config (or log) I can enable to further investigate or maybe mitigate, as for example XON XOFF?

nophead, 2018-01-15:
G0 does not have a fast response once the motion buffer is filled.

tvi675, 2018-01-16:
Rolling back to Prusa firmware v3.0.12 and maintaining Octoprint 1.3.6 seems to have solved the serial disconnects, so I suspect it is related to the wider Prusa firmware issues mentioned above.

From my point of view I’m happy that this issue can be closed as a not Octoprint issue, thank you for your help. 

@token47 it may be worth creating your own issue to continue getting to the bottom of your problem.

tvi675, 2018-01-16:
Issue opened on the Prusa Firmware site at https://github.com/prusa3d/Prusa-Firmware/issues/410

foosel, 2018-01-18:
Closing since it was confirmed as a firmware issue.

@token47 

> Shouldn't there be a reaction for this kind of thing, like repeating a command or anything when an "ok" is not received in a short time in the middle of a print?

OctoPrint will "tickle" the printer with an `M105` when it repeatedly runs into a timeout to do exactly that. If the printer keeps sending messages in the meantime however (e.g. due to temperatures being auto reported), that won't trigger. So I guess it makes sense to add some additional checks here to ensure such a "tickling" also in such cases. I'll add that to my TODO list.

> Secondly, is there any other config (or log) I can enable to further investigate or maybe mitigate, as for example XON XOFF?

There's a PR in the pipeline for XON/XOFF support, but it's not yet ready. I also can't think of any other settings or logging to enable here to get to the bottom of this - the problem is a lost new line or ok somewhere between OctoPrint and your firmware. It might be related to #2370 (which I took a look at in another context and most likely is rather an issue one layer beneath OctoPrint in PySerial or even the OS driver, but further investigation is needed), but your output lines don't look long enough for that.

In any case, this issue of your's is not the same as the one observed by @tvi675 (and also behaves differently), so I suggest to rather open up a new ticket as well.

closed 2018-01-18 by foosel

token47, 2018-01-23:
Thank you! I opened that issue on marlin and I will wait a little to see what they think. Depending on that and more tests I'll open a new issue here. Thanks for now.