OctoPrint/OctoPrint #2366: Copying the API key is impossible

skorokithakis, 2018-01-14:
I am running version 1.3.5, and when trying to copy the API key with Firefox 57 on Ubuntu, it's just impossible. The control is grayed out, not selectable, the few times I do manage to select the key, I can't copy it, and there's just no way for me to get the key out of the control. All I can do is use the QR code, but I want to access the key on the same computer as the browser! I'm afraid this is too much "security" for even the legitimate user.

ntoff, 2018-01-15:
This issue has already been fixed in 1.3.6

foosel, 2018-01-15:
As @ntoff mentioned, this has already been fixed in the current stable version. Please update your instance.

closed 2018-01-15 by foosel