OctoPrint/OctoPrint #2368: Cannot load Cura profile

agribov, 2018-01-15:
#### What were you doing?

1) Set up OctoPrint as a fresh Octopi install on a Pi-3
2) Went through the "first time setup" process. It got stuck on loading a Cura profile so I skipped that step and finished the process.
3) Everything else works fine, tried loading Cura profile from the settings page and it always got hung up.

#### What did you expect to happen?

Cura profile to be loaded

#### What happened instead?

After selecting the profile to upload, it either automatically hits "confirm" and gets stuck, or sometimes I can hit confirm and it will get stuck. (screenshot below, wheel next to confirm button is endlessly spinning, I've let it spin for 10+ minutes). I've tried this in both Chrome and Firefox on Ubuntu.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.3.6

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Alunar DIY Prusa i3. Should not matter since profile loading can happen even before printer is connected.

#### Browser and Version of Browser, Operating System running Browser

Chrome and Firefox on Ubuntu 16.04

#### Link to octoprint.log

https://gist.github.com/agribov/fe13d967006124dbc8f22ad30f3cd7d5

#### Link to contents of terminal tab or serial.log

None?

#### Link to contents of Javascript console in the browser

See Screenshot below

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

agribov, 2018-01-15:
Also, I did read through issue #795 which is very similar. I just went back there and noticed that another user had the same issue and resolved it by closing the instance of Octoprint and starting a new one. I tried restarting Octoprint through the browser interface multiple times to no avail -- Is there another way to do this through the terminal?

Either way I think this issue should be up since it is causing problems, and that post had been closed years ago.

ntoff, 2018-01-15:
>UnicodeEncodeError: 'ascii' codec can't encode character u'\uff0d' in position 4: ordinal not in range(128)

It looks like there's a weird character somewhere in the file, have you perhaps manually edited the file after creating it or does it have language characters with accents (umlaut's etc)

It'd probably help figure it out if you could please include your profile that you're trying to upload as well.

foosel, 2018-01-15:
It definitely would help to have that file available. I'd actually say it's necessary to have it to debug this :)

agribov, 2018-01-17:
@foosel @ntoff, 
https://gist.github.com/agribov/d37d8e254a4d67cb7577f5e09b971465
That is the profile I was trying to upload. This profile came with my printer (They gave an SD card with useful files). Cura 15.04 reads it with no problem (both Windows and Linux).

I uploaded it to Cura a few days ago, may have changed a setting, and have been using that. Just now I went into Cura to export (the same profile), and that freshly exported profile works fine.

So it sounds like this won't affect most users, though there is something that Cura is able to read, but Octoprint cannot, so it may be worth resolving. I haven't contributed to Octoprint (just got into it last week), so I don't have the background knowledge to debug this at the moment.

closed 2018-04-09 by foosel

wwwiesel, 2018-07-29:
Hi, I try to import a ini file in version 1.3.7 with the same behavior.
Its not possible to import a profile... 
[image]

a5m0, 2018-11-05:
I also had this issue trying to import cura profiles from a previous octoprint install, ended up just SCP'ing them and they were picked up ok.

commit 67005577 (2018-01-22): Cura plugin: str to unicode

Closes #2368