OctoPrint/OctoPrint #2370: Printer stalls during Dual Extrusion Print

b-morgan, 2018-01-16:
#### What were you doing?
1. Generate gcode file for a dual color 3DBenchy using Simplify3D for a Lulzbot TAZ 6 with a Dual Extruder V2.
2. Upload gcode file to Octoprint (.stl and .gcode included in zip of octoprint.log and serial.log below)
3. Start the print

Note: This only happens with a Dual Extruder. It did not happen with the previous version of Marlin 1.0.2.22 (although I didn't have the Dual Extruder for that long before Lulzbot released the newer firmware). 

There are a couple of workarounds. One, change Temperature interval (autoreport) to 0 (i.e. poll for temperatures. Two, add a delay in Marlin at the end of "void print_heater_state(const float &c, const float &t," ... "  safe_delay(2)
  }"

Additional information can be found in:
https://github.com/MarlinFirmware/Marlin/issues/9111
https://code.alephobjects.com/T1598

#### What did you expect to happen?
Print to finish without error

#### What happened instead?
Printer stalls randomly while printing. If caught immediately, a Fake Acknowledgement will recover. If not caught immediately, there is a blob of melted filament at the nozzle and the print is ruined.

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.14.0 

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Lulzbot TAZ 6 with Dual Extruder V2, Marlin firmware 1.1.5.64

#### Browser and Version of Browser, Operating System running Browser

Firefox 57.0.4 (64-bit), Windows 10 Pro (64-bit) build 1709

#### Link to octoprint.log
#### Link to contents of terminal tab or serial.log
[PrinterStall.zip](https://github.com/foosel/OctoPrint/files/1633306/PrinterStall.zip)

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.
Yes

Phits, 2018-02-03:
I am having the same issue.

foosel, 2018-03-02:
I just realized that while I thought I had replied to this already, in fact I haven't. Sorry.

In the meantime, #2454 was opened, and that looks like it's describing a similar situation as observed here, albeit with a slightly different initial setup and also differing symptoms. 

OctoPrint uses pyserial to communicate with the printer, it doesn't create the serial connection on its own so to speak. What I can't see is any way to adjust the read buffer size (and I think that's even determined by the driver). What I might be able to do so is find an alternative to pyserial's `readline` that is currently used by OctoPrint, since that is only ever reading one byte a time, which definitely is slower than trying bulk reads.

foosel, 2018-03-08:
I've just pushed a commit that changes to a bulk read from the serial line (with internal buffer for what goes beyond the first line). Would be interesting to know if that improves things.

Phits, 2018-03-08:
I can try it out this weekend. Thanks!

b-morgan, 2018-03-08:
I downgraded the firmware in my printer to one without the added delay in the auto temperature report, set the auto report interval to 1 second (stess test), and started a print which had failed before.

Unfortunately, it failed again. 

I think I did the correct changes to Software Update plugin, i.e. changed first dropdown to Commit and left /home/pi/OctoPrint in the second dropdown. Octoprint reported one commit which I installed. I don't think I did this correctly as the comm.py does not show the changes in the above commit.

foosel, 2018-03-09:
If it doesn't show you probably didn't checkout the `maintenance` branch but are still on master or something like that. Which is actually good because yesterday's commit still had a compatibility issue with the lower range of the supported pyserial versions and simply didn't compile against pyserial 2.7. See #2468. Now fixed though.

Steps to switch to `maintenance` via SSH (assuming OctoPi, adjust as necessary if you did a manual install) since that's not possible via the UI:

```
source ~/oprint/bin/activate
cd ~/OctoPrint
git fetch
git checkout maintenance
python setup.py install
sudo service octoprint restart
```

That should work.

b-morgan, 2018-03-09:
I'm pretty sure I got the update correct. I am sad to report that the problem still exists with these fixes...

```
2018-03-09 08:59:27,619 - Send: N29617 G1 X169.149 Y128.444 E1.7332*80
2018-03-09 08:59:27,628 - Recv:  T:189.97 /190.00 B:60.00 /60.00 T0:190.07 /190.00 T1:189.97 /190.00 @:82 B@:33 @0:82 N29617ok N29617 P11 B3
2018-03-09 08:59:28,644 - Recv:  T:190.03 /190.00 B:59.94 /60.00 T0:190.13 /190.00 T1:190.03 /190.00 @:81 B@:41 @0:80 @1:81
2018-03-09 08:59:29,644 - Recv:  T:190.07 /190.00 B:59.89 /60.00 T0:190.07 /190.00 T1:190.07 /190.00 @:81 B@:48 @0:83 @1:81
```
If needed, can I use the Software Update plugin to get back to 1.3.6 release? 

I'll stay on this maintenance branch unless something else goes wrong. I can workaround this issue by using different firmware or turning off auto temperature reports.

foosel, 2018-03-09:
> I am sad to report that the problem still exists with these fixes

Hmpf. That fix was sadly the only idea that I had for now. As I said, no actual control over buffer sizes and such, the only thing I can do is read as fast as possible.

> If needed, can I use the Software Update plugin to get back to 1.3.6 release?

I think so, just switch back to release tracking.

foosel, 2018-10-29:
So... I'm going to close this, not because the problem is solved, but because it's basically a duplicate of #2454, only that that ticket has way more "meat" by now due to the analysis that took place in there. So let's continue there (even though as I just wrote there, I'm currently at a loss as to how best to continue in the first place).