OctoPrint/OctoPrint #2371: Formatting error in temperature graph

azalanono, 2018-01-16:
Hello, foosel.

I notice some little issues related to the temperature graph. Easy to fix.

#### What were you doing?

I print something. Is not related to a printing problem.
#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

Sure, as this is a formatting issue in WEB interface.

#### Version of OctoPrint

Version 1.3.6

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Mendel Max with Marlin RC08  (not important)

#### Browser and Version of Browser, Operating System running Browser

FireFox 50.1.x browser on Linux. 

#### Link to octoprint.log

Not necessary

#### Link to contents of terminal tab or serial.log

Also not necessary

#### Link to contents of Javascript console in the browser

I think this is also not necessary

#### Screenshot(s)/video(s) showing the problem:

A verbal description follows. 

I have read the FAQ.

**What happens:**
Watching the UI's temperature graph, I notice that the width of the graph area changes from time to time. This is caused by this unnecessary text "gerade eben" which appears from time to time. When the graph scrolls a little bit further, it reverts back to normal width.

**Suggestions for fixing this problem:**
It will solve the problem perfectly when -15 -14 -13 .... 0 is written right-aligned to the vertical grid lines, without repeating "Min" behind every value.  Instead, write "Min" once and right of the grid.
This will solve the problem.

You currently write in X-Axis the unit "Min" after every particular value. Why aren't you also writing °C after every particular temperature value along Y-Axis? To keep consistency, I would recommend to follow technical standards and write the unit _once_ and outside of the diagram: Above temperatures write once "°C", and right of time values write once "Min".

kantlivelong, 2018-01-16:
Hi @azalanono,

Could you provide a screenshot of the issue occurring in Safe Mode?

Thanks

foosel, 2018-02-05:
Confirmed, seems to be caused by how flot positions the axis ticks. I've removed the "just now"/"gerade eben" label and left that tick simply unlabled now (-0 or just 0 looked weird). Also added "°C" to the y axis ticks. Would have preferred to have some central axis labels, but flot doesn't support that out of the box and the existing plugins to do it that I found create way too much whitespace in the process.

Fix is pushed to `maintenance`, will soon be merged to `devel` and be released with 1.3.7

commit 3e236dc5 (2018-02-05): Temperature graph: Small axis fixes

  * remove "just now" X axis tick, we just don't show anything for less
    than a minute ago now
  * add unit to Y axis ticks

Closes #2371
closed 2018-04-09 by foosel