OctoPrint/OctoPrint #2372: Unstable Connection to Sbase Smoothieware board.

Thisismydigitalself, 2018-01-16:
#### What were you doing?

Trying to control my new printer running Sbase with Smoothieware. there is connection, i can control the printer and monitor it but connection is not stable. not like my solid rock connection to my other printer running MKS GEN 1.4 board.

#### What did you expect to happen?
Have stable connection througout the print - not the case. print fails. when printing from SD all is well.

#### What happened instead?
Printer's nozzle temp dropped to zero resulting in clogged nozzle and ruined print(s).

#### Did the same happen when running OctoPrint in safe mode?
Yes.

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint
OctoPrint 1.3.6, OctoPi 0.14

[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]

#### Printer model & used firmware incl. version
Little Monster, not sure about the firmware version.

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Chrome, Latest.
[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

kantlivelong, 2018-01-16:
Hi @Thisismydigitalself,

- Please supply logs.
- How are you connecting to the controller from OctoPrint?
- Connection settings?

ntoff, 2018-01-16:
as well as the logs, please also provide the gcode that fails to make testing easier.

Thisismydigitalself, 2018-01-16:
USB @ 115200.
Thank you. I will print a test case, gather the needed information and report back.
Should OctoPrint manage Smoothieware "right out of the box" like with MKS GEN running Marlin or  some modifications are needed beforehand?

kantlivelong, 2018-01-16:
I use 250000 baud over USB with the official Smoothieboard without issues.

Thisismydigitalself, 2018-01-16:
Just heads-up:
Sent a model from Cura using its built in API to OctoPrint. file loaded, but never finishes "Still Stabilizing" stage. after 2 minutes i Canceled printing and manually pressed the PRINT buttom on OctoPrint main page and printing started immediately and still going..

More will follow.

Thisismydigitalself, 2018-01-16:
1 Hour print finished successfully.
What i don't understand is why i had to manually start the print cause this does not happen with my other printer running Marlin.

[image]

kantlivelong, 2018-01-16:
logs?

Thisismydigitalself, 2018-01-16:
[octoprint.log](https://github.com/foosel/OctoPrint/files/1636493/octoprint.log)

kantlivelong, 2018-01-16:
What happens if you print from OctoPrint instead of SD?

Thisismydigitalself, 2018-01-16:
my last posts are showing printing from OctoPrint, not from SD unless i am missing something.
Are you saying that when I manually pressed Print on OctoPrint main page the file was streaming from SD?

kantlivelong, 2018-01-16:
The first picture has "State: Sending files to SD". Is there an option in Cura to specify the upload target? Perhaps you have it set to SD.

foosel, 2018-01-16:
Your first screenshot (the one with "still stabilizing") showed streaming a file to SD. The picture you made of your printer's display also only hinted at a file printed from SD. Your second screenshot ("1 hour print finished") shows an actual print via OctoPrint. I don't know about @kantlivelong but at least to me right now it's completely unclear with which operation mode you are having what problems.

kantlivelong, 2018-01-16:
You stated that you had to cancel the print when it said "stabilizing." but in the same screenshot it says "Sending files to SD".  The 3rd screenshot with the gcode viewer doesn't indicate that you are printing from SD. It certainly appears that you have Cura configured to print from SD.

Thisismydigitalself, 2018-01-16:
> Your first screenshot (the one with "still stabilizing") showed streaming a file to SD

> 

I guess this is it. i now see i left the "Store on Printer's SD card" check box checked on Cura's API. 
this might be the culprit, I will uncheck it, test again and will report back.

Thank you.

Thisismydigitalself, 2018-01-16:
OK, To sum things up:

1.
The fact that my printer did not start printing even after couple of minutes was related to Cura's API settings where the defaults include "Store G-Code on the printer SD card" which i left checked without knowing it has such an impact on performance and that it takes so long to copy a file to printer's SD card cause the file is not that big to begin with.

2. 
I still don't know why the other day my printer stopped mid-print dropping the nozzle temp down to zero and ruined the print.

I hope things will stay as they are now. looking good.
Cura & OctoPrint makes a great couple.

Thank you all for being here for me.
Take Care.

ntoff, 2018-01-17:
If you're using the latest version of Cura, try a previous one. I've seen people here and on reddit having issues with the most recent version outputting incorrect gcodes related to temperatures.

https://github.com/foosel/OctoPrint/issues/2283#event-1420163871

While it might not be the exact same error (or maybe it is if it's temperature related), it wouldn't surprise me if there's more ~~bugs~~ undocumented "features" in the latest version of Cura.

Thisismydigitalself, 2018-01-17:
Regarding Cura's "stability" issues, I myself filed many issues on Cura's github repository.
I still like to work with Cura and I hope they will iron out most of the "Undocumented features" sooner than later.

I still don't know why and how to fix the issue of my smoothie touch screen not getting updated with printer current status while printing.

kantlivelong, 2018-01-17:
Generally the LCD only works with SD print but most firmware out there will allow a line of text or a progress indicator update. Check out the [plugins repo](https://plugins.octoprint.org)

Thisismydigitalself, 2018-01-17:
I checked but none worked. what i need to double check is why it works with my other printer running Marlin in oppose to Smoothieware. it might be that with the other one i use this save g-code to SD option or - Smoothieware does not support M73 nor M117 commands.

kantlivelong, 2018-01-18:
Closing as this is not octoprint and it looks like the original issue is resolved.

closed 2018-01-18 by kantlivelong