OctoPrint/OctoPrint #2373: [Request] H.264 timelapse output

povlhp, 2018-01-16:
I have a feature request. My iPhone will not play the current MPEG2 videos created by the timelapse function. So I want a H.264 output.
It is pretty trivial to modify timelapse.py to do it manually.

I just changed to the line below. I am not a python developer. But the way I see it implemented more universal is to let the user enter/pick a vcodec in the GUI settings, a field for codec options - Default could be -b bitrate, and I could replace with -preset fast
And then we need a way to pick the format of the stream. Was vob, changed to mp4.
The file can keep the .mpg extension, but even this could be user configurable.

libx264 is available for the pi.
                 
command = [
                        ffmpeg, '-framerate', str(fps), '-loglevel', 'error', '-i', '"{}"'.format(input), '-vcodec', 'libx264', "-preset fast",
                        '-threads', str(threads), '-r', "25", '-y',
#'-b', str(bitrate),
                        '-f', 'mp4'
]

ntoff, 2018-01-17:
Not quite a duplicate as such, but there's already a brainstorming thread on more advanced timelapse functionality (with one of the items being able to add your own custom command line for ffmpeg) https://github.com/foosel/OctoPrint/issues/1980

povlhp, 2018-01-17:
I agree. If  #1980 is implemented, then I am happy, and I can get my H.264.
But since H.264 is more well supported than MPEG2 these days, changing the default (and requiring libx264) would make sense if nothing else. So I think a simple patch switching default format would be easy to implement, and a gain for most users.

ntoff, 2018-01-19:
If you want to get into the advanced stuff right now, you can create wrapper scripts that render out videos using anything you want https://github.com/guysoft/OctoPi/issues/184 It's a bit outdated now as the timelapse image names have changed a bit since then, but should give you a general idea of what to do in the mean time.

povlhp, 2018-01-19:
#184 will allow me to do it. But I still think it should be made userfriendly. I close it for now, and leave it to #1980 to work on a more user friendly solution.

closed 2018-01-19 by povlhp

foosel, 2019-11-13:
This has actually been implemented by #2410 which will be part of 1.4.0.