OctoPrint/OctoPrint #2374: OctoPrint Times out after 0-45 seconds of connection

SyntaxWarrior, 2018-01-20:
I'm pretty sure this is not an octoprint issue but a hardware issue with my printer. But I'm not sure and at this point I have no idea how to proceed. And I dont know where else I might get help as the Tevo support does not answer my emails.
If this really is a hardware problem, I would really appreciate suggestions for what controller board I should buy that works well with octoprint.

#### What were you doing?
I can connect to my Tevo Tarantula using octoprint. But some where between 0-45 seconds after the connection the printer stops responding and eventually I get the following message in the terminal: 
```
"Send: M105
No response from printer after 11 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from 'Operational' to 'Offline: Too many consecutive timeouts, printer still connected and alive?'
Connection closed, closing down monitor"
```

#### Ideally provide exact steps to follow in order to reproduce your problem:

Press Connect with baud 115200 and dev/ttyUSB0.
View messages in terminal
Usually the printer gets all the way though returning a list of files that are on the SD card, and then Octoprint sends M105 continously untill the printer stops responding.
Sometimes the communication stops half way through the SD card file list. 
Sometimes it takes much longer, like 45 seconds for it to disconnect. It never makes it to 1 minute of connection time.

#### What did you expect to happen?
For it to stay connected indefinitely (at least longer than 45 seconds preferably)

#### What happened instead?
Disconnect/timeout, at best, after 45 seconds.

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
OctoPrint 1.3.6

####Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version
Tevo Tarantula. Says Migbot 3D printer on display. I have not touched printer firmware.
Have not figured out how to get the actual version of it yet.

#### Browser and Version of Browser, Operating System running Browser
Chrome Version 63.0.3239.132

#### Link to octoprint.log
https://pastebin.com/3MXprTMQ

#### Link to contents of terminal tab or serial.log
https://pastebin.com/LMjPPbFh

#### Screenshot(s)/video(s) showing the problem:
https://www.youtube.com/watch?v=Vn37gt8VnFc

I have read the FAQ.

Skystriker, 2018-01-20:
Im getting the same issues with my Tevo Tornado.
It works perfectly for 1-3minutes then there is no feedback from the printer whatsoever.
Seems to be a problem between Octoprint and Tevo Frimware.
It is mostly occouring after heating up, but sometimes during the print or idle aswell.
I´ll provide my data aswell:

Link to octoprint.log
https://pastebin.com/u3Cw59B2

Link to contents of terminal tab or serial.log:
https://pastebin.com/pv69UV2A

foosel, 2018-01-22:
As you already noted, sadly your firmware simply stops responding altogether and all attempts by OctoPrint to get it to stop its silent treatment fail, forcing it to finally give up and declare defeat. Not sure if it's a hardware or a firmware issue, but it's entirely outside of OctoPrint's abilities to work around.

Instead of directly replacing the hardware, I'd suggest trying to reflash the firmware first, maybe even with another firmware entirely if possible. Worth a shot at least.

With regards to good replacement electronics, I have to admit I'm too out of the loop to make a qualified comment on this. Best rather do some research in some general purpose 3d printing community (something like r/3dprinting on reddit maybe), and if push comes to shove ask there.

ntoff, 2018-01-23:
I saw a post recently on G+ relating to failed prints on a Tevo Tarantula, same symptoms (but "marlin v2.0"), it just stops responding. I'd definitely say try out the official release of Marlin from here first https://github.com/MarlinFirmware/Marlin before looking for replacement hardware.

@Skystriker you have a LOT of errors from plugins in your log, I would stop using those plugins and create an issue ticket with the individual plugin authors, and stop using them until the errors are fixed. Figure out one issue at a time (and rule out any faulty plugins being the cause).

closed 2018-02-08 by foosel

foosel, 2018-02-12:
Follow-up on that by @Skystriker:

> Solution: Flash a custom Marlin to the board. The baudrate should be set to 115200  NOT the Tevo default of 250000. Additionally the line `dwc_otg.speed=1` has to be inserted into `/boot/cmdline.txt` just as described in the FAQ.
>
> That solved the problem for me.