OctoPrint/OctoPrint #2375: Issue with printing stored gcode files

electrickery, 2018-01-21:
#### What were you doing?

I selected an object from the Files list. When trying to Load or Load/Print the object, OctoPi complains the object doesn't fit print volume. 
"Object doesn't fit print volume
 Object in beamClip2.gcode exceeds the 
 print volume of the current selected
 printer profile, be careful when printing this.

 More
 Object exceeds print volume in depth.
 Object's bounding box: (0.00, -3.00, 0.00)
 x (136.00, 118.96, 27.95)
 Print volume: (0.00, 0.00, 0.00) x (250.00,
 210.00, 200.00)
 You can disable this check via settings > Features
 > "Enable model size detection [...]"
" 

#### What did you expect to happen?

No complaining for objects that actually fit. There was no warning at initial loading and printing was done without problem. Ignoring the warning results in another succesful print.

#### What happened instead?

The pop-up described above appeared.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.3.6 running on OctoPi 0.14.0, on a Raspberry Pi B2. 

#### Operating System running OctoPrint

Raspberry Pi B2. 

#### Printer model & used firmware incl. version

The printer is a Prusa i3 MK2 (printing volume 250x210x200), firmware 3.1.0. These values are present in the Settings > Printer Profiles tab.  
The slicer is Prusa-Slic3r 1.38.5-Prusa3d-linux64.

#### Browser and Version of Browser, Operating System running Browser

Firefox Quantum 57.0.4 (64-bit)

ntoff, 2018-01-22:
Please provide all the requested logs, and also the gcode which fails the test.

foosel, 2018-01-22:
@electrickery 🤫 Log files are relevant even if they do not contain any error messages. In OctoPrint's case they provide important information about your environment that might become a key factor in debugging.

I currently cannot reproduce this against a Prusa MK2 profile with correctly set custom bounding box that includes the initial nozzle swipe off-bed (that btw is part of your file and the reason for the message). Please verify that you actually have the correct printer profile selected - your problem very much sounds like the correct profile was selected on your first print, and a wrong one on your second attempt. 

Alternatively, it might just be that you started printing it the first time before the file analysis had been done, meaning the volume check couldn't be done, and on second attempt there was enough time for the volume to be calculated and your printer profile - lacking the custom bounding box to prevent this exact issue with off-bed nozzle wipes - didn't match up.

For comparison:

[image]

Note the negative minimum Y value in the custom bounding box definition.

electrickery, 2018-01-22:
Custom bounding box:
I never checked the "Custom bounding box" so I never did set those values. I missed that in the Prusa release notes.

Printing before check is executed:
Usually I look at the gcode viewer to make sure it is the correct file. 

Additional:
The error message is only for objects uploaded and printed at a previous session. A reboot system makes the newly uploaded 'correct'  objects 'too big'.

foosel, 2018-01-23:
The GCODE analysis used for the size check (and some other things like print time estimation) is run server side (and is paused when printing for performance reasons) and has nothing to do with the client side gcode viewer. The bounding box check also has nothing to do with the gcode viewer. 

Since your problem solved itself after you fixed your printer profile to actually match your printer's bounding box, this sounds indeed like you started your first print job before that analysis had run, hence didn't get a warning despite the file containing printed parts outside your printer's default bounding box, and then got a warning on second print since OctoPrint had had enough time to analyse the file in between. So... not a bug in OctoPrint but rather an incomplete printer profile. Closing.

closed 2018-01-23 by foosel