OctoPrint/OctoPrint #2376: API upload of file without extension/invalid extension causes exception, 500

tedder, 2018-01-21:
#### What were you doing?

When uploading gcode from slic3r through the Octoprint API, specifying a filename in slic3r like `myprint` will result in a 500 error. The trace from Octoprint is below- it isn't a "caught" 500 error, it's a raw one. Seems like it should at least be caught. My preference would be to assume a file sans extension/type is gcode by default, but you may want to reject it. Either way, it should be caught, I think.

I can reproduce this with any invalid file extension, not just a missing extension.

I can also reproduce this in the UI- it at least gives a prettier error message ("Could not upload the file. Make sure that it is a valid file with..."), but the iframe type window in that modal says "internal server error", so it's ultimately the same thing.

#### What did you expect to happen?
gcode to be uploaded

#### What happened instead?
500 error

#### Did the same happen when running OctoPrint in safe mode?
Haven't tried. Safe to assume it isn't plugin-related.

#### Version of OctoPrint
1.3.6

#### Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version
n/a

#### Browser and Version of Browser, Operating System running Browser
n/a

#### Link to octoprint.log
Nice trace here: https://gist.github.com/tedder/5ac8faaaa25738e64ed7a6bbdf791c5d

#### Link to contents of terminal tab or serial.log
n/a

#### Link to contents of Javascript console in the browser
n/a

#### Screenshot(s)/video(s) showing the problem:
Screenshots, don't think they are handy, but for completeness: https://imgur.com/a/NwfzF

I have read the FAQ.

kantlivelong, 2018-01-21:
Thanks for the report. 

Duplicate of #2316

closed 2018-01-21 by kantlivelong

tedder, 2018-01-21:
Ah, great. I missed that in my search. I see #2324 now.