OctoPrint/OctoPrint #2379: [Request] Reduce scroll sensitivity in file selection box.

Crowlord, 2018-01-23:
Possibly a bug as I did not notice this behaviour before. 

When scrolling the file list in file selection one "click" of the scroll wheel moves through several pages of items. This occurs both in chrome on raspi display and on web terminal on the remote PC (chrome again)

ntoff, 2018-01-23:
version of octoprint? seems normal here on 1.3.7-dev branch, one click = scroll by one file

Crowlord, 2018-01-23:
1.3.6 Octopi 0.13.0

Perhaps its local to my install. Its perfectly possible and would explain why I am the only person to notice it.

ntoff, 2018-01-23:
I've not noticed any issues in the year or two I've been using octoprint. I was going to suggest checking your scroll wheel settings (by line / by page) but you seem to be using different OS's? Or are you  using the same OS/mouse for local and remote access? Not really sure on your setup.

ntoff, 2018-01-24:
I just tried 1.3.6, chrome on windows 10 and the scroll is still fine. 

I'd suggest trying it in safe mode just to rule out any other plugins messing with it, go to the system menu and click restart in safemode.

Also in the past there have been issues with certain things with touch screens that don't appear on non-touch devices, does your windows 10 device also have a touch screen?

foosel, 2018-01-24:
I don't know if any way to influence scroll speed. As far as I know that's entirely in the hands of your browser and the OS, and [this comment on stack overflow appears to agree](https://stackoverflow.com/a/7408178).

Crowlord, 2018-01-24:
Hmm. Definitely a local issue then. Best close the thread :D. It's not breaking anything so I will just work around it. 

Thanks

BillyBlaze, 2018-01-24:
In addition to @ntoff, could you specify if you have a touchscreen attached? perhaps you have a laptop with a built-in touchsreen that might cause bugs in the library.

One other observation:
1. slimscroll.js has been upgraded in OctoPrint v1.3.6 to Slimscroll v1.3.8 ([af4f97f2](https://github.com/foosel/OctoPrint/commit/af4f97f2df2fbeb31a5c56d105c93e90978ae262))

@foosel, I completly agree with that comment, any reason we could not get rid of slimscroll?

Crowlord, 2018-01-24:
The desktop is standard config. No touchscreen just a graphics card to monitor via DVI.

foosel, 2018-02-08:
@BillyBlaze ah, good point, I forgot about that upgrade. In the same boat as @ntoff though, can't reproduce. 

Do you have alternatives that provide a *slim* scroll bar there? I don't want a regular one (too fat, too ugly) and we do need something there on desktop.

BillyBlaze, 2018-02-08:
Al through I get the aesthetic part, but on usability, I do miss the big default (not touch) scrollbar.

What about using the same technique as YouTube? They hide the scrollbar if you're mouse pointer is not active on the main screen, once you hover over the main view, the big scrollbar appears:

1. Hidden scrollbar: (mouse active on left part menu)
[image]

2. Visible scrollbar: (mouse active on a thumbnail)
[image]

foosel, 2018-03-26:
I've finally gotten around to testing this and I'm pleasantly surprised, somehow I thought this would look way way worse:

[image]

So I've now switched all three places in core + bundled plugins that used slimscroll over to stock scrollbars per the above commit. I'm leaving the `slimscrollForEach` plus the library in though since third party plugins might still be using those (to be honest, I'm not even sure if one of *mine* isn't...).

foosel, 2018-03-28:
@Crowlord thanks for the update! Switching to regular scrollbars should hopefully make this not reappear in the future as well I hope.

foosel, 2018-04-13:
Quick update on this: will be released with OctoPrint **1.3.9**, 1.3.8 was just turned into a hotfix release.

foosel, 2018-07-25:
1.3.9 has just been released.

closed 2018-07-25 by foosel