OctoPrint/OctoPrint #2380: printing pauses while octoprint waits for "ok"

token47, 2018-01-25:
This is kind of a continuation of  #2363 

Ok, this thing is starting to get me in the nerves so I gotta do something about it before I revert back to the old setup.

- worked for months without issues on a manually installed octoprint on a ODROID / ubuntu 16.04, updated to the last version (1.3.6).
- it is random (I mean the timing, does not seem associated with any specific event)
- happens always between commands, on the (missing) acknowledgement
- the problem is always exactly the same, it returns other 2 chars in place of "ok", so the firmware seems to be sending the ok but they get replaced when/before reaching octoprint
- it's always 2 chars, and it's always the same first 2 chars of the last line sent to firmware (see example below)
- the firmware sits there, waiting for commands as if nothing happened and as if octoprint decided to not send anything anymore, no errors, no message on display, temperatures stay updated, everything seems fine
- if I babysit the printer and click FAKE ACKNOWLEDGEMENT (on the Terminal Tab) the printing happily continues, until it happens again
- it can be eletrical noisy on the usb connection, but I doubt it. the same cable as before, same RAMPS, same everything, except for the ODROID -> RPI3 change
- Before: octoprint 1.3.6, ubuntu 16.04.4 LTS, ODROID U3. marlin 1.1.[78]. After: octoprint 1.3.6, octopi 0.14, raspberry pi 3, marlin 1.1.8 (i used 1.1.8 for a very short time on the old setup so I'm not sure 1.1.7 was better in this respect)
- this smells like a software bug. hardware bugs are not this precise in repetition (specially noisy cables, it could scramble the response but not fabricate this specific one)
- as it's the same octoprint, I bet there is a bug in the underlying parts that came with the change, I need to go deeper but don't know how
- can't think of any config in my firmware that would influence this. serial config in Marlin is shown below

Below an excerpt or the serial.log at the exact time of freezing:

```
2018-01-24 23:35:37,999 - Send: N2594 G1 X145.492 Y87.974 E1.4852*88
2018-01-24 23:35:38,016 - Recv: ok
2018-01-24 23:35:38,020 - Send: N2595 G1 X145.175 Y87.657 E1.5120*86
2018-01-24 23:35:38,040 - Recv: ok
2018-01-24 23:35:38,043 - Send: N2596 G1 X145.175 Y88.336 E1.5527*91
2018-01-24 23:35:38,062 - Recv: ok
2018-01-24 23:35:38,064 - Send: N2597 G1 X145.492 Y88.653 E1.5795*91
2018-01-24 23:35:38,089 - Recv: N2
2018-01-24 23:35:38,964 - Recv:  T:205.12 /205.00 B:60.02 /60.00 @:43 B@:59
2018-01-24 23:35:40,271 - Send: N2598 G1 X145.492 Y89.332 E1.6202*95
2018-01-24 23:35:40,290 - Recv: ok
2018-01-24 23:35:40,293 - Send: N2599 G1 X145.175 Y89.015 E1.6470*87
2018-01-24 23:35:40,306 - Recv: ok
2018-01-24 23:35:40,309 - Send: N2600 G1 X145.175 Y89.694 E1.6877*80
2018-01-24 23:35:40,319 - Recv: ok
```

Notice the "N2" at time 38,089, it hangs the printing. It continued after about 2 seconds when I clicked FAKE ACKNOWLEDGEMENT on the web interface.

Just to have an idea of the randomness, it happened at this times in the last print. Sometimes seconds apart, sometimes minutes apart:

```
pi@tp002:~/.octoprint/logs $ grep "Recv: ..$" serial.log | grep -v "ok"
2018-01-24 23:30:21,287 - Recv: N2
2018-01-24 23:30:50,066 - Recv: N2
2018-01-24 23:34:25,006 - Recv: N1
2018-01-24 23:35:38,089 - Recv: N2
2018-01-24 23:36:27,145 - Recv: N2
2018-01-24 23:37:17,527 - Recv: N3
2018-01-24 23:40:26,890 - Recv: N4
2018-01-24 23:41:37,676 - Recv: N6
2018-01-24 23:41:45,820 - Recv: N6
2018-01-24 23:46:51,706 - Recv: N8
```

The serial section of my Configuration_adv.h (all defaults)

```
// @section serial

// The ASCII buffer for serial input
#define MAX_CMD_SIZE 96
#define BUFSIZE 4

// Transmission to Host Buffer Size
// To save 386 bytes of PROGMEM (and TX_BUFFER_SIZE+3 bytes of RAM) set to 0.
// To buffer a simple "ok" you need 4 bytes.
// For ADVANCED_OK (M105) you need 32 bytes.
// For debug-echo: 128 bytes for the optimal speed.
// Other output doesn't need to be that speedy.
// :[0, 2, 4, 8, 16, 32, 64, 128, 256]
#define TX_BUFFER_SIZE 0

// Host Receive Buffer Size
// Without XON/XOFF flow control (see SERIAL_XON_XOFF below) 32 bytes should be enough.
// To use flow control, set this buffer size to at least 1024 bytes.
// :[0, 2, 4, 8, 16, 32, 64, 128, 256, 512, 1024, 2048]
//#define RX_BUFFER_SIZE 1024

#if RX_BUFFER_SIZE >= 1024
  // Enable to have the controller send XON/XOFF control characters to
  // the host to signal the RX buffer is becoming full.
  //#define SERIAL_XON_XOFF
#endif

#if ENABLED(SDSUPPORT)
  // Enable this option to collect and display the maximum
  // RX queue usage after transferring a file to SD.
  //#define SERIAL_STATS_MAX_RX_QUEUED

  // Enable this option to collect and display the number
  // of dropped bytes after a file transfer to SD.
  //#define SERIAL_STATS_DROPPED_RX
#endif

// Enable an emergency-command parser to intercept certain commands as they
// enter the serial receive buffer, so they cannot be blocked.
// Currently handles M108, M112, M410
// Does not work on boards using AT90USB (USBCON) processors!
//#define EMERGENCY_PARSER

// Bad Serial-connections can miss a received command by sending an 'ok'
// Therefore some clients abort after 30 seconds in a timeout.
// Some other clients start sending commands while receiving a 'wait'.
// This "wait" is only sent when the buffer is empty. 1 second is a good value here.
//#define NO_TIMEOUTS 1000 // Milliseconds

// Some clients will have this feature soon. This could make the NO_TIMEOUTS unnecessary.
//#define ADVANCED_OK
```

Could an octoprint plugin do this? I'm using basically the same ones as before.

Also, I'll try some prints on 1.1.7 as soon as I get some spare time.

Thank you!

token47, 2018-01-25:
To give a little more contex:

``
pi@tp002:~/.octoprint/logs $ grep -B 1 "Recv: N.$" serial.log
2018-01-24 23:30:21,270 - Send: N251 G1 X93.140 Y73.360 E2.5049*80
2018-01-24 23:30:21,287 - Recv: N2
--
2018-01-24 23:30:50,042 - Send: N275 G1 X99.473 Y73.360 E2.5049*89
2018-01-24 23:30:50,066 - Recv: N2
--
2018-01-24 23:34:24,985 - Send: N1613 G1 X139.159 Y119.655 E23.2150*89
2018-01-24 23:34:25,006 - Recv: N1
--
2018-01-24 23:35:38,064 - Send: N2597 G1 X145.492 Y88.653 E1.5795*91
2018-01-24 23:35:38,089 - Recv: N2
--
2018-01-24 23:36:27,120 - Send: N2949 G1 X86.508 Y71.077 E80.1031*90
2018-01-24 23:36:27,145 - Recv: N2
--
2018-01-24 23:37:17,519 - Send: N3329 M117 2.20p  complete*88
2018-01-24 23:37:17,527 - Recv: N3
--
2018-01-24 23:40:26,878 - Send: N4598 G1 X105.060 Y71.992 E2.6709*94
2018-01-24 23:40:26,890 - Recv: N4
--
2018-01-24 23:41:37,553 - Send: N6142 G1 X103.561 Y156.986 E43.3126*87
2018-01-24 23:41:37,676 - Recv: N6
--
2018-01-24 23:41:45,799 - Send: N6194 G1 X96.926 Y148.008 E58.2160*103
2018-01-24 23:41:45,820 - Recv: N6
--
2018-01-24 23:46:51,689 - Send: N8345 G1 X89.860 Y92.640 E2.0874*103
2018-01-24 23:46:51,706 - Recv: N8
--
2018-01-25 00:00:57,649 - Send: N15235 G1 X149.864 Y110.992 E4.3803*90
2018-01-25 00:00:57,696 - Recv: N1
``

token47, 2018-01-25:
I'm curious about the line change, as the linux serial driver should not care about text line changes, seems like someone buffering the initial string is messing up later reading the wrong pointer.

foosel, 2018-01-25:
So... If I understand this correctly, you changed three things: you switched from an odroid to a pi, you switched from Ubuntu to octopi, and you upgraded Marlin. Your OctoPrint version is the same. That doesn't make it sound to me like OctoPrint would be the most likely culprit here.

What I'd do: downgrade marlin. Test with that. If it still persists, check the power supply on the pi. The pi is notoriously finicky with regards to it's power requirements, so something that was fine for the odroid doesn't necessarily have to be fine for the pi. That's only a shot in the dark based on communication issues reported in the past, I've never before seen something as consistent as this.

You *could* also write a plugin that basically turns `N{number}` alone on the line into `ok`, but I would consider that a last resort since it really is a hack.

PS: this sounds more like a support request, so the mailing list would have been a better place for this.

MasterPIC, 2018-01-25:
Is including line numbers and checksums for gcode commands required by your printer?

MasterPIC, 2018-01-26:
@token47 I think it's a firmware issue. The wrong answer starts with the first two chars of the last gcode. Maybe disabling line numbers for gcode commands is a possible workaround.

closed 2018-02-09 by GitIssueBot

token47, 2018-05-01:
I fought with this for some time. in my case, after much debuging, it seemed to be some corruption on the serial on the linux driver, but it only occured if I tried to use the USB at the same time to read the file and to control the printer. In my case, I put the whole system on usb (rasp 3 can boot from usb)

took me a few months and a bunch of tests in different scenarios to uncover it

interestingly it does not mater if I use the webcam while printing.

at some point the octoprint receive garbled data, an "ok" turns to garbage (generally the first two chars from the previous sent string) and octoprint hangs witing for the ok that never comes. at this exact time, if you go to the terminal tab, advanced (down the page) and click FAKE ACKNOWLEDGE, it goes on with the print, until it happens again

I even hacked the python code from the serial routine to change the text to "ok" and it solved most problems, but eventually would not catch a new string and stop again.

changed the rpi 3 three times, no changes.

the same printer went for 10 hour print without problem with direct USB to the pc.

it all stopped when I quit using the USB for the mass storage at the same time. weird. just dd'ed the install back to a sd card and the problem was gone.

Just to have in mind...