OctoPrint/OctoPrint #2382: After update to 1.36 Samba write is not permitted anymore

alfaraser, 2018-01-25:
#### What were you doing?
Hi,
I have an Octoprint set up on an OrangePi, it prints fine, all was working fine.

**Since I upgraded to ver. 1.3.6 I can not write to the Samba shared folders anymore to transfer the prints to do.**

I haven't changed the credentials nor any other setting than just doing the upgrade.

Upload via Web interface works fine, 
Mapping of the folders shared by the Octoprint works fine,
just no upload via mounted folder possible,

Do I miss something ?

ntoff, 2018-01-25:
Please see here https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report and provide *all* the requested information, without logs of any kind it's a guessing game as to why your setup no longer works.

kantlivelong, 2018-02-05:
Closing due to lack of response.

closed 2018-02-05 by kantlivelong